MichaIng/DietPi #1995: WiFi modules turn off after reboot/shutdown | Pine A64

andrewsudz, 2018-08-07:
Hello, I setup my Pine A64+ on the latest version of DietPi over Ethernet SSH. I am now trying to switch from Ethernet to onboard WiFi. I have both modules activated in dietpi-config and they are both connected.

I then run `ifdown eth0` to stop the Ethernet, unplug the ethernet cable, and switch over to the WiFi's IP on SSH and it connects. When I connect it is running on wlan0 and the Internet Test works.

I then reboot to make sure everything will work but now it won't connect. I replug my ethernet cable back in and my WiFi adapter is turned off / says Not Found. I did this a couple times now and it seems that reboots and shutdowns turn off my WiFi adapter.

Please help, thanks!

______________________________________________________________________

Also, the only way to access my Pine A64 is through SSH because their is no video output. I have tried many HDMI cables, monitors, and TVs. No success.

Other people also say this version does not support video output so I don't know if this is intentional or not. I have tried version 1.44 and the video output works on that one.

Fourdee, 2018-08-07:
@andrewsudz 

Hi, thanks for the report 👍 

Please run the following command and paste results:
```
G_CONFIG_INJECT '8723bs' '8723bs' /etc/modules
```

If the entry did not previously exist, please reboot system and see if the WiFi issue after reboot is resolved.

Fourdee, 2018-08-07:
@andrewsudz 

> Also, the only way to access my Pine A64 is through SSH because their is no video output. I have tried many HDMI cables, monitors, and TVs. No success.

Thanks for reporting this, i'll run some local testing to verify.

Fourdee, 2018-08-07:
REF: https://github.com/Fourdee/DietPi/issues/1844

andrewsudz, 2018-08-07:
It works! Ran the code above and used the same procedure from my original post and it is staying connected after reboots and shutdowns!

Thanks so much

Fourdee, 2018-08-08:
@andrewsudz 

Excellent, thanks for confirming resolved 👍 

I'll apply this fix to v6.13 during patching.

Fourdee, 2018-08-08:
Completed, resolved for v6.13.

closed 2018-08-08 by Fourdee