MichaIng/DietPi #1996: FreshRSS

msongz, 2018-08-07:
## Creating a software request:

- Software title | FreshRSS
- Short description | A free, self-hostable rss reader
- GitHub URL (if available) | https://github.com/FreshRSS/FreshRSS
- External URL (if available) | http://freshrss.org
- (Official) install documentation | 
https://github.com/FreshRSS/FreshRSS#installation

MichaIng, 2018-08-07:
@msongz 
Thanks for your request.

Jep that is possible to easily implement in comparison to feedbin.

Install steps: https://github.com/FreshRSS/FreshRSS#example-of-full-installation-on-linux-debianubuntu

MichaIng, 2018-08-08:
I will implement this with v6.14, if nothing urgent comes in between. v6.13 has already much new stuff the now needs to be tested and fine tuned, before we go on with further new features.

MichaIng, 2018-08-17:
PR is up and needs testing on all distro + webserver configurations: https://github.com/Fourdee/DietPi/pull/2016

Fourdee, 2018-08-19:
@MichaIng 

Legend, works a charm , great work adding this one, thank you 👍 

I believe this is now completed? Please close as needed.

MichaIng, 2018-08-19:
@Fourdee 
Great, jep can be closed then.

closed 2018-08-19 by MichaIng