MichaIng/DietPi #1999: Sonarr | Reckeck install after user change

MichaIng, 2018-08-09:
Ref: https://dietpi.com/phpbb/viewtopic.php?f=11&t=4368

There were some more reports about wrong permissions, we need to review what might going wrong when updating existing installs.

MichaIng, 2018-08-09:
Okay as long as all directories, where external software needs access as well (Sonarr => Samba|NZBGet) are `chown <user>:dietpi` and `chmod 664` (or better 660?), everything should work fine due to service files are executed as User=<user>; Group=dietpi. This has the same affect, as if the one user is added to the other users group, e.g. "sonarr" being member of "dietpi" group, but just for the service and the processes it executes.

Rechecked our code and permissions function. Actually everything as as above:
- $G_FP_DIETPI_USERDATA 775 dietpi:dietpi
- $G_FP_DIETPI_USERDATA/nzbget 775 nzbget:dietpi
- Sonarr service runs as sonarr:dietpi
Thus Sonarr should have access.

This was already the case with v6.12, thus not sure how the issue from first link could have occurred (644 permissions on nzbget folder).

If it's about samba, if I see it right, currently samba runs as root user. Doesn't this mean that files created via samba client are created as `root:root`, which breaks access for software?
I remember there was some reason to run samba as root, but hmm actually run it as "dietpi" should solve permissions issues.

MichaIng, 2018-08-09:
Copy&Paste about Sonarr + NZBGet:

> [mention]kazgor[/mention] 
> Okay I checked a bid:
> The NZBGet folder should be owned by nzbget:dietpi, that's correct, and it should be chmod to 775, like the whole dietpi_userdata folder.
> As Sonarr service runs as "sonarr" user but "dietpi" group, it should have write access to the the nzbget folder.
> Adding sonarr to nzbget group does not change something, as the above has the same effect.
> 
> But now, hmm you say the files are created by nzbget with 644 permissions? Okay yeah that would indeed breaks write access for group members. Either nzbget needs to create files with 664 permissions (660 would be possible as well), or Sonarr needs to run as same user (not just same group).
> 
> So for now in you case, chmod 777 hammer is not needed, chmod 660 is sufficient. But yeah we need to find a way to make NZBGet and possibly other software creating files by default with 664 (or 660) permissions.

Fourdee, 2018-08-10:
@MichaIng 

Should be resolved with commit, or, run now with
```
dietpi-services stop
sed -i "/UMask=/c\UMask=0002" $G_FP_DIETPI_USERDATA/nzbget/nzbget.conf 
chmod 775 -R $G_FP_DIETPI_USERDATA
dietpi-services start

```

MichaIng, 2018-08-10:
@Fourdee 
Jep in forum we also found the umask setting 👍.
Use `G_CONFIG_INJECT` to set it 😃?

MichaIng, 2018-08-10:
@Fourdee 
About https://dietpi.com/phpbb/viewtopic.php?f=9&t=4253

When I do fresh sonarr install, `~/.config/NzbDrone` does not exist and is also not created after first UI access. I did't want to setup indexer + downloader now, especially since I don't have any accounts or what might need to be required 😉.

We moved data dir to dietpi_userdata and triggered reinstall on dietpi-update.
New service file starts new (?) daemon location with new data dir as command argument: https://github.com/Fourdee/DietPi/blob/testing/dietpi/dietpi-software#L11483-L11496

User says:
> But Sonarr is still looking into /root/.config/NzbDrone for the userdata
> ...
> the location of Sonarr's userdata can only be changed accesing its sqlite database, not through the UI

Indeed within the UI I can't find a way to change data dir.
Perhaps, after Sonarr first run init, after data dir was written to database, even `-data=$G_FP_DIETPI_USERDATA/sonarr` argument does not have any effect, thus data dir stays at old location?

In this case indeed we would need to create symlinks, at least as I am not familiar/keen to adjust an sqlite database?

MichaIng, 2018-08-10:
@Fourdee 
So in your case the data dir was ~/.config/nzbdrone as well first and successfully moved with dietpi-update?

I will ask user to try manually overwriting with new service and see if then the new data dir is used.

Fourdee, 2018-08-11:
@MichaIng 

> So in your case the data dir was ~/.config/nzbdrone as well first and successfully moved with dietpi-update?

Yep, tested the patch for existing installs prior to v6.12 release, was fine. We'll need to see what user response is, need more info.

Fourdee, 2018-08-13:
No response from user and unable to replicate, marking as closed.

closed 2018-08-13 by Fourdee