MichaIng/DietPi #2001: hdparm | Recheck settings/auto apply to new drives

MichaIng, 2018-08-11:
Ref: https://dietpi.com/phpbb/viewtopic.php?f=9&t=4382
Man: https://manpages.debian.org/stretch/hdparm/hdparm.8.en.html

- We just apply settings to `/dev/sda`. Maybe we should move this to drive manager, automatically applying/adding an entry for each /dev/sdX drive fount?
- Adding entries then as one-liner should be possible: `/dev/sda { spindown_time = 120; apm = 254 }` and with this enabling selective parsing and enables us by times to even add power management control to drive manager UI for each drive.

Not sure about the current settings. On my RPi anyway power management is controlled by the docking station, thus can't test it here.
- `apm = 254` =>` no spindown` according to man page
- `spindown_time = 120` => 10 minutes, does this even apply with the above set?

MichaIng, 2018-08-11:
Indeed apm = 127 seems to solve it.

Fourdee, 2018-08-16:
I finally got there with the title lol

Fourdee, 2018-08-16:
Note to self:
- Write new file each time with database of drives?
```
root@DietPi:~# grep -v '^#' /etc/hdparm.conf

quiet

/dev/sda {
        #10 mins
        spindown_time = 120

        #
        apm = 254
}
```

Fourdee, 2018-08-16:
Cant use this, unless we scrape results, always returns 0:
```
root@DietPi:~# hdparm -M /dev/mmcblk0; echo $?

/dev/mmcblk0:
 HDIO_DRIVE_CMD(identify) failed: Invalid argument
0
```

Fourdee, 2018-09-06:
```hdparm``` always returns 0 exit code, even if the device does not support its feature. So we have no way of knowing if the value has been correctly applied ☹️ 
```
root@DietPi:~# G_RUN_CMD hdparm -M 255 /dev/sda
[  OK  ] hdparm -M 255 /dev/sda
root@DietPi:~# hdparm -M 255 /dev/sda

/dev/sda:
 setting acoustic management to 255
SG_IO: bad/missing sense data, sb[]:  f0 00 05 00 00 00 00 14 00 00 00 00 20 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
SG_IO: bad/missing sense data, sb[]:  f0 00 05 00 00 00 00 14 00 00 00 00 20 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 acoustic      = not supported

```

Fourdee, 2018-09-06:
Change to a global option for all drives, to set a desired spindown duration. APM is hard coded to 127 to allow spin down.

Applied directly with ```hdparm``` and saved to config with ```/etc/hdparm.conf```

To be honest, this is the best I can do at the moment, considering the lack of standardization from HDD manufactures, USB<>SATA converter manufactures, and, compatibility/features of/with ```hdparm```. Nightmare.

Fourdee, 2018-09-06:
### Testing pending:
- Set spindown to 60 seconds, wait, verify
- Reboot, wait, verify

Fourdee, 2018-09-06:
Completed. However, end user results will vary due to https://github.com/Fourdee/DietPi/issues/2001#issuecomment-418939193

closed 2018-09-06 by Fourdee

Fourdee, 2018-09-06:
[FAILED] DietPi-Drive_Manager | hdparm -B 127 /dev/mmcblk0
[FAILED] DietPi-Drive_Manager | hdparm -B 127 /dev/mmcblk0
[FAILED] DietPi-Drive_Manager | hdparm -B 127 /dev/mmcblk0

--------------------------------------------------------------------
- DietPi has encounted an error, and, is unable to continue        -
- Please create a ticket: https://github.com/Fourdee/DietPi/issues -
- Copy and paste the BLUE lines below, into the ticket             -
--------------------------------------------------------------------

#### Details:
- Date           | Thu  6 Sep 04:05:47 BST 2018
- Bug report     | N/A
- DietPi version | v6.15
- Img creator    | DietPi Core Team
- Pre-image      | Meveric
- SBC device     | Odroid C2 (aarch64) (index=12)
- Kernel version | #1 SMP PREEMPT Sun Aug 19 15:27:56 CEST 2018
- Distro         | stretch (index=4)
- Command        | hdparm -B 127 /dev/mmcblk0
- Exit code      | 22
- Software title | DietPi-Drive_Manager

#### Steps to reproduce:

1. ...
2. ...

#### Expected behaviour:

- ...

#### Actual behaviour:

- ...

#### Extra details:

- ...

#### Additional logs:
```
Log file contents:
 HDIO_DRIVE_CMD failed: Invalid argument
 HDIO_DRIVE_CMD(identify) failed: Invalid argument

/dev/mmcblk0:
 setting Advanced Power Management level to 0x7f (127)
```

--------------------------------------------------------------------

[FAILED] DietPi-Drive_Manager | Unable to continue, DietPi-Drive_Manager will now terminate.