MichaIng/DietPi #2004: DietPi-Update | GitHub update archive currently not available

MichaIng, 2018-08-12:
https://github.com/Fourdee/DietPi/archive/testing.zip

I currently get error "502: Failure", thus update breaks. But this only affects non-master branch updates, as master branch will be downloaded from dietpi.com, luckily 😉.

Fourdee, 2018-08-12:
@MichaIng 

Working fine here:
```
[  OK  ] DietPi-Update | Unpack update archieve
[  OK  ] DietPi-Update | Copy DietPi core files to ramdisk
[  OK  ] DietPi-Update | Copy rootfs files in place
```

Could be:
- flooding prevention (eg: you've downloaded that file too many times). Refresh router IP from ISP.
- Gateway issue, router/ISP/DNS playing up?

MichaIng, 2018-08-12:
@Fourdee 
Jep, also worked here now. Temporary issue. As G_CHECK_URL does not fail, good that we use dietpi.com for master branch update as first attempt.

closed 2018-08-12 by MichaIng