MichaIng/DietPi #2006: Connecting to dietpi.com (dietpi.com)|185.101.93.93|:80... failed: Connection timed out.

Valdhor, 2018-08-13:
#### Details:
- Date           | Mon 13 Aug 11:57:46 EDT 2018
- Bug report     | N/A
- DietPi version | v6.12
- Img creator    | DietPi Core Team
- Pre-image      | Meveric
- SBC device     | Odroid C2 (aarch64) (index=12)
- Kernel version | #1 SMP PREEMPT Wed Apr 18 16:59:34 CEST 2018
- Distro         | stretch (index=4)
- Command        | Connection test: dietpi.com
- Exit code      | 4
- Software title | DietPi-Survey

#### Steps to reproduce:

Select Sonarr to install and run the install

#### Expected behaviour:
Installs Sonarr

#### Actual behaviour:
Connection to diet.com fail. All other devices still have internet access

#### Extra details:

- ...

#### Additional logs:
```
Log file contents:
Spider mode enabled. Check if remote file exists.
--2018-08-13 11:57:19--  http://dietpi.com/
Resolving dietpi.com (dietpi.com)... 185.101.93.93
Connecting to dietpi.com (dietpi.com)|185.101.93.93|:80... failed: Connection timed out.
Giving up.
```

Fourdee, 2018-08-13:
@Valdhor 

Hi, thanks for the report 👍 

Appears to be online at the moment. 

Please can you try the following, and paste results?
```
G_CHECK_URL http://dietpi.com
```

Valdhor, 2018-08-13:
This killed dietpi altogether (Just got "host is down" every time I tried to ssh in) and I had to reflash the card. Note that this Odroid C2 does not have WIFI and is directly connected to the router via ethernet. It has just completed initial software setup after reflash.

`root@DietPi:~# G_CHECK_URL http://dietpi.com
[  OK  ] Root access verified.
[  OK  ] Connection test: http://dietpi.com
root@DietPi:~# `

Valdhor, 2018-08-13:
Hit wrong button

Valdhor, 2018-08-13:
Ordered from Ameridroid when I ordered the C2 - https://ameridroid.com/products/5v-2a-power-supply 5V/2A. Generic Chinese.

Valdhor, 2018-08-13:
root@DietPi:~# dmesg | grep eth
[    0.000000] psci: probing for conduit method from DT.
[    0.498155] eth0: PHY ID 001cc916 at 0 IRQ POLL (stmmac-0:00) active
[    0.498166] eth0: PHY ID 001cc916 at 7 IRQ POLL (stmmac-0:07)
[    0.498563] usbcore: registered new interface driver cdc_ether
[    4.285129] eth0: device MAC address 00:1e:06:34:53:1f
[    8.363271] stmmaceth c9410000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[  334.744976] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[  338.733367] stmmaceth c9410000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[  338.734200] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
root@DietPi:~#

Valdhor, 2018-08-13:
New error while trying to install Plex:

#### Details:
- Date           | Mon 13 Aug 18:40:38 BST 2018
- Bug report     | N/A
- DietPi version | v6.12
- Img creator    | DietPi Core Team
- Pre-image      | Meveric
- SBC device     | Odroid C2 (aarch64) (index=12)
- Kernel version | #1 SMP PREEMPT Wed Apr 18 16:59:34 CEST 2018
- Distro         | stretch (index=4)
- Command        | G_AGI: binutils:armhf plexmediaserver-installer:armhf
- Exit code      | 100
- Software title | DietPi-Software

#### Steps to reproduce:

1. ...
2. ...

#### Expected behaviour:

- ...

#### Actual behaviour:

- ...

#### Extra details:

- ...

#### Additional logs:
```
Log file contents:
E: Error, pkgProblemResolver::Resolve generated breaks, this may be caused by held packages.
```

Valdhor, 2018-08-13:
Already installed:
[ SUB1 ] DietPi-Services > stop
[  OK  ] DietPi-Services | stop : cron
[  OK  ] DietPi-Services | stop : sonarr
[  OK  ] DietPi-Services | stop : jackett
[  OK  ] DietPi-Services | stop : couchpotato
[  OK  ] DietPi-Services | stop : sabnzbd
[  OK  ] DietPi-Services | stop : proftpd

Valdhor, 2018-08-13:
Boot media is Sandisk Extreme 16GB

Fourdee, 2018-08-13:
@Valdhor 

Thanks for the info. Trying to replicate your installation now.

In the mean time, please run and paste results of:
```
apt-mark showhold
apt-get install -f
apt-get install -y binutils:armhf plexmediaserver-installer:armhf
```

Valdhor, 2018-08-13:
root@DietPi:~# apt-mark showhold
root@DietPi:~# apt-get install -f
Reading package lists... Done
Building dependency tree       
Reading state information... Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
root@DietPi:~# apt-get install -y binutils:armhf plexmediaserver-installer:armhf
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libmono-system-configuration4.0-cil : Depends: libmono-system-security4.0-cil (>= 1.0) but it is not going to be installed
 libmono-system-data4.0-cil : Depends: libmono-system-enterpriseservices4.0-cil (>= 1.0) but it is not going to be installed
                              Depends: libmono-system-numerics4.0-cil (>= 1.0) but it is not going to be installed
                              Depends: libmono-system-transactions4.0-cil (>= 1.0) but it is not going to be installed
 libmono-system-web4.0-cil : Depends: libmono-sqlite4.0-cil (>= 3.0.6) but it is not going to be installed
                             Depends: libmono-system-componentmodel-dataannotations4.0-cil (>= 4.0.0~alpha1) but it is not going to be installed
                             Depends: libmono-system-drawing4.0-cil (>= 3.0.6) but it is not going to be installed
                             Depends: libmono-system-enterpriseservices4.0-cil (>= 1.0) but it is not going to be installed
                             Depends: libmono-system-runtime-serialization-formatters-soap4.0-cil (>= 1.0) but it is not going to be installed
                             Depends: libmono-system-web-applicationservices4.0-cil (>= 1.0) but it is not going to be installed
                             Depends: libmono-system-web-services4.0-cil (>= 1.0) but it is not going to be installed
 mono-runtime : Depends: mono-runtime-sgen (= 5.14.0.177-0xamarin3+debian9b1) but it is not going to be installed
 mono-xsp4 : Depends: mono-xsp4-base (= 4.5-0xamarin3+debian9b1) but it is not going to be installed
E: Error, pkgProblemResolver::Resolve generated breaks, this may be caused by held packages.
root@DietPi:~#

Fourdee, 2018-08-13:
@Valdhor 

Thanks, I was able to replicate via local testing, leave it with us, we'll try and resolve and provide a solution.

Fourdee, 2018-08-13:
Note to self:
Fresh install testing required without ```binutils:armhf```. Must be a reason why we added it in the 1st place.

Fourdee, 2018-08-13:
@Valdhor 

Should have a fix tomorrow, please check back then 😃

Fourdee, 2018-08-14:
Hangs on "Add media library":
```
Aug 14 11:41:33 DietPi sh[996]: /usr/lib/plexmediaserver/Plex Tuner Service: err
or while loading shared libraries: libstdc++.so.6: cannot open shared object fil
e: No such file or directory
Aug 14 11:41:33 DietPi sh[996]: /usr/lib/plexmediaserver/Plex Tuner Service: err
or while loading shared libraries: libstdc++.so.6: cannot open shared object fil
e: No such file or directory
Aug 14 11:41:33 DietPi sh[996]: /usr/lib/plexmediaserver/Plex Tuner Service: err
or while loading shared libraries: libstdc++.so.6: cannot open shared object fil
e: No such file or directory
```

Fourdee, 2018-08-14:
Install tests:
- 🈯️ With mono pre-installed and the same applications as per @Valdhor 
- 🈯️ Fresh install without mono

Fourdee, 2018-08-14:
@Valdhor 

Plex install is fixed in v6.13, it will be released today. After updating your version of DietPi, simply install plex with:
```
dietpi-software install 42
```

If problems persist with ethernet, please:
- Try another ethernet cable
- Verify attached USB devices/drives do not exceed a total of 1A
- Reboot router

Failing that, please open a new ticket regarding the networking issue.

closed 2018-08-14 by Fourdee

Valdhor, 2018-08-14:
I thought this may have had something to do with the external USB drive I have. This would normally mount at a point with a whole string of letters and numbers. I wanted it to mount somewhere I could remember. After initial startup (Before installing any software) I would use vi to change /etc/fstab to a new mount point (I would create a new directory first in /mnt):

root@DietPi:# cd /mnt
root@DietPi:/mnt# mkdir WD8TB
root@DietPi:/mnt# chmod 1777 WD8TB/
root@DietPi:/mnt# vi /etc/fstab
root@DietPi:/mnt# mount -a
root@DietPi:/mnt# reboot
Connection to 192.168.1.237 closed by remote host.
Connection to 192.168.1.237 closed.

I would then go to DietPi-Drive_Manager, select /mnt/WD8TB and transfer User data and Swapfile (set to 1) to it. Reboot.

Then I would try installing Sonarr, SABnzbd, CouchPotato, qBitTorrent and Plex. At some point it would fail.

You may like to try with an external USB drive setup the way I did it.