MichaIng/DietPi #2007: Pi-hole | Add support for blocking page

Fourdee, 2018-08-13:
At webserver conf level?

MichaIng, 2018-08-13:
Quick fix attempt to symlink either
- `ln -sf /var/www/html/pihole /var/www/pihole` (to have correct subfolder to webroot relative paths)

or
- `ln -sf /var/www/html/pihole/index.php /var/www/index.php`
and maybe needed `ln -s /var/www/html/pihole/blockingpage.css /var/www/blockingpage.css`

as without further configuration, blocks are redirected to webroot and this is finally the same that we did before with out own `/var/www/index.php` = "blocked by Pi-hole..."

MichaIng, 2018-08-14:
Okay made some test, although without internet for my VM, as I am currently not at home.

Pi-hole starts now automatically. The only thing I did additionally was `systemctl enable pihole-FTL` (systemd-sysv forward), maybe somehow that is missing during install steps? I remember an error message on gravity step, maybe that led to service being/staying disabled?

About the blockpage:
`ln -sf /var/www/html/pihole/index.php /var/www/index.php` indeed works fine here. The .css file seems not even needed. Checking its html code, it links stylesheet to an online version, thus the local one seems unused?

With this also the 404 forward to blocking page is not needed. Not sure if it still makes sense or in which case a block actually returns 404 (instead of redirecting back to host/webroot).
Also in case of other web applications and browsing there are other reasons for 404 where the block page simply is wrong in place?

But the other /admin/ folder additions to Lighttpd should be added on our install step to `/etc/lighttpd/conf-available/dietpi-pihole.conf`. By times we can add respective entries for Apache and Nginx (adding at least the `.git*` file serve denial).

Fourdee, 2018-08-14:
@MichaIng 

> Pi-hole starts now automatically. The only thing I did additionally was systemctl enable pihole-FTL (systemd-sysv forward), maybe somehow that is missing during install steps?

Legend, works a treat 👍 

Blocking page with symlink does not work for me on lighttpd:
[image]
```
root@DietPi:~# ls -lha /var/www/index.php
lrwxrwxrwx 1 www-data www-data 30 Aug 14 13:18 /var/www/index.php -> /var/www/html/pihole/index.php
```

MichaIng, 2018-08-14:
@Fourdee
> Blocking page with symlink does not work for me on lighttpd

Strange, will retest with web access. Can't be that hard.

So issue not yet resolved right? I reopen for v6.14

MichaIng, 2018-08-14:
Okay, long testing up and down.

🈯️ Enabling blocking page from Pi-hole system, starting from our install (with Lighttpd):
```
cd /var/www
rm admin pihole
ln -s html/admin admin
ln -s html/pihole pihole
ln -s html/pihole/index.php index.php
```
Now accessing a blocked page, shows the new blocking page.
- Default Pi-hole install does not contain any index page within webroot (/var/www/html), instead just admin/ and pihole/ folders.
- Without index page available, server answers with 403, however [server.error-handler-404 from Pi-hole lighttpd.conf](https://github.com/pi-hole/pi-hole/blob/master/advanced/lighttpd.conf.debian#L30) jumps in, showing blocking page.
- If we link it directly to webroot, the 404 handler is obsolete, which is for my impression the cleaner solution?
- `ln -s html/pihole pihole` is needed, since blocking page needs `blockingpage.css` being available there:
https://github.com/pi-hole/pi-hole/blob/master/advanced/index.php#L66
https://github.com/pi-hole/pi-hole/blob/master/advanced/index.php#L231 (`pi.hole` is redirected to Pi-hole server via own DNS entry, thus not external, as I thought first.)

🈴 I tried the while night to enable blocking page for other systems that use Pi-hole server as DNS. Pages are blocked as expected, but no blocking page shows up.
I also tried it with a default non-manipulated Pi-hole install, including lighttpd etc, but still no blocking page for other systems. Maybe it's just not intended?
Maybe someone can verify.

quyentruong, 2018-08-28:
I also had the same issue after upgrading pihole to 4.0. I had to restore to previous version 3.x.
Does anyone figure out how to solve this issue?

MichaIng, 2018-09-01:
Can someone verify, that Pi-hole does not show blocking page on clients?

I tested again. Blocked pages are redirected by Pi-hole DNS server to 0.0.0.0.
On clients of course that cannot involve any blocking page:
```
root@VM-Buster:~# curl -v google.de
* Rebuilt URL to: google.de/
*   Trying 0.0.0.0...
* TCP_NODELAY set
* connect to 0.0.0.0 port 80 failed: Connection refused
* Failed to connect to google.de port 80: Connection refused
* Closing connection 0
curl: (7) Failed to connect to google.de port 80: Connection refused
```
On the server, 0.0.0.0 is redirected to localhost, which is blocking page on Pi-hole default install or DietPi install with above steps:
```
root@VM-Stretch:~# curl -v google.de
* Rebuilt URL to: google.de/
*   Trying 0.0.0.0...
* TCP_NODELAY set
* Connected to google.de (127.0.0.1) port 80 (#0)
> GET / HTTP/1.1
> Host: google.de
> User-Agent: curl/7.52.1
> Accept: */*
>
< HTTP/1.1 200 OK
< X-Pi-hole: A black hole for Internet advertisements.
< Content-type: text/html; charset=UTF-8
< Content-Length: 3921
< Date: Sat, 01 Sep 2018 15:00:52 GMT
< Server: lighttpd/1.4.45
<
<!DOCTYPE html>
<!-- Pi-hole: A black hole for Internet advertisements
*  (c) 2017 Pi-hole, LLC (https://pi-hole.net)
*  Network-wide ad blocking via your own hardware.
...
```

I tried it as said with official Pi-hole install method and also had no blocking page on clients. If someone can prove it on different network setup, then this is properly a bug, or missing feature that would be actually easy to implement:
- In case of installed web UI, redirect blocked pages to Pi-hole server IP instead of 0.0.0.0.

MichaIng, 2018-09-03:
@bboffey 
Many thanks for the link. This really clarifies everything.

So default is `NULL` blocking mode, which leads to no blocking page is shown on clients.
The benefits are obvious, not I am not sure if users mostly expect a blocking page or not. Advantage of the blocking page is also that you have a clear info that the page is blocked, instead of white page that without a bid investigation (which IP was about to open) could have several reasons.
Also that you can directly add pages to whitelist via blocking page is nice.

I mark this as no DietPi issue then. Also as blocking page does by default not show up on clients, does it make sense to enable it for the Pi-hole server machine then?

But the question really is now if we want to overwrite default and enable blocking page (for clients and server) or not. If it's about ads blocking that are embedded with web pages, then the connection attempt itself is really unnecessary indeed. On pop-ups and if you simply want to disallow access to certain not advertising-only web pages within your local network (tracking social media, unwanted search sites and such 😄), then a blocking page is quite informative and helpful for users.

Or we add a whiptail menu to ask users? This would be cool to have within Pi-holes own install script, but we could add it ourself (before/after we inform user about changed web UI login password) while waiting for Pi-hole to decide over PR implementation? This is properly the best solution here?

MichaIng, 2018-09-09:
@Fourdee 
To finish this, what do you think:
- Follow Pi-hole default and not show blocking page at all?
- Show blocking page for Pi-hole machine only? Not very consistent for my point of view and the Pi-hole machine, I guess, is usually not used directly for web browsing etc. Steps on existing Pi-hole install below.
```
cd /var/www
rm admin pihole
ln -s html/admin admin
ln -s html/pihole pihole
ln -s html/pihole/index.php index.php
```
- Enable blocking page for all clients via `/etc/pihole/pihole-FTL.conf` `BLOCKINGMODE=IP-NODATA-AAAA` (most compatible method, is seems): https://docs.pi-hole.net/ftldns/blockingmode/

k-plan, 2018-10-21:
@MichaIng 

I would prefer this:

-  > Or we add a whiptail menu to ask users? This would be cool to have within Pi-holes own install script, but we could add it ourself (before/after we inform user about changed web UI login password) while waiting for Pi-hole to decide over PR implementation? This is properly the best solution here?

-  > Enable blocking page for all clients via /etc/pihole/pihole-FTL.conf BLOCKINGMODE=IP-NODATA-AAAA (most compatible method, is seems): https://docs.pi-hole.net/ftldns/blockingmode/

So, durning setup whiptail, menu ask user to for his choice.
`BLOCKINGMODE=NULL`=>  is default and activated in whiptail menu, if he don't change anything.
`BLOCKINGMODE=IP-NODATA-AAAA` => is marked in whiptail menu as recommended option.

MichaIng, 2018-10-21:
@k-plan 
Agree, Pi-hole install is anyway very interactive currently (not possible differently at the moment), so one whip question more does not hurt.

I would then just use choice options Enable/Disable blocking page, which enables/disables (default) it for server (see the needed symlinks above), as well as clients. Use `BLOCKINGMODE=IP-NODATA-AAAA`, if enabled, as I see no benefit in using the others. Offering all possible options might confuse users and those who want to additionally configure, are free to manually edit the settings.

k-plan, 2018-10-21:
@MichaIng 

> ... as I see no benefit in using the other possible options. Offering all possible options might as well confuse users ..

👍  yes, spot-on. 
No need to show users any parameters.

MichaIng, 2018-10-30:
PR is up: https://github.com/Fourdee/DietPi/pull/2201

Fourdee, 2018-10-30:
Tests:
- 🈴 Nginx + yes page
```
Forbidden
You don't have permission to access /pihole on this server.
Apache/2.4.25 (Debian) Server at 192.168.0.118 Port 80
```
- 🈯️ Apache2 + no page

Fourdee, 2018-10-30:
@MichaIng 

> /admin works right? Will test as well.

Same error with admin.

html folder missing?
```
root@DietPi:~# ls -lha /var/www
total 88K
drwxrwxr-x  2 www-data www-data 4.0K Oct 30 17:00 .
drwxr-xr-x 12 root     root     4.0K Oct 30 16:58 ..
lrwxrwxrwx  1 www-data www-data   10 Oct 30 17:00 admin -> html/admin
-rwxrwxr-x  1 www-data www-data  38K Oct 30 17:00 apc.php
-rwxrwxr-x  1 www-data www-data  11K Oct 30 16:58 index.html
-rwxrwxr-x  1 www-data www-data  23K Oct 30 17:00 opcache.php
-rwxrwxr-x  1 www-data www-data   20 Oct 30 17:00 phpinfo.php
lrwxrwxrwx  1 www-data www-data   11 Oct 30 17:00 pihole -> html/pihole
```

MichaIng, 2018-10-30:
More stange:
I just did a `dietpi-software reinstall 93` to check for error messages during Pi-hole installer run that I might have overseen. Chose "reconfigure" to emulate a fresh install.
Afterwards the html folder exists:
```
root@VM-Stretch:/var/www# l
total 84
lrwxrwxrwx 1 www-data www-data    10 Oct 30 18:02 admin -> html/admin
lrwxrwxrwx 1 www-data www-data    10 Oct 30 17:53 admin.bak -> html/admin
-rwxrwxr-x 1 www-data www-data 37914 Oct 30 17:53 apc.php
drwxrwxr-x 4 www-data www-data  4096 Oct 30 18:01 html
-rwxrwxr-x 1 www-data www-data 10701 Oct 30 17:50 index.html
lrwxrwxrwx 1 www-data www-data    16 Oct 30 18:02 index.php -> pihole/index.php
lrwxrwxrwx 1 www-data www-data    16 Oct 30 17:53 index.php.bak -> pihole/index.php
-rwxrwxr-x 1 www-data www-data 22915 Oct 30 17:53 opcache.php
-rwxrwxr-x 1 www-data www-data    20 Oct 30 17:53 phpinfo.php
lrwxrwxrwx 1 www-data www-data    11 Oct 30 18:02 pihole -> html/pihole
lrwxrwxrwx 1 www-data www-data    11 Oct 30 17:53 pihole.bak -> html/pihole
```
And webUI + blocking page works now.

Note to self: symlinks got backup, which should not be done if those match the ones created.

Fourdee, 2018-10-30:
@MichaIng 

> I just did a dietpi-software reinstall 93 

Yep, just did the same lol 👍

Fourdee, 2018-10-30:
@MichaIng 

Apache2 APT install, generates its own html dir:
```
root@DietPi:~# ls -lha /var/www
total 12K
drwxr-xr-x  3 root root 4.0K Oct 30 17:09 .
drwxr-xr-x 12 root root 4.0K Oct 30 17:09 ..
drwxr-xr-x  2 root root 4.0K Oct 30 17:09 html
```

Fourdee, 2018-10-31:
@MichaIng 

Legend, works a charm 👍 Thanks.

Blocking page with Apache2 doesn't appear to be working for me:
[image]
[image]
```
root@DietPi:~# cat /etc/pihole/pihole-FTL.conf
# Added by DietPi:
BLOCKINGMODE=IP-NODATA-AAAA
```

MichaIng, 2018-10-31:
@Fourdee 
> Blocking page with Apache2 doesn't appear to be working for me:

So not yet like a charm 😄.

```
root@DietPi:~# cat /etc/pihole/pihole-FTL.conf
# Added by DietPi:
BLOCKINGMODE=IP-NODATA-AAAA
```
??? So `pihole-FTL.conf` was empty when `G_CONFIG_INJECT` injected the setting, only in this case `# Added by DietPi:` will be added to allow `sed` add to end of file. There is supposed to be much more content, did something go wrong with Pi-hole installer?

Fourdee, 2018-10-31:
@MichaIng 

I believe ```/etc/pihole/pihole-FTL.conf``` are optional vars you can use to override defaults:
https://docs.pi-hole.net/ftldns/configfile/

Install vars are in ```/etc/pihole/setupVars.conf```
```
root@DietPi:~# cat /etc/pihole/setupVars.conf
PIHOLE_INTERFACE=eth0
IPV4_ADDRESS=192.168.0.113/24
IPV6_ADDRESS=
PIHOLE_DNS_1=8.8.8.8
PIHOLE_DNS_2=8.8.4.4
QUERY_LOGGING=true
INSTALL_WEB_SERVER=false
INSTALL_WEB_INTERFACE=true
LIGHTTPD_ENABLED=false
WEBPASSWORD=54a6673c814b32309806ff1067ede0155de1af6c0550d6ce51d6f62ba76e4101
```

So, is the issue Pi-hole ignoring the var, or, something our end, needs further testing/debugging.

MichaIng, 2018-10-31:
@Fourdee 
Jep, AFAIK most settings are commented by default. But still during my tests, /etc/pihole/pihole-FTL.conf contained all settings, at least as comments, and explanations about them. This should be preserved. No idea how it could be empty in your case. Will do another test.

MichaIng, 2018-10-31:
I think I was mistaken, the pihole-FTL.conf is clearly initiated as empty file: https://github.com/pi-hole/pi-hole/blob/master/automated%20install/basic-install.sh#L1241-L1246

`install` = `cp` with -o <user> -g <group> -m <permissions> and done from /dev/null, so empty.

Hmm what is the difference:
```
root@VM-Stretch:~# l /dev/nul
-rw-r--r-- 1 root root 0 Oct 31 18:16 /dev/nul
root@VM-Stretch:~# l /dev/null
crw-rw-rw- 1 root root 1, 3 Oct 31 18:12 /dev/null
```
_________
Rested with VM Stretch Pi-hole server and VM Jessie client. Installed LXDE and accessed manually blocked domain. Blocking page showed up correctly 🤔.
@Fourdee 
Which device and distro did you use during your test?

Fourdee, 2018-11-04:
@MichaIng 

Still not working for me:
```
root@DietPi:~# cat /var/log/pihole.log | grep micro
Nov  4 12:59:51 dnsmasq[692]: 424 192.168.0.5/63824 query[A] www.microsoft.com from 192.168.0.5
Nov  4 12:59:51 dnsmasq[692]: 424 192.168.0.5/63824 <unknown> www.microsoft.com is 192.168.0.118
```

Could be a DNS cache issue my end.

> Installed LXDE and accessed manually blocked domain. Blocking page showed up correctly

Works with local system only?

MichaIng, 2018-11-06:
@Fourdee 
> Works with local system only?

Will recheck

MichaIng, 2018-11-08:
@Fourdee 
Did you find time to retest, in case with another device? I tested now in multiple VM combinations. But yeah, VM only, so perhaps network or device related.

I will retest with production RPi as client and Windows system. Already wanted to, but outdated/broken VMware and/or VirtualBox network drivers didn't allow to change any adapter setting (error thrown on apply), until full purged and reinstalled both VM software + full network adapter reset 🙄.

Fourdee, 2018-11-09:
@MichaIng 

Retest with ASUS TB and block page:
- 🈯️ Local browsing
- Remote via DNS

MichaIng, 2018-11-20:
@Fourdee 
Don't get it, just retested with dedicated RPi as client:
1. Install Pi-hole on Stretch VM (default Lighttpd)
2. Enter local router/gateway as DNS server, otherwise follow installer defaults (+web UI etc.)
3. Choose YES when blocking page prompt appears.
... install finishes, pihole-FTL service starts automatically
4. Choose Stretch VM IP as DNS server via /etc/resolv.conf on RPi.
5. Add test.com (whildcard) to blacklist via Pi-hole web UI
6. 🈯️ `curl -v test.com` on RPi results in blocking page code shown.

Fourdee, 2018-11-24:
@Fourdee 

Might be a DNS cache/routing issue my end. 

Ok, retest:
- ASUS TB with pihole + blocking page lighttpd
- Odroid C2 with static IP + dns setup for ASUS TB. + vnc + chromium

- 🈯️ ```curl -v http://www.microsoft.com```
```
root@DietPi:~# curl -v http://www.microsoft.com
* Rebuilt URL to: http://www.microsoft.com/
*   Trying 192.168.0.118...
* TCP_NODELAY set
* Connected to www.microsoft.com (192.168.0.118) port 80 (#0)
> GET / HTTP/1.1
> Host: www.microsoft.com
> User-Agent: curl/7.52.1
> Accept: */*
>
< HTTP/1.1 200 OK
< X-Pi-hole: A black hole for Internet advertisements.
< Content-type: text/html; charset=UTF-8
< Content-Length: 3527
< Date: Sat, 24 Nov 2018 03:48:47 GMT
< Server: lighttpd/1.4.45
```
- 🈴 ```curl -v https://www.microsoft.com```
```
root@DietPi:~# curl -v https://www.microsoft.com
* Rebuilt URL to: https://www.microsoft.com/
*   Trying 192.168.0.118...
* TCP_NODELAY set
* connect to 192.168.0.118 port 443 failed: Connection refused
* Failed to connect to www.microsoft.com port 443: Connection refused
* Closing connection 0
curl: (7) Failed to connect to www.microsoft.com port 443: Connection refused
```

Seems https is not redirected to blocking page. Verified results in Chromium.

closed 2018-11-24 by Fourdee