MichaIng/DietPi #2008: RPi webcam boot loop issue on RPi Zero W

MichaIng, 2018-08-13:
Ref: https://dietpi.com/phpbb/viewtopic.php?f=11&t=4346

Maybe some firmware related issue, needs replication.

Fourdee, 2018-08-21:
@MichaIng 

I suspect fault with camera unit or loose ribbon/sunny cable.

However, found an issue during testing:
[image]

debugging now, then will continue with test.

Fourdee, 2018-08-21:
Notes:
- Us locale
- RPi webcam install only

Fourdee, 2018-08-21:
- 🈯️ Unable to replicate with RPi3 B+.
- 🈴 Unable to test RPi Zero, as I lack the smaller ribbon cable: https://thepihut.com/products/raspberry-pi-zero-camera-adapter. Will order one once Pateron donations come in at start of month. Spent the majority (if not all and then a little more) of last months Pateron donations on rewards and shipping out competition items. No more free competitions for a while ☹️. I waffled on there a bit 🤣

Fourdee, 2018-09-10:
🈯️ Cable on order.

Fourdee, 2018-09-12:
Unable to replicate:
- RPi Cam Control v6.4.30: mycam@DietPi-ZeroW

I suspect faulty:
- Camera
- Cable
- Bad connection
- Sunny connector loose

Marking as closed.

closed 2018-09-12 by Fourdee