MichaIng/DietPi #2010: Update RPi image to v6.13

Fourdee, 2018-08-14:
> https://twitter.com/xmijo/status/1029321254925488129

MichaIng, 2018-08-14:
Makes sense, I will also update the pending VM image once v6.13 release.

MichaIng, 2018-08-16:
🈯️ VirtualBox v6.13 image online for testing: https://dietpi.com/downloads/testing/DietPi_VirtualBox-x86_64-Stretch.7z
- [ ] VirtualBox v6.13 Buster experimental image
- [ ] VMware v6.13 image
- [ ] VMware v6.13 Buster experimental image

Fourdee, 2018-08-16:
NB: RPi image is already done, did it a few days ago.

Fourdee, 2018-08-18:
@MichaIng 

Tested VMWare images, both working great 👍 

There is an issue with our survey uploads, fails connection test. It occurred after our host went offline yesterday. However, i'll take a look.

Will mark this as closed. Great work @MichaIng 👍

closed 2018-08-18 by Fourdee