MichaIng/DietPi #2013: DietPi-Drive_Manager | Allow custom bind and tmpfs mounts

MichaIng, 2018-08-14:
## Creating a feature request:

#### Is your feature request related to a problem? Please describe:

- When users create custom bind or tmpfs mounts, next DietPi-Drive_Manager call removes them from fstab.

#### Describe the solution you'd like:

- Allow custom entries, e.g. scrape them and readd or allow adding custom entries via dietpi.txt or other settings file.

Fourdee, 2018-08-15:
```bind``` scrape and re-dump into ```fstab``` is a little risky, could pull in unintentional mounts with ```bind``` in the name. Although, cant see another way currently
https://serverfault.com/questions/613179/how-do-i-do-mount-bind-in-etc-fstab

```tmpfs``` scrape and dump should be fine if we exclude ```/DietPi```, else PREP will not generate the entry.

Fourdee, 2018-08-15:
@MichaIng 

Legend, works a charm 👍 

```
root@DietPi:~# grep '[[:blank:],]bind[[:blank:],]' /etc/fstab
/mount/source /mount/target none defaults,bind 0 0
/mount/source /mount/target none defaults,bind,test 0 0
/mount/source /mount/target none bind 0 0
```

Fourdee, 2018-08-15:
I believe for our current user base, we should not allow tmpfs/bind entries to be listed in main menu, or, allow for additions/removal.

Leave this down to manual ```/etc/fstab``` via advanced users.

Fourdee, 2018-08-15:
Passed testing, marking as completed.

Nice one @MichaIng with https://github.com/Fourdee/DietPi/issues/2013#issuecomment-413212699 👍

paradix, 2018-08-22:
Guys thnx for the effort on that but unfortunately it is still not working properly.
When I run DietPi-Drive_Manager it than changes the mount points for my HDD when there is a bind mount in the fstab.

Here is my /etc/fstab
```
#-----------------------------------------------------------
#TMPFS, MISC
#-----------------------------------------------------------
proc /proc proc defaults 0 0
tmpfs /tmp tmpfs defaults,size=1023M,noatime,nodev,nosuid,mode=1777 0 0
tmpfs   /DietPi         tmpfs   defaults,size=10m,noatime,nodev,nosuid,mode=1777                                        0 0
/mnt/6940d1c0-8dcd-4386-b726-0cda8cb87b75/downloads/site /jail/glftpd/site none defaults,nofail,bind 0 0

#-----------------------------------------------------------
#SWAPFILE
#-----------------------------------------------------------
/var/swap    none    swap    sw    0   0

#-----------------------------------------------------------
#PHYSICAL DRIVES
#-----------------------------------------------------------
UUID=e139ce78-9841-40fe-8823-96a304a09859 / auto defaults,noatime,rw 0 1
UUID=0966-01D4 /boot auto defaults,noatime,rw,nofail,x-systemd.automount 0 0
UUID=6940d1c0-8dcd-4386-b726-0cda8cb87b75 /mnt/6940d1c0-8dcd-4386-b726-0cda8cb87b75/ auto defaults,noatime,rw,nofail,x-systemd.automount 0 0
```
When I run dietpi-drive_manager with the upper /etc/fstab it shows the following 
[image]

After exiting dietpi-drive_manager the TMPFS, MISC section is untouched, but PHYSICAL DRIVES change to
```
UUID=e139ce78-9841-40fe-8823-96a304a09859 / auto defaults,noatime,rw 0 1
UUID=0966-01D4 /boot auto defaults,noatime,rw,nofail,x-systemd.automount 0 0
UUID=6940d1c0-8dcd-4386-b726-0cda8cb87b75 /jail/glftpd/site auto defaults,noatime,rw,nofail,x-systemd.automount 0 0
```

Fourdee, 2018-08-22:
@paradix 

Thanks for the report 👍 

I believe I know what the cause is, leave it with me, will hopefully reply with a fixed script soon.

Fourdee, 2018-08-22:
Strange, unable to replicate the issue. 

Ok, we'll need to debug your system please use the following ```/etc/fstab``` (which removes the additional ```/``` from ```/mnt/6940d1c0-8dcd-4386-b726-0cda8cb87b75/```:
```
#-----------------------------------------------------------
#TMPFS, MISC
#-----------------------------------------------------------
proc /proc proc defaults 0 0
tmpfs /tmp tmpfs defaults,size=1023M,noatime,nodev,nosuid,mode=1777 0 0
tmpfs   /DietPi         tmpfs   defaults,size=10m,noatime,nodev,nosuid,mode=1777                                        0 0
/mnt/6940d1c0-8dcd-4386-b726-0cda8cb87b75/downloads/site /jail/glftpd/site none defaults,nofail,bind 0 0

#-----------------------------------------------------------
#SWAPFILE
#-----------------------------------------------------------
/var/swap    none    swap    sw    0   0

#-----------------------------------------------------------
#PHYSICAL DRIVES
#-----------------------------------------------------------
UUID=e139ce78-9841-40fe-8823-96a304a09859 / auto defaults,noatime,rw 0 1
UUID=0966-01D4 /boot auto defaults,noatime,rw,nofail,x-systemd.automount 0 0
UUID=6940d1c0-8dcd-4386-b726-0cda8cb87b75 /mnt/6940d1c0-8dcd-4386-b726-0cda8cb87b75 auto defaults,noatime,rw,nofail,x-systemd.automount 0 0
```

Once completed, please run
```
systemctl daemon-reload
mount -a
DEBUG=1 dietpi-drive_manager
```

And paste results please.

paradix, 2018-08-22:
Here is the debug log from running ```dietpi-drive_manager``` after removing the trailing ```/``` and ```demon-reload```
```
[  OK  ] DietPi-Drive_Manager | Root access verified.
[ SUB1 ] DietPi-Services > stop
[  OK  ] DietPi-Services | stop : cron
[  OK  ] DietPi-Services | stop : gogs
[  OK  ] DietPi-Services | stop : nginx
[ INFO ] DietPi-Drive_Manager | Detecting drives, please wait...
[  OK  ] DietPi-Drive_Manager | Detecting currently mounted drives, please wait...
[ INFO ] DietPi-Drive_Manager |  - Detected: /dev/mmcblk0p2 > /
[ INFO ] DietPi-Drive_Manager |  - Detected: /dev/mmcblk0p1 > /boot
[ INFO ] DietPi-Drive_Manager |  - Detected: /dev/sda1 > /jail/glftpd/site
[  OK  ] DietPi-Drive_Manager | Detecting unmounted drives, please wait...
[  OK  ] DietPi-Drive_Manager | DEBUG INFO:
[  OK  ] DietPi-Drive_Manager | Index=0
aDRIVE_UUID e139ce78-9841-40fe-8823-96a304a09859
aDRIVE_PART_UUID d3630000-02
aDRIVE_MOUNT_SOURCE /dev/mmcblk0p2
aDRIVE_MOUNT_TARGET /
aDRIVE_SOURCE_DEVICE mmcblk0
aDRIVE_FSTYPE ext4
aDRIVE_APM
aDRIVE_SIZE_TOTAL 30G
aDRIVE_SIZE_USED 3.8G
aDRIVE_SIZE_PERCENTUSED 13%
aDRIVE_SIZE_FREE 26G
aDRIVE_ISFILESYSTEM 1
aDRIVE_ISMOUNTED 1
aDRIVE_ISREADONLY_CURRENTLY 0
aDRIVE_ISNETWORKED 0
aDRIVE_ISROM 0
aDRIVE_ISPARTITIONTABLE 1

[  OK  ] DietPi-Drive_Manager | Index=1
aDRIVE_UUID 0966-01D4
aDRIVE_PART_UUID d3630000-01
aDRIVE_MOUNT_SOURCE /dev/mmcblk0p1
aDRIVE_MOUNT_TARGET /boot
aDRIVE_SOURCE_DEVICE mmcblk0
aDRIVE_FSTYPE vfat
aDRIVE_APM
aDRIVE_SIZE_TOTAL 128M
aDRIVE_SIZE_USED 48M
aDRIVE_SIZE_PERCENTUSED 37%
aDRIVE_SIZE_FREE 81M
aDRIVE_ISFILESYSTEM 1
aDRIVE_ISMOUNTED 1
aDRIVE_ISREADONLY_CURRENTLY 0
aDRIVE_ISNETWORKED 0
aDRIVE_ISROM 0
aDRIVE_ISPARTITIONTABLE 1

[  OK  ] DietPi-Drive_Manager | Index=2
aDRIVE_UUID 6940d1c0-8dcd-4386-b726-0cda8cb87b75
aDRIVE_PART_UUID 7e9d79a1-f45a-42a3-be1b-bf131c00cb05
aDRIVE_MOUNT_SOURCE /dev/sda1
aDRIVE_MOUNT_TARGET /jail/glftpd/site
aDRIVE_SOURCE_DEVICE sda
aDRIVE_FSTYPE ext4
aDRIVE_APM
aDRIVE_SIZE_TOTAL 5.5T
aDRIVE_SIZE_USED 2.0T
aDRIVE_SIZE_PERCENTUSED 38%
aDRIVE_SIZE_FREE 3.3T
aDRIVE_ISFILESYSTEM 1
aDRIVE_ISMOUNTED 1
aDRIVE_ISREADONLY_CURRENTLY 0
aDRIVE_ISNETWORKED 0
aDRIVE_ISROM 0
aDRIVE_ISPARTITIONTABLE 1
```

Here it is before running ```dietpi-drive_manager```
```
#-----------------------------------------------------------
#TMPFS, MISC
#-----------------------------------------------------------
proc /proc proc defaults 0 0
tmpfs /tmp tmpfs defaults,size=1023M,noatime,nodev,nosuid,mode=1777 0 0
tmpfs   /DietPi         tmpfs   defaults,size=10m,noatime,nodev,nosuid,mode=1777                                        0 0
/mnt/6940d1c0-8dcd-4386-b726-0cda8cb87b75/downloads/site /jail/glftpd/site none defaults,nofail,bind 0 0

#-----------------------------------------------------------
#SWAPFILE
#-----------------------------------------------------------
/var/swap    none    swap    sw    0   0

#-----------------------------------------------------------
#PHYSICAL DRIVES
#-----------------------------------------------------------
UUID=e139ce78-9841-40fe-8823-96a304a09859 / auto defaults,noatime,rw 0 1
UUID=0966-01D4 /boot auto defaults,noatime,rw,nofail,x-systemd.automount 0 0
UUID=6940d1c0-8dcd-4386-b726-0cda8cb87b75 /mnt/6940d1c0-8dcd-4386-b726-0cda8cb87b75 auto defaults,noatime,rw,nofail,x-systemd.automount 0 0
```
The ```fstab``` keeps changing even when I remove the trailing ```/```

Running ```dietpi-drive_manager``` shows this
[image]

And ```/etc/fstab``` is changed again
```
#-----------------------------------------------------------
#TMPFS, MISC
#-----------------------------------------------------------
proc /proc proc defaults 0 0
tmpfs /tmp tmpfs defaults,size=1023M,noatime,nodev,nosuid,mode=1777 0 0
tmpfs   /DietPi         tmpfs   defaults,size=10m,noatime,nodev,nosuid,mode=1777                                        0 0
/mnt/6940d1c0-8dcd-4386-b726-0cda8cb87b75/downloads/site /jail/glftpd/site none defaults,nofail,bind 0 0

#-----------------------------------------------------------
#SWAPFILE
#-----------------------------------------------------------
/var/swap    none    swap    sw    0   0

#-----------------------------------------------------------
#PHYSICAL DRIVES
#-----------------------------------------------------------
UUID=e139ce78-9841-40fe-8823-96a304a09859 / auto defaults,noatime,rw 0 1
UUID=0966-01D4 /boot auto defaults,noatime,rw,nofail,x-systemd.automount 0 0
UUID=6940d1c0-8dcd-4386-b726-0cda8cb87b75 /jail/glftpd/site auto defaults,noatime,rw,nofail,x-systemd.automount 0 0
```

Additional info is that after rebooting with the "correct" ```fstab``` in ```/proc/mounts``` I find the following entries regardig ```/dev/sda1```
```
/dev/sda1 /mnt/6940d1c0-8dcd-4386-b726-0cda8cb87b75 ext4 rw,noatime,data=ordered 0 0
/dev/sda1 /jail/glftpd/site ext4 rw,noatime,data=ordered 0 0
/dev/sda1 /jail/glftpd/site ext4 rw,noatime,data=ordered 0 0
/dev/sda1 /mnt/6940d1c0-8dcd-4386-b726-0cda8cb87b75/downloads/site ext4 rw,noatime,data=ordered 0 0
```

Fourdee, 2018-08-22:
@paradix 

Many thanks. Ok, so it appears the ```df``` result is showing the jail as mounting point. DietPi then generates the fstab based on that information.

Please can you paste results:
```
df -h
```

paradix, 2018-08-23:
Yeah, I have been checking out the ```dietpi-drive-manager``` script and realized already that it is due to ```df``` but couldn't google out anything helpful why this happens.

Here you go with the ```df``` output
```
Filesystem      Size  Used Avail Use% Mounted on
/dev/mmcblk0p2   30G  3.8G   26G  13% /
udev             10M     0   10M   0% /dev
tmpfs           344M   10M  334M   3% /run
tmpfs           859M     0  859M   0% /dev/shm
tmpfs           5.0M     0  5.0M   0% /run/lock
tmpfs           859M     0  859M   0% /sys/fs/cgroup
/dev/mmcblk0p1  128M   48M   81M  37% /boot
/dev/sda1       5.5T  2.0T  3.3T  38% /jail/glftpd/site
tmpfs            10M  1.6M  8.5M  16% /DietPi
tmpfs          1023M  536K 1023M   1% /tmp
```

Fourdee, 2018-08-24:
@paradix 

> Maybe that will help https://unix.stackexchange.com/questions/195293/bind-mounting-and-df-output
Seems to be the same issue.

Legend 👍 

Ok we can try changing our scrape to ```df -a``` or ```cat /proc/mounts```.

Please can you paste results of both those commands?

Fourdee, 2018-08-24:
@paradix 

I've updated our script to use ```df -a``` scrape. Hopefully this resolves, please update the script with:
```
wget https://raw.githubusercontent.com/Fourdee/DietPi/testing/dietpi/dietpi-drive_manager -O /DietPi/dietpi/dietpi-drive_manager
```

paradix, 2018-08-25:
works kind of :) does not mess up the mount path of the bind mounts after running ```dietpi-drive_manager``` 

here are the results.

```df -a```
```
Filesystem      1K-blocks       Used  Available Use% Mounted on
rootfs                  -          -          -    - /
sysfs                   0          0          0    - /sys
proc                    0          0          0    - /proc
udev                10240          0      10240   0% /dev
devpts                  0          0          0    - /dev/pts
tmpfs              351820       5764     346056   2% /run
/dev/mmcblk0p2   30639116    3973556   26649176  13% /
securityfs              0          0          0    - /sys/kernel/security
tmpfs              879544          0     879544   0% /dev/shm
tmpfs                5120          0       5120   0% /run/lock
tmpfs              879544          0     879544   0% /sys/fs/cgroup
cgroup                  0          0          0    - /sys/fs/cgroup/systemd
cgroup                  0          0          0    - /sys/fs/cgroup/cpuset
cgroup                  0          0          0    - /sys/fs/cgroup/debug
cgroup                  0          0          0    - /sys/fs/cgroup/cpu,cpuacct
cgroup                  0          0          0    - /sys/fs/cgroup/memory
cgroup                  0          0          0    - /sys/fs/cgroup/devices
cgroup                  0          0          0    - /sys/fs/cgroup/freezer
cgroup                  0          0          0    - /sys/fs/cgroup/net_cls,net_prio
cgroup                  0          0          0    - /sys/fs/cgroup/blkio
cgroup                  0          0          0    - /sys/fs/cgroup/perf_event
cgroup                  0          0          0    - /sys/fs/cgroup/hugetlb
systemd-1               0          0          0    - /proc/sys/fs/binfmt_misc
systemd-1               -          -          -    - /boot
systemd-1               -          -          -    - /mnt/6940d1c0-8dcd-4386-b726-0cda8cb87b75
debugfs                 0          0          0    - /sys/kernel/debug
hugetlbfs               0          0          0    - /dev/hugepages
mqueue                  0          0          0    - /dev/mqueue
configfs                0          0          0    - /sys/kernel/config
tmpfs               10240       1544       8696  16% /DietPi
tmpfs             1047552        536    1047016   1% /tmp
/dev/mmcblk0p1     130798      48218      82580  37% /boot
/dev/sda1      5814215184 2046120560 3475052192  38% /mnt/6940d1c0-8dcd-4386-b726-0cda8cb87b75
/dev/sda1      5814215184 2046120560 3475052192  38% /jail/glftpd/site
/dev/sda1      5814215184 2046120560 3475052192  38% /jail/glftpd/site
/dev/sda1      5814215184 2046120560 3475052192  38% /mnt/6940d1c0-8dcd-4386-b726-0cda8cb87b75/downloads/site
```

```/etc/fstab``` before running ```dietpi-drive_manager```
```
#-----------------------------------------------------------
#NETWORK
#-----------------------------------------------------------
#Please use DietPi-Drive_Manager to setup network mounts

#-----------------------------------------------------------
#TMPFS, MISC
#-----------------------------------------------------------
proc /proc proc defaults 0 0
tmpfs /tmp tmpfs defaults,size=1023M,noatime,nodev,nosuid,mode=1777 0 0
tmpfs   /DietPi         tmpfs   defaults,size=10m,noatime,nodev,nosuid,mode=1777                                        0 0
/mnt/6940d1c0-8dcd-4386-b726-0cda8cb87b75/downloads/site /jail/glftpd/site none defaults,nofail,bind 0 0

#-----------------------------------------------------------
#SWAPFILE
#-----------------------------------------------------------
/var/swap    none    swap    sw    0   0

#-----------------------------------------------------------
#PHYSICAL DRIVES
#-----------------------------------------------------------
UUID=e139ce78-9841-40fe-8823-96a304a09859 / auto defaults,noatime,rw 0 1
UUID=0966-01D4 /boot auto defaults,noatime,rw,nofail,x-systemd.automount 0 0
UUID=6940d1c0-8dcd-4386-b726-0cda8cb87b75 /mnt/6940d1c0-8dcd-4386-b726-0cda8cb87b75 auto defaults,noatime,rw,nofail,x-systemd.automount 0 0
```

When running ```dietpi-drive_manager``` you see

[image]

And ```/etc/fstab``` looks like that after running ```dietpi-drive_manager```
```
#-----------------------------------------------------------
#NETWORK
#-----------------------------------------------------------
#Please use DietPi-Drive_Manager to setup network mounts

#-----------------------------------------------------------
#TMPFS, MISC
#-----------------------------------------------------------
proc /proc proc defaults 0 0
tmpfs /tmp tmpfs defaults,size=1023M,noatime,nodev,nosuid,mode=1777 0 0
tmpfs   /DietPi         tmpfs   defaults,size=10m,noatime,nodev,nosuid,mode=1777                                        0 0
/mnt/6940d1c0-8dcd-4386-b726-0cda8cb87b75/downloads/site /jail/glftpd/site none defaults,nofail,bind 0 0

#-----------------------------------------------------------
#SWAPFILE
#-----------------------------------------------------------
/var/swap    none    swap    sw    0   0

#-----------------------------------------------------------
#PHYSICAL DRIVES
#-----------------------------------------------------------
UUID=e139ce78-9841-40fe-8823-96a304a09859 / auto defaults,noatime,rw 0 1
UUID=0966-01D4 /boot auto defaults,noatime,rw,nofail,x-systemd.automount 0 0
UUID=6940d1c0-8dcd-4386-b726-0cda8cb87b75 /mnt/6940d1c0-8dcd-4386-b726-0cda8cb87b75 auto defaults,noatime,rw,nofail,x-systemd.automount 0 0
UUID=6940d1c0-8dcd-4386-b726-0cda8cb87b75 /jail/glftpd/site auto defaults,noatime,rw,nofail,x-systemd.automount 0 0
UUID=6940d1c0-8dcd-4386-b726-0cda8cb87b75 /jail/glftpd/site auto defaults,noatime,rw,nofail,x-systemd.automount 0 0
UUID=6940d1c0-8dcd-4386-b726-0cda8cb87b75 /mnt/6940d1c0-8dcd-4386-b726-0cda8cb87b75/downloads/site auto defaults,noatime,rw,nofail,x-systemd.automount 0 0
```

Fourdee, 2018-08-26:
@paradix 

Thanks for testing and results 👍 

Ok, so it appears its unable to detect if the mount is a bind from ```df -a``` scrape,  thus, ```fstab``` entries are set to standard physical drive mount options.

I'll need to see if we can find a way to detect if the mount is a ```jail```, then apply the correct mount options. then we can remove the ```jail``` detection for ```#TMPFS, MISC``` which is causing the dupe entry.

Fourdee, 2018-08-27:
Notes:
```
root@DietPi:~# echo "$bind_mounts"
/mnt/C0F3-5D6B/test /jail/glftpd/site none defaults,nofail,bind 0 0
/mnt/C0F3-5D6B/test2 /jail/glftpd/site2 none defaults,nofail,bind 0 0

root@DietPi:~# findmnt --kernel -n --list | grep '\['
/jail/glftpd/site               /dev/mmcblk0p2[/mnt/C0F3-5D6B/test] ext4        rw,noatime
/jail/glftpd/site               /dev/sda1[/test]                    vfat        rw,noatime,fmask=0022,dmask=0022,codepage=437,iocharset=iso8859-1,shortname=mixed,errors=remount-ro
/mnt/C0F3-5D6B/test             /dev/sda1[/test]                    vfat        rw,noatime,fmask=0022,dmask=0022,codepage=437,iocharset=iso8859-1,shortname=mixed,errors=remount-ro
/jail/glftpd/site2              /dev/sda1[/test2]                   vfat        rw,noatime,fmask=0022,dmask=0022,codepage=437,iocharset=iso8859-1,shortname=mixed,errors=remount-ro
```

Fourdee, 2018-08-27:
@paradix

Apologies for the delay.

Resolved for v6.15. 

Testing will need to be completed on the testing branch, due to global var name changes. To test, please run:
```
#Create backup
dietpi-backup 1

#Install testing branch
G_DEV_1

#Test program
dietpi-drive_manager
```

To revert to previous state:
```
#Restore backup
dietpi-backup -1
```

I'll mark this as completed, as my testing has passed. However @paradix if you are able to test and verify, I would be grateful.

paradix, 2018-08-28:
@Fourdee What delay? :) I wish all project open source or commercial would have that responsiveness :+1: 

So, I have done the testing on the test branch. Semms all fine. ```fstab``` look fine after running ```dietpi-drive_manager```. 

Here are the logs you guys wanted
```DEBUG=1 dietpi-drive_manager```
```[  OK  ] DietPi-Drive_Manager | Root access verified.
[ SUB1 ] DietPi-Services > stop
[  OK  ] DietPi-Services | stop : cron
[  OK  ] DietPi-Services | stop : gogs
[  OK  ] DietPi-Services | stop : nginx
[ INFO ] DietPi-Drive_Manager | Detecting drives, please wait...
[  OK  ] DietPi-Drive_Manager | Detecting currently mounted drives, please wait...
[  OK  ] DietPi-Drive_Manager | BIND: match removed from df scrape: /jail/glftpd/site
[ INFO ] DietPi-Drive_Manager |  - Detected: /dev/mmcblk0p2 > /
[ INFO ] DietPi-Drive_Manager |  - Detected: /dev/mmcblk0p1 > /boot
[  OK  ] DietPi-Drive_Manager | Detecting unmounted drives, please wait...
[ INFO ] DietPi-Drive_Manager |  - Detected unmounted drive: /dev/sda1, updating fstab
[  OK  ] DietPi-Drive_Manager | DEBUG INFO:
[  OK  ] DietPi-Drive_Manager | Index=0
aDRIVE_UUID e139ce78-9841-40fe-8823-96a304a09859
aDRIVE_PART_UUID d3630000-02
aDRIVE_MOUNT_SOURCE /dev/mmcblk0p2
aDRIVE_MOUNT_TARGET /
aDRIVE_SOURCE_DEVICE mmcblk0
aDRIVE_FSTYPE ext4
aDRIVE_APM
aDRIVE_SIZE_TOTAL 30G
aDRIVE_SIZE_USED 3.7G
aDRIVE_SIZE_PERCENTUSED 13%
aDRIVE_SIZE_FREE 26G
aDRIVE_ISFILESYSTEM 1
aDRIVE_ISMOUNTED 1
aDRIVE_ISREADONLY_CURRENTLY 0
aDRIVE_ISNETWORKED 0
aDRIVE_ISROM 0
aDRIVE_ISPARTITIONTABLE 1

[  OK  ] DietPi-Drive_Manager | Index=1
aDRIVE_UUID 0966-01D4
aDRIVE_PART_UUID d3630000-01
aDRIVE_MOUNT_SOURCE /dev/mmcblk0p1
aDRIVE_MOUNT_TARGET /boot
aDRIVE_SOURCE_DEVICE mmcblk0
aDRIVE_FSTYPE vfat
aDRIVE_APM
aDRIVE_SIZE_TOTAL 128M
aDRIVE_SIZE_USED 48M
aDRIVE_SIZE_PERCENTUSED 37%
aDRIVE_SIZE_FREE 81M
aDRIVE_ISFILESYSTEM 1
aDRIVE_ISMOUNTED 1
aDRIVE_ISREADONLY_CURRENTLY 0
aDRIVE_ISNETWORKED 0
aDRIVE_ISROM 0
aDRIVE_ISPARTITIONTABLE 1

[  OK  ] DietPi-Drive_Manager | Index=2
aDRIVE_UUID 6940d1c0-8dcd-4386-b726-0cda8cb87b75
aDRIVE_PART_UUID 7e9d79a1-f45a-42a3-be1b-bf131c00cb05
aDRIVE_MOUNT_SOURCE /dev/sda1
aDRIVE_MOUNT_TARGET /mnt/6940d1c0-8dcd-4386-b726-0cda8cb87b75
aDRIVE_SOURCE_DEVICE sda
aDRIVE_FSTYPE ext4
aDRIVE_APM
aDRIVE_SIZE_TOTAL
aDRIVE_SIZE_USED
aDRIVE_SIZE_PERCENTUSED
aDRIVE_SIZE_FREE
aDRIVE_ISFILESYSTEM 1
aDRIVE_ISMOUNTED 0
aDRIVE_ISREADONLY_CURRENTLY 0
aDRIVE_ISNETWORKED 0
aDRIVE_ISROM 0
aDRIVE_ISPARTITIONTABLE 1
```

```/etc/fstab``` after  running ```dietpi-drive_manager```
```#-----------------------------------------------------------
#NETWORK
#-----------------------------------------------------------
#Please use DietPi-Drive_Manager to setup network mounts

#-----------------------------------------------------------
#TMPFS, MISC
#-----------------------------------------------------------
proc /proc proc defaults 0 0
tmpfs /tmp tmpfs defaults,size=1023M,noatime,nodev,nosuid,mode=1777 0 0
tmpfs   /DietPi         tmpfs   defaults,size=10m,noatime,nodev,nosuid,mode=1777                                        0 0
/mnt/6940d1c0-8dcd-4386-b726-0cda8cb87b75/downloads/site /jail/glftpd/site none defaults,nofail,bind 0 0

#-----------------------------------------------------------
#SWAPFILE
#-----------------------------------------------------------
/var/swap    none    swap    sw    0   0

#-----------------------------------------------------------
#PHYSICAL DRIVES
#-----------------------------------------------------------
UUID=e139ce78-9841-40fe-8823-96a304a09859 / auto defaults,noatime,rw 0 1
UUID=0966-01D4 /boot auto defaults,noatime,rw,nofail,x-systemd.automount 0 0
#UUID=6940d1c0-8dcd-4386-b726-0cda8cb87b75 /mnt/6940d1c0-8dcd-4386-b726-0cda8cb87b75 auto defaults,x-systemd.automount,noatime,rw,nofail 0 0
```

The only minor thing is that ```dietpi-drive_manager``` shows the drive as not mounted
[image]

Great job guys. Thnx a lot again.

Fourdee, 2018-08-29:
@paradix 

> [  OK  ] DietPi-Drive_Manager | Index=2
aDRIVE_UUID 6940d1c0-8dcd-4386-b726-0cda8cb87b75
aDRIVE_PART_UUID 7e9d79a1-f45a-42a3-be1b-bf131c00cb05
aDRIVE_MOUNT_SOURCE /dev/sda1
aDRIVE_MOUNT_TARGET /mnt/6940d1c0-8dcd-4386-b726-0cda8cb87b75
aDRIVE_SOURCE_DEVICE sda
aDRIVE_FSTYPE ext4
aDRIVE_ISMOUNTED 0

Thanks for testing, but this is very strange. 

If you are still using the testing branch:
- is the drive mounted in ```df -h``` ?
- what happens if you try mounting/mounting the drive in dietpi-drive_manager?

paradix, 2018-08-30:
>    is the drive mounted in df -h ?
```
Filesystem      Size  Used Avail Use% Mounted on
/dev/mmcblk0p2   30G  3.7G   26G  13% /
udev             10M     0   10M   0% /dev
tmpfs           344M   10M  334M   3% /run
tmpfs           859M     0  859M   0% /dev/shm
tmpfs           5.0M     0  5.0M   0% /run/lock
tmpfs           859M     0  859M   0% /sys/fs/cgroup
/dev/mmcblk0p1  128M   48M   81M  37% /boot
/dev/sda1       5.5T  2.0T  3.3T  38% /jail/glftpd/site
tmpfs            10M  1.6M  8.5M  16% /DietPi
tmpfs          1023M  536K 1023M   1% /tmp
```

>    what happens if you try mounting/unmounting the drive in dietpi-drive_manager?
[image]

Fourdee, 2018-08-30:
@paradix 

```
Filesystem      Size  Used Avail Use% Mounted on
/dev/mmcblk0p2   30G  3.7G   26G  13% /
udev             10M     0   10M   0% /dev
tmpfs           344M   10M  334M   3% /run
tmpfs           859M     0  859M   0% /dev/shm
tmpfs           5.0M     0  5.0M   0% /run/lock
tmpfs           859M     0  859M   0% /sys/fs/cgroup
/dev/mmcblk0p1  128M   48M   81M  37% /boot
/dev/sda1       5.5T  2.0T  3.3T  38% /jail/glftpd/site
tmpfs            10M  1.6M  8.5M  16% /DietPi
tmpfs          1023M  536K 1023M   1% /tmp
```

Strange how this is not showing the original and true ```/dev/sda1``` mount, only the jail. Maybe we do need to move back to ```df -a```.

Fourdee, 2018-08-30:
@paradix 

Script updated in testing branch, should be good to go this time 😃

Fourdee, 2018-08-31:
Note to self:

Current state breaks ASUS TB testing:
```
root@DietPi:~# DEBUG=1 dietpi-drive_manager
[ SUB1 ] DietPi-Services > stop
[  OK  ] DietPi-Services | stop : cron
[ INFO ] DietPi-Drive_Manager | Detecting drives, please wait...
[  OK  ] DietPi-Drive_Manager | Detecting currently mounted drives, please wait...
[  OK  ] DietPi-Drive_Manager | BIND: match removed from df scrape:  >
[  OK  ] DietPi-Drive_Manager | Detecting unmounted drives, please wait...
[ INFO ] DietPi-Drive_Manager |  - Detected drive with no partitions: mmcblk1
[  OK  ] DietPi-Drive_Manager | DEBUG INFO:
[  OK  ] DietPi-Drive_Manager | Index=0
aDRIVE_UUID
aDRIVE_PART_UUID
aDRIVE_MOUNT_SOURCE /dev/mmcblk1
aDRIVE_MOUNT_TARGET /tmp/mmcblk1
aDRIVE_SOURCE_DEVICE mmcblk1
aDRIVE_FSTYPE
aDRIVE_APM
aDRIVE_SIZE_TOTAL
aDRIVE_SIZE_USED
aDRIVE_SIZE_PERCENTUSED
aDRIVE_SIZE_FREE
aDRIVE_ISFILESYSTEM 0
aDRIVE_ISMOUNTED 0
aDRIVE_ISREADONLY_CURRENTLY 0
aDRIVE_ISNETWORKED 0
aDRIVE_ISROM 0
aDRIVE_ISPARTITIONTABLE 0
```

Fourdee, 2018-09-02:
🈯️Final testing:
- Testing with 2 binds
- All fstab entries are correct after running the main application
- Bind entries are not show in dietpi-drive_manager menu (as per design)
```
root@DietPi:~# DEBUG=1 dietpi-drive_manager
[  OK  ] DietPi-Drive_Manager | Root access verified.
[ SUB1 ] DietPi-Services > stop
[  OK  ] DietPi-Services | stop : cron
[ INFO ] DietPi-Drive_Manager | Detecting drives, please wait...
[  OK  ] DietPi-Drive_Manager | Detecting currently mounted drives, please wait...
[  OK  ] DietPi-Drive_Manager | BIND: match removed from df scrape: /mnt/C0F3-5D6B/test > /jail/glftpd/site
[  OK  ] DietPi-Drive_Manager | BIND: match removed from df scrape: /mnt/C0F3-5D6B/test2 > /jail/glftpd/site2
[ INFO ] DietPi-Drive_Manager |  - Detected: /dev/mmcblk0p2 > /
[ INFO ] DietPi-Drive_Manager |  - Detected: /dev/mmcblk0p1 > /boot
[ INFO ] DietPi-Drive_Manager |  - Detected: /dev/sda1 > /mnt/C0F3-5D6B
[  OK  ] DietPi-Drive_Manager | Detecting unmounted drives, please wait...
[  OK  ] DietPi-Drive_Manager | DEBUG INFO:
[  OK  ] DietPi-Drive_Manager | Index=0
aDRIVE_UUID e139ce78-9841-40fe-8823-96a304a09859
aDRIVE_PART_UUID d3630000-02
aDRIVE_MOUNT_SOURCE /dev/mmcblk0p2
aDRIVE_MOUNT_TARGET /
aDRIVE_SOURCE_DEVICE mmcblk0
aDRIVE_FSTYPE ext4
aDRIVE_APM
aDRIVE_SIZE_TOTAL 115G
aDRIVE_SIZE_USED 1.4G
aDRIVE_SIZE_PERCENTUSED 2%
aDRIVE_SIZE_FREE 114G
aDRIVE_ISFILESYSTEM 1
aDRIVE_ISMOUNTED 1
aDRIVE_ISREADONLY_CURRENTLY 0
aDRIVE_ISNETWORKED 0
aDRIVE_ISROM 0
aDRIVE_ISPARTITIONTABLE 1

[  OK  ] DietPi-Drive_Manager | Index=1
aDRIVE_UUID 0966-01D4
aDRIVE_PART_UUID d3630000-01
aDRIVE_MOUNT_SOURCE /dev/mmcblk0p1
aDRIVE_MOUNT_TARGET /boot
aDRIVE_SOURCE_DEVICE mmcblk0
aDRIVE_FSTYPE vfat
aDRIVE_APM
aDRIVE_SIZE_TOTAL 128M
aDRIVE_SIZE_USED 45M
aDRIVE_SIZE_PERCENTUSED 36%
aDRIVE_SIZE_FREE 83M
aDRIVE_ISFILESYSTEM 1
aDRIVE_ISMOUNTED 1
aDRIVE_ISREADONLY_CURRENTLY 0
aDRIVE_ISNETWORKED 0
aDRIVE_ISROM 0
aDRIVE_ISPARTITIONTABLE 1

[  OK  ] DietPi-Drive_Manager | Index=2
aDRIVE_UUID C0F3-5D6B
aDRIVE_PART_UUID 87b427d4-c076-4a64-a02f-0f6f4a758508
aDRIVE_MOUNT_SOURCE /dev/sda1
aDRIVE_MOUNT_TARGET /mnt/C0F3-5D6B
aDRIVE_SOURCE_DEVICE sda
aDRIVE_FSTYPE vfat
aDRIVE_APM
aDRIVE_SIZE_TOTAL 15G
aDRIVE_SIZE_USED 24K
aDRIVE_SIZE_PERCENTUSED 1%
aDRIVE_SIZE_FREE 15G
aDRIVE_ISFILESYSTEM 1
aDRIVE_ISMOUNTED 1
aDRIVE_ISREADONLY_CURRENTLY 0
aDRIVE_ISNETWORKED 0
aDRIVE_ISROM 0
aDRIVE_ISPARTITIONTABLE 1

```

Resolved for v6.15.

closed 2018-09-02 by Fourdee

Joulinar, 2020-06-16:
@MichaIng 
Sorry for hijacking the old topic. Do you know if Drive Manager will allow custom entries in `/etc/fstab`? There is a user on the board who like to mount file systems using `sshfs`. As Drive Manager is not supporting `sshfs`, the only way is to create custom entry within `/etc/fstab`

https://dietpi.com/phpbb/viewtopic.php?f=9&t=7781

On my test, the manual created `/etc/fstab` entry was working fine as well as Drive Manager was detecting the mount correctly
```
                ┌───────────────────────────────────────────────┤ DietPi-Drive_Manager ├───────────────────────────────────────────────┐
                │ Please select a drive to see available options.                                                                      │
                │  - User data location: RootFS (/mnt/dietpi_userdata)                                                                 │
                │                                                                                                                      │
                │                    ●─ mmcblk0 ───────────────────────────────────────────────────────                                │
                │  /                 : /dev/mmcblk0p2 | ext4 | Capacity: 29G | Used: 2.5G (9%)                                         │
                │  /boot             : /dev/mmcblk0p1 | vfat | Capacity: 252M | Used: 52.5M (21%)                                      │
                │                    ●─ 192.168.0.13:/ ─────────────────────────────────────────────────────                           │
                │  /mnt/nfs_client   : 192.168.0.13:/ | Net | Capacity: 9.8G                                                           │
                │                    ●─ root@192.168.0.13:/mnt/dietpi_userdata ─────────────────────────────────────────────           │
                │  /mnt/sshfs        : root@192.168.0.13:/mnt/dietpi_userdata | Net | Capacity: 9.8G                                   │
                │                    ●─ //192.168.0.10/Share ───────────────────────────────────────────────────                       │
                │  /mnt/samba        : //192.168.0.10/Share | Net | Capacity: 2.6T | Used: 583.8G (22%)                                │
                │                    ●─ Global Options ─────────────────────────────────────────────────────                           │
                │  Idle Spindown     : Set a global idle duration, before drives power down                                            │
                │                    ●─ Add / Refresh Drives ───────────────────────────────────────────────────                       │
                │  Add network drive : Select to mount networked drives                                                                │
                │  Refresh           : Scan for recently added/removed drives                                                          │
                │                                                                                                                      │
                │                                                                                                                      │
                │                                  <Ok>                                      <Exit>                                    │
                │                                                                                                                      │
                └──────────────────────────────────────────────────────────────────────────────────────────────────────────────────────┘

```
However Drive Manager is deleting the line within `/etc/fstab`

MichaIng, 2020-06-16:
sshfs mounts are something I wanted to add for a long time as replacement for the insecure FTP mounts to be used for mounts across www, as NFS and Samba are designed for local network shares only.

Good to see that drive manager detect it correctly, but is the entry preserved correctly? For this the a UUID must exist and it is added with some default mount options aimed for physical drives. I'll add it to network drive exclusions, so the entry won't be touched and added to the network drive part of fstab.

Joulinar, 2020-06-16:
I was testing it yesterday and indeed automatic mount on reboot is not working ootb. It was needed to exchange keys between client and server to allow password free access. I was following this guide

https://www.linode.com/docs/networking/ssh/using-sshfs-on-linux/

MichaIng, 2020-06-16:
Based on guides there are two ways to add the fstab entry:
```
sshfs#user@host/remote/path /local/path fuse options
```
and
```
user@host/remote/path /local/path fuse.sshfs options
```
Could you verify that both work just the same?

> It was needed to exchange keys between client and server to allow password free access.

Yes that makes sense, although I guess there is a way to add credentials right into the fstab entry as well, e.g. via `user:pass@host/remot/path` which is usual SSH syntax. But this is nothing that I'd ship like that since highly insecure (reminds me of curlftpfs which worked like that). But probably it is possible to add password stored in a file. Else key authentication is anyway the better solution for SSH in general.

Joulinar, 2020-06-16:
ok will do some test. Should I create a new request for `sshfs` or should we stay on this old topic.

MichaIng, 2020-06-16:
We'd create a new topic for full SSHFS support, means that there is a menu entry to add SSHFS mounts like it exist for NFS and Samba.

Joulinar, 2020-06-16:
done #3610