MichaIng/DietPi #2015: Remove file `/etc/machine-id` from image to be automatically generated on first boot

carlosedp, 2018-08-16:
Every Linux distribution that uses systemd (DietPi included) contains a file called `/etc/machine-id` that is an hex value to make that install unique.

The current DietPi rootfs has this file already set so all installations have the same value. There are a couple of applications that rely on `/etc/machine-id` to generate it's IDs on a cluster (Weave Net for example) creating a conflict.

It's needed to add a step on DietPi's image generation to remove `/etc/machine-id` as it is are generated dynamically by systemd on first boot (if not present). Also, if exists, the file `/var/lib/dbus/machine-id` must be removed.

Fourdee, 2018-08-16:
@carlosedp 

Thanks for the request, we'll take a look 👍 

We actually use our own generated UUID during 1st run, unique to each device:
https://github.com/Fourdee/DietPi/blob/master/dietpi/func/dietpi-obtain_hw_model#L661-L670

> There are a couple of applications that rely on /etc/machine-id to generate it's IDs on a cluster (Weave Net for example) creating a conflict.

Ah I see, ok we should be able to patch removal of the file during update. I'll send commit and test.

Fourdee, 2018-08-16:
Test removal + reboot
```
root@DietPi:~# cat /etc/machine-id
15a3221b01944fef846859d61ce7789e
root@DietPi:~# rm /etc/machine-id
```
Reboot:
```
root@DietPi:~# cat /etc/machine-id
cat: /etc/machine-id: No such file or directory
```

Fourdee, 2018-08-16:
Completed. Resolved during v6.14 patch.

Can also be applied now:
```
rm /etc/machine-id &> /dev/null
rm /var/lib/dbus/machine-id &> /dev/null
systemd-machine-id-setup
```

closed 2018-08-16 by Fourdee

carlosedp, 2018-08-16:
Just a question @Fourdee , the solution just remove the files on the first boot right? After this the id must be kept between future reboots.

Fourdee, 2018-08-17:
@carlosedp 

> Just a question @Fourdee , the solution just remove the files on the first boot right? After this the id must be kept between future reboots.

Ok, as we did not set unique codes for each install originally:
- Pre v6.14 DietPi.img = new code will be generated during v6.14 update
- v6.14 images and above = new code will be generated during 1st run of DietPi during the last of the pre-boot stages.