MichaIng/DietPi #2017: First setup on Odroid XU4 from SD card

AKonstantin1337, 2018-08-17:
#### Details:
- Date           | Thu  3 Nov 17:17:42 GMT 2016
- Bug report     | N/A
- DietPi version | v6.13
- Img creator    | DietPi Core Team
- Pre-image      | Meveric
- SBC device     | Odroid XU3/XU4/HC1/HC2 (armv7l) (index=11)
- Kernel version | #1 SMP PREEMPT Thu Apr 5 12:46:33 UTC 2018
- Distro         | stretch (index=4)
- Command        | Connection test: https://deb.debian.org/debian/
- Exit code      | 5
- Software title | DietPi-Software

#### Steps to reproduce:
1. get a fresh XU4
2. get a fresh install of dietpi for XU4
3. get an error
4. ....
5. PROFIT

#### Expected behaviour:
- I don't really know 

#### Actual behaviour:

- error

#### Extra details:

- ...

#### Additional logs:
```
Log file contents:
Spider mode enabled. Check if remote file exists.
--2016-11-03 17:17:40--  https://deb.debian.org/debian/
Resolving deb.debian.org (deb.debian.org)... 130.89.148.14, 128.31.0.62, 149.20.4.15, ...
Connecting to deb.debian.org (deb.debian.org)|130.89.148.14|:443... connected.
ERROR: The certificate of ‘deb.debian.org’ is not trusted.
ERROR: The certificate of ‘deb.debian.org’ is not yet activated.
The certificate has not yet been activated
```

Fourdee, 2018-08-17:
@AKonstantin1337 

Thanks for the report 👍 

Please check the date on system:
```
date
```

This will occur if timesync is not successful.

To resolve, simply set the date, then during DietPi update, the new timesync system will be enabled:
```
date +%Y%m%d -s "20180817"
```

Then re-run dietpi-software to resume setup:
```
dietpi-software
```

Fourdee, 2018-08-20:
Hi,

Issue is resolved with v6.14 patch. 
We'll release this as a hotfix ASAP.

In the mean time, i've also updated the XU4 image which resolves the issue:
https://dietpi.com/downloads/images/DietPi_OdroidXU4-ARMv7-Stretch.7z

closed 2018-08-20 by Fourdee