MichaIng/DietPi #2018: Please Help

akidiep, 2018-08-17:
--------------------------------------------------------------------
- DietPi has encounted an error, and, is unable to continue        -
- Please create a ticket: https://github.com/Fourdee/DietPi/issues -
- Copy and paste the BLUE lines below, into the ticket             -
--------------------------------------------------------------------

#### Details:
- Date           | Fri 17 Aug 23:38:27 AEST 2018
- BR sent?       | 1 : 6e852772-126c-431d-97e5-5de939cd64da
- DietPi version | v6.9
- Img creator    | 0
- Pre-image      | Raspbian Lite
- SBC device     | RPi 3 Model B (armv7l) (index=3)
- Distro         | stretch (index=4)
- Command        | occ maintenance:mode --on
- Exit code      | 1
- Software title | DietPi-Services

#### Expected behaviour:

#### Actual behaviour:

#### Steps to reproduce:

#### Additional Information:

#### Additional logs:
```
Log file contents:
An unhandled exception has been thrown:
Doctrine\DBAL\DBALException: Failed to connect to the database: An exception occured in driver: SQLSTATE[HY000] [2002] No such file or directory in /var/www/owncloud/lib/private/DB/Connection.php:62
Stack trace:
#0 /var/www/owncloud/lib/composer/doctrine/dbal/lib/Doctrine/DBAL/Connection.php(429): OC\DB\Connection->connect()
#1 /var/www/owncloud/lib/composer/doctrine/dbal/lib/Doctrine/DBAL/Connection.php(389): Doctrine\DBAL\Connection->getDatabasePlatformVersion()
#2 /var/www/owncloud/lib/composer/doctrine/dbal/lib/Doctrine/DBAL/Connection.php(328): Doctrine\DBAL\Connection->detectDatabasePlatform()
#3 /var/www/owncloud/lib/composer/doctrine/dbal/lib/Doctrine/DBAL/Connection.php(623): Doctrine\DBAL\Connection->getDatabasePlatform()
#4 /var/www/owncloud/lib/private/DB/Connection.php(145): Doctrine\DBAL\Connection->setTransactionIsolation(2)
#5 /var/www/owncloud/lib/composer/doctrine/dbal/lib/Doctrine/DBAL/DriverManager.php(172): OC\DB\Connection->__construct(Array, Object(Doctrine\DBAL\Driver\PDOMySql\Driver), Object(Doctrine\DBAL\Configuration), Object(Doctrine\CommonentManager))
#6 /var/www/owncloud/lib/private/DB/ConnectionFactory.php(145): Doctrine\DBAL\DriverManager::getConnection(Array, Object(Doctrine\DBAL\Configuration), Object(Doctrine\CommonentManager))
#7 /var/www/owncloud/lib/private/Server.php(526): OC\DB\ConnectionFactory->getConnection('mysql', Array)
#8 /var/www/owncloud/lib/composer/pimple/pimple/src/Pimple/Container.php(113): OC\Server->OC\{closure}(Object(OC\Server))
#9 /var/www/owncloud/lib/private/AppFramework/Utility/SimpleContainer.php(111): Pimple\Container->offsetGet('DatabaseConnect...')
#10 /var/www/owncloud/lib/private/ServerContainer.php(87): OC\AppFramework\Utility\SimpleContainer->query('DatabaseConnect...')
#11 /var/www/owncloud/lib/private/Server.php(1211): OC\ServerContainer->query('DatabaseConnect...')
#12 /var/www/owncloud/lib/private/Server.php(402): OC\Server->getDatabaseConnection()
#13 /var/www/owncloud/lib/composer/pimple/pimple/src/Pimple/Container.php(113): OC\Server->OC\{closure}(Object(OC\Server))
#14 /var/www/owncloud/lib/private/AppFramework/Utility/SimpleContainer.php(111): Pimple\Container->offsetGet('AppConfig')
#15 /var/www/owncloud/lib/private/ServerContainer.php(87): OC\AppFramework\Utility\SimpleContainer->query('AppConfig')
#16 /var/www/owncloud/lib/private/Server.php(1140): OC\ServerContainer->query('AppConfig')
#17 /var/www/owncloud/lib/private/Server.php(580): OC\Server->getAppConfig()
#18 /var/www/owncloud/lib/composer/pimple/pimple/src/Pimple/Container.php(113): OC\Server->OC\{closure}(Object(OC\Server))
#19 /var/www/owncloud/lib/private/AppFramework/Utility/SimpleContainer.php(111): Pimple\Container->offsetGet('AppManager')
#20 /var/www/owncloud/lib/private/ServerContainer.php(87): OC\AppFramework\Utility\SimpleContainer->query('AppManager')
#21 /var/www/owncloud/lib/private/Server.php(1410): OC\ServerContainer->query('AppManager')
#22 /var/www/owncloud/lib/private/legacy/app.php(346): OC\Server->getAppManager()
#23 /var/www/owncloud/lib/private/legacy/app.php(110): OC_App::getEnabledApps()
#24 /var/www/owncloud/lib/base.php(586): OC_App::loadApps(Array)
#25 /var/www/owncloud/lib/base.php(1004): OC::init()
#26 /var/www/owncloud/console.php(64): require_once('/var/www/ownclo...')
#27 /var/www/owncloud/occ(11): require_once('/var/www/ownclo...')
#28 {main}

Fourdee, 2018-08-18:
@akidiep 

Hi,

Thanks for the report 👍 

@MichaIng 
Any chance you could take a look at this one please?

Fourdee, 2018-08-18:
Confirmed error on testing branch:

[image]

akidiep, 2018-08-18:
Hi there, already sorted. Thanks for looking at this issue.

MichaIng, 2018-08-19:
@akidiep 
Great to here. I guess MariaDB was down, or pdo_mysql PHP module?

@Fourdee 
The Nextcloud occ installation must have been failed and thrown an error message.
`Failed loading /usr/lib/php/20151012/opcache.so:  /usr/lib/php/20151012/opcache.so: cannot open shared object file: Permission denied`
- Was this the only error message of that kind? If PHP failed, then the command line install failed. I will also try on VM.

Fourdee, 2018-08-19:
@MichaIng 

> €: Could not replicate. Found and fixed minor issue on Jessie, but that did not break the install: 

Strange, issue occurred on my C2, fine on NanoPC-T4, trying reinstall + test again.

Fourdee, 2018-08-19:
@MichaIng 

🈯️ Fresh image install + update to testing is fine

Device hasn't been re-flashed for over 2 months, been using it as a testing system, so non-issue.

Marking as closed, please reopen if required.

closed 2018-08-19 by Fourdee