MichaIng/DietPi #2019: tinyMediaManager

msdos, 2018-08-18:
## Creating a software request:

#### Give us some formal software information:
- Software title | tinyMediaManager
- Short description | A multi-OS media managment tool 
- GitHub URL (if available) | https://github.com/tinyMediaManager/tinyMediaManager
- External URL (if available) | https://www.tinymediamanager.org
- (Official) install documentation | https://github.com/tinyMediaManager/tinyMediaManager/wiki/Installation#user-content-linux

#### Are there similar/alternative software titles available with DietPi-Software?

- AFAIK no.

#### What makes your requested software better than the above solutions, if available?

- N/A

#### How can DietPi make the installation easier or compatible, than following the install instructions or do APT installation, if available?

- N/A

#### Can you provide the installation steps that you would suggest DietPi-Software to do?
1. Install mentioned dependencies in https://github.com/tinyMediaManager/tinyMediaManager/wiki/Installation#user-content-linux. Java can be a problem, because it's installing a specific version.
2. The database and the software itself would be in the same folders. Maybe a symlink to the folder `data` in `.config` would be needed to be implemented to avoid increased hd usage of `/opt`.
3. The upgrade os the application is manual, calling a `.sh` from the folder itself.

#### Are you willing to help maintaining the software installation, e.g. in case of needed setup changes due to updates etc.? This is not needed, but could speed up our decision to implement it, as man power is always a topic :wink:.

- Yes

MichaIng, 2018-08-18:
@msdos
Thanks for your request.

Java 8 is fine, we use the same as default on DietPi already, since java 10 is not yet supported by all 3rd party software.

Usually the .config location can be moved to a chosen one via systemd unit WorkingDirectory value. Eg. we could move all, data and install dir to /mnt/dietpi_userdata/

msdos, 2018-08-18:
The last version is in https://release.tinymediamanager.org/dist/tmm_2.9.13_5cfe5d1_linux.tar.gz.

This application has a GUI mode and a commanline interface. All it's configuration parameters are defined in `data/config.xml` so it's a good idea to first execute it in GUI mode to properly configure the xml before using the commandline interface.

When running this application, it creates in it's own folder the following subfolders:

```
backup
├── movies.db.2018-08-16.zip
(...)
├── tvshows.db.2018-08-16.zip

cache
├── http
│   ├── 001dcff08ef6c5e390257c0c5bb091dd.0
│   └── journal
├── image
│   ├── 007d87df74d5634c7f247b8791dda2a1.jpg
├── movies.json
└── movies.json.cache

data
├── config.xml
├── movies.db
├── scraper_tmdb.conf
├── tmm.prop
└── tvshows.db

logs
├── tmm.2018-08-16.0.log.gz
(...)
└── tmm.log
```

It zipped logs and backups were done by the application itself.

I'm not familiar with systemd and WorkingDirectory to be able to give an opinion on how everything should be done so I decided to complement here with more information about the application behavior.

msdos, 2018-08-18:
Creating symbolic links for the subfolders mentioned in https://github.com/Fourdee/DietPi/issues/2019#issuecomment-414079563 **before** running the application works, the application doesn't crash. So if the application is installed in `/opt`, symbolic links can be created to point to `/mnt/dietpi_userdata/` (for example, `/mnt/dietpi_userdata/tinyMediaManager/data`, `/mnt/dietpi_userdata/tinyMediaManager/logs` and so on. I just don't know if this is a good approach or if it's already used in dietpi.

msdos, 2018-09-14:
I'm thinking if we should wait for the 3.0 release of this package since it's a new version in beta.

msdos, 2018-09-15:
Symbolic link for the `cache` folder gives problems. https://github.com/tinyMediaManager/tinyMediaManager/issues/390. Better wait.

MichaIng, 2018-12-26:
Added to FeatHub, feel free to vote for it: https://feathub.com/Fourdee/DietPi/+12