MichaIng/DietPi #2020: Error running dietpi-config first time - "The Certificate has not yet been activated"

carlosedp, 2018-08-18:
I've been facing this problem every time I reflash my test boards. The first run of dietpi-config (or dietpi-software) present me with this error:

```
───────────────────────────┤ DietPi Error Handler: ├────────────────────────────

DietPi-Software: Connection test: http://deb.debian.org/debian/
 - Exit code: 5
 - DietPi version: v6.9 | HW_MODEL:69 | HW_ARCH:3 | DISTRO:4
 - Image creator: 0
 - Pre-image: DietPi/FriendlyARM

Log file contents:
URL transformed to HTTPS due to an HSTS policy
Spider mode enabled. Check if remote file exists.
--2018-06-10 02:48:47--  https://deb.debian.org/debian/
Resolving deb.debian.org (deb.debian.org)... 128.31.0.62, 5.153.231.4,
149.20.4.15, ...
Connecting to deb.debian.org (deb.debian.org)|128.31.0.62|:443... connected.
ERROR: The certificate of ‘deb.debian.org’ is not trusted.
ERROR: The certificate of ‘deb.debian.org’ is not yet activated.
The certificate has not yet been activated

                                     <Ok>

────────────────────────────────────────────────────────────────────────────────

```

I tried editing the apt sources.list to use the http instead of https mirrors but no success.  Internet connection is Ok and I can run apt-get update / upgrade / install etc normally.

The problem only happens within dietpi apps.

Fourdee, 2018-08-18:
@carlosedp 

Hi,

Thanks for the report 👍 

We've had a few reports of this, i'll try to replicate.

NTP was replaced with systemd-timesyncd system in v6.12 I believe. Once updated, this should provide more stability of timesync.

Fourdee, 2018-08-20:
Ok, i was unable to replicate this issue on my network. However, I believe this is the same issue users have experienced with our previous NTP system (now replaced with systemd-timesyncd on recent images).

In short, until we are able to update our images to v6.12> (to include the above change by default):
**If you experience the above error, manually set the date:**
```
#YYYY MM DD
date +%Y%m%d -s "20180820"
```

Then resume 1st run setup and update:
````
dietpi-software
```

Fourdee, 2018-08-20:
I'll mark this as closed in favor of:
https://github.com/Fourdee/DietPi/issues/2026

closed 2018-08-20 by Fourdee

adamotte, 2018-08-24:
Reproduced with dietPI v6.14

ahmedmoselhi, 2020-08-15:
> BTW, updating the date to the correct one manually solved the problem. Apparently something wrong with NTP. After reboot the problem returned.

this solves a similar issue with me. thanks