MichaIng/DietPi #2021: Security Hole: Plaintext Global password saves to dietpi.txt

christophernewman, 2018-08-18:
## Creating a bug report/issue:

#### Required Information:
- DietPi version	| 6.13
- Distro version	| 9.4
- Kernel version	| Linux DietPi 4.14.62-v7+ #1134 SMP Tue Aug 14 17:10:10 BST 2018 armv7l GNU/Linux            
- SBC device		| RPi 3 Model B+
- Power supply used | 5V 3A
- SDcard used		|  SanDisk ultra

#### Steps to reproduce:
Simply nano or cat /boot/dietpi.txt

#### Expected behaviour:
global password should be encrypted

#### Actual behaviour:
global password shows in plain text

#### Extra details:
This opens up a boot vulnerability allowing data to be parsed from plain text

Fourdee, 2018-08-18:
@christophernewman 

Thanks for the request. 👍 

Generally speaking, to obtain the global password which is used for software installations only, user would need full SSH access to the system via the root/dietpi user password which is prompted to be changed during 1st run.
Outside of that, access to the physical flash media would be required to obtain this information.

Regardless, we'll look into encrypting the password and moving it to ```/var/lib/dietpi/dietpi-software/.X.bin``` with root access only and 700.

Fourdee, 2018-08-18:
Testing:
- 🈯️ Patch testing ```dietpi-update -1```
- Automated install testing on fresh image

Fourdee, 2018-08-19:
@christophernewman 

Resolved for v6.14 update (ETA next week):
https://github.com/Fourdee/DietPi/issues/2021

Passed testing, marking as completed.

closed 2018-08-19 by Fourdee

MichaIng, 2018-08-19:
@Fourdee 
Quick cool 😎.

I mean yeah, the global password was really just intended as a first access default for new software. This should then always be changed by user individually for each software, same as login passwords etc. But feels good, if even this first access default is secured.

But to finish it consequently, there need to be done some additional steps:
- We have to go through the installs where the password is used and check how it is stored.
- It is up to user to change the password of e.g. web UIs etc., but e.g. MariaDB user passwords are mostly saved as plain text within software config files (eg. /var/www/nextcloud/config/config.php) and cannot be changed by user via web UI, and we cannot expect user to take care each MariaDB user password, adjust it within software settings etc.
- Thus instead of using the global PW here (thus expose it again in most cases), I suggest we follow the way how Nextcloud does this: Silently generating a random database password on software install, apply it and directly unset the variable. The password is never needed by end user. In case root (and only "root", since unix_socket authentication) can access all databases, change db user passwords etc.
- It can be done similar in all cases, where a password is needed for non-interactive software access. Only for interactive user login, global PW makes totally sense.