MichaIng/DietPi #2023: IPv6 not working

antiabed, 2018-08-19:
## Creating a bug report/issue:

#### Required Information:
- DietPi version	| 6.13
- Distro version	| 9.4
- Kernel version	| 4.14.52-v7+ #1123 SMP Wed Jun 27 17:35:49 BST 2018 armv7l GNU/Linux
- SBC device		| RPi 3 Model B+ (armv7l)
- Power supply used	| 5V Power supply by raspberrypi.org
- SDcard used		| SanDisk ultra

#### Problem:
1. ping google.com gets a normal reply
2. ping6 google.com gets "connect: Network is unreachable"

I run Pi-hole on my DietPi install and I used CloudFlare as an upstream DNS server before. With this setup I could reach the ipv6 internet on all my devices, the internet.nl test gave me a 100% score.
Recently I switched to unbound, running my own DNS server on the Pi. I discovered the internet.nl test giving me a full score, with the exception of "IPv6 connectivity of DNS resolver" I tried everything to fix this, and in the process I found out that I can ping google.com and other websites from my Pi, but I can't do the same with ping6.
So I think unbound might not support IPv6 because the Pi it runs on can't connect to IPv6. Would this be a correct assumption and how do I get my DietPi install to support IPv6?

Fourdee, 2018-08-19:
@antiabed 

Hi,

IPv6 is enabled on all DietPi systems by default:
```
root@DietPi:~# ip a
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether 9e:8c:1d:da:87:f1 brd ff:ff:ff:ff:ff:ff
    inet 192.168.0.121/24 brd 192.168.0.255 scope global eth0
       valid_lft forever preferred_lft forever
    inet6 fe80::9c8c:1dff:feda:87f1/64 scope link
       valid_lft forever preferred_lft forever
```

To allow for a working IPv6, you need to ensure a IPv6 compatible gateway is used.

antiabed, 2018-08-19:
Hi Dan, thank you for the quick reply. I am no programmer by any stretch and I'm pretty new to this, so how would I go about using an IPv6 compatible gateway? And do you think this might be the source of my unbound DNS not being able to resolve IPv6?

Thanks again!

antiabed, 2018-08-21:
Running ip a and ls -lha /proc/net/if_inet6 I get the exact same response you posted. How would I go about using a IPv6 compatible gateway?

Fourdee, 2018-08-21:
> How would I go about using a IPv6 compatible gateway?

To be honest, I disable IPv6 for all my networks. So probably the wrong person to ask lol.

If you have a Windows PC, try testing the IPv6 ping there aswell, if this also fails, indicates your router/ISP is not fully IPv6 compatible:
```
C:\Users\4D>ping -6 ipv6.google.com
Ping request could not find host ipv6.google.com. Please check the name and try again.
```

I enabled IPv6 on my router, seems unable to find an IP, so I can only assume Virgin Media has not rolled IPv6 out yet. Regardless, check your router settings page for something similar:
[image]

antiabed, 2018-08-21:
Thanks again for the reply. On my MacBook on the same network all the ping6 requests work. I tried everything but ping6 on the dietpi keeps getting "connect: Network is unreachable" even pinging the router. Normal ping works.

antiabed, 2018-08-22:
After a lot of digging and messing around I finally found the issue. I had uncommented the `net.ipv6.conf.all.forwarding=1`line in `/etc/sysctl.conf` earlier, to enable IPv6 for my VPN. Because of that my DietPi couldn't do IPv6 itself. I probably didn't configure that right, and should have found a way to forward IPv6 and keep it working on the device itself but for now I've just commented the line.

MichaIng, 2018-10-14:
@antiabed 
I mark this issue as solved. Please feel free to reopen if we can/should do something when IPv6 is enabled via `dietpi-config`. Note that we handle IPv6 toggle differently now, to keep it enabled in kernel and just disable via `sysctl.conf` for network interfaces: https://github.com/Fourdee/DietPi/pull/2099

closed 2018-10-14 by MichaIng