MichaIng/DietPi #2024: bash: man: command not found

msdos, 2018-08-19:
## Creating a bug report/issue:

#### Required Information:
- DietPi version	| 6.13
- Distro version	| 9.4
- Kernel version	| Linux DietPi 4.14.62-v7+ #1134 SMP Tue Aug 14 17:10:10 BST 2018 armv7l GNU/Linux
- SBC device		| RPi 3 Model B (armv7l)
- Power supply used	| 5V 2A
- SDcard used		| SanDisk SDHC 16GB

#### Additional Information (if applicable):
- Was the software title installed freshly or updated/migrated?
Freshly installed image.

#### Steps to reproduce:

1. Install DietPi.
2. Try to ran `man`.

#### Expected behaviour:

- Terminal showing `What manual page do you want?` or something like that.

#### Actual behaviour:

- `bash: man: command not found`

#### Extra details:

```
root@DietPi:/opt# whereis man
man: /usr/local/man /usr/share/man
root@DietPi:/opt# echo $PATH
/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
root@DietPi:/opt#
```

Just another opinion... shouldn't we add a `dietpi-requiredinformation` or something like that creates the required information in the beginning of this issue instead of copying and pasting 4 commands?

msdos, 2018-08-19:
My bad. This issue can be closed. `man` needs to be installed using `apt`.

> Just another opinion... shouldn't we add a dietpi-requiredinformation or something like that creates the required information in the beginning of this issue instead of copying and pasting 4 commands?

But I'll keep this suggestion. And if someone searches for this error using DietPi in the future is still useful.

closed 2018-08-19 by msdos

Fourdee, 2018-08-20:
@msdos 

Hi,

Yes, we remove Manpage from the system during PREP to reduce the size of the image.

Must manually be installed with:
```
G_AGI man
```