MichaIng/DietPi #2026: Images | Update all images to resolve issues with previous NTP system

Fourdee, 2018-08-20:
As per:
https://github.com/Fourdee/DietPi/issues/2020

Some users are experiencing this during 1st run, where NTP system fails.

This has been replaced with systemd-timesyncd, which appears more stable, however, all v6.11 or lower images will need updating to have this system by default.

Fourdee, 2018-08-20:
Completed:
- DietPi_v6.13_RPi-ARMv6-Stretch.img
- DietPi_v6.13_SparkySBC-ARMv7-Stretch.img
- All VM's @MichaIng thanks 👍 

Pending:
- DietPi_v6.0_NanoPiM1-ARMv7-Stretch.img
- DietPi_v6.0_NanoPiM1Plus-ARMv7-Stretch.img
- DietPi_v6.0_NanoPiM2-ARMv7-Stretch.img
- DietPi_v6.0_NanoPiM3-ARMv7-Stretch.img
- DietPi_v6.0_NanoPiNEO2-ARMv8-Stretch.img
- DietPi_v6.0_OdroidC1-ARMv7-Jessie.img
- DietPi_v6.0_OdroidC2-ARMv8-Jessie.img
- DietPi_v6.0_OdroidXU4-ARMv7-Jessie.img
- DietPi_v6.0_RPi-ARMv6-Jessie.img
- DietPi_v6.11_NanoPiM3-ARMv7-Stretch.img
- DietPi_v6.1_PineA64-ARMv8-Stretch.img
- DietPi_v6.2_NanoPiNEO-ARMv7-Stretch.img
- DietPi_v6.2_NativePC-BIOS-x86_64-Stretch.img
- DietPi_v6.4_OdroidN1-ARMv8-Stretch.img
- DietPi_v6.4_Rock64-ARMv8-Stretch.img
- DietPi_v6.6_RPi-ARMv6-Stretch_AmiBerry.img
- DietPi_v6.7_ASUSTB-ARMv7-Stretch.img
- DietPi_v6.7_NanoPiK1Plus-ARMv8-Stretch.img
- DietPi_v6.7_OdroidC2-ARMv8-Stretch.img
- DietPi_v6.7_OdroidXU4-ARMv7-Stretch.img
- DietPi_v6.7_Rock64-ARMv8-Stretch.img
- DietPi_v6.9_ASUSTB-ARMv7-Stretch.img

Fourdee, 2018-08-20:
Rock64:
- Kernel upgrade test | https://github.com/Fourdee/DietPi/issues/1892
- CPU freq check | https://github.com/Fourdee/DietPi/issues/2025

Fourdee, 2018-08-20:
NanoPi M1/M1+:
- Use existing RootFS, upgrade kernel to 4.x from official image (which runs jessie)

Fourdee, 2018-08-20:
### XU4:
- Hangs during post
7 urandom warnings missed due to ratelimiting.

Fourdee, 2018-08-20:
@MichaIng 

> Maybe something with run_ntpd being called from old dietpi-update but with already new scripts in place. Have to investigate. Hope we can also find a solution for pre v6.9 users, perhaps with a pre-patch that is done before even dietpi-update goes too far, putting new script versions on place.

Yep, it needs resolving. If i can replicate it during image updates, i'll start debugging.

> 6.9

Yeah, but, may as well do the lot while we are here 👍

Fourdee, 2018-08-20:
Simple fix for the date issue?
- on VPS echo date > file
- EMS patch grab file and set date

Fourdee, 2018-08-20:
@MichaIng 

🈯️ Time issue resolved with: https://github.com/Fourdee/DietPi/commit/7635d6ccc9383751fe1bbe8d0b87b7a085a85b76

Cause:
- Removing NTP and not updating SystemD-TimeSyncD afterwards
- Date would be rolled back after reboot
- Sync would then fail as systemd needing daemon-reloading (probably due to NTP removal)

Fourdee, 2018-08-21:
### Rock64:
- 1st run is fine
- Reboot results in no post, serial debug required.

Fourdee, 2018-08-21:
Note to self:
https://github.com/midwan/amiberry/issues/264#issuecomment-414215685

Fourdee, 2018-08-22:
As the original issue which caused this ticket to be created, is now resolved, and, the most popular images have been updated, i'll flag this as low priority.

adamotte, 2018-08-24:
Same error with 6.14  (#2020) 

#### Details:
- Date           | Thu  3 Nov 17:17:21 GMT 2016
- Bug report     | N/A
- DietPi version | v6.14
- Img creator    | DietPi Core Team
- Pre-image      | TinkerOS 2.0.5
- SBC device     | Asus Tinker Board (armv7l) (index=52)
- Kernel version | #1 SMP Thu Feb 22 18:23:31 CST 2018
- Distro         | stretch (index=4)
- Command        | Connection test: https://deb.debian.org/debian/
- Exit code      | 5
- Software title | DietPi-Software

#### logs:
```
Log file contents:
Spider mode enabled. Check if remote file exists.
--2016-11-03 17:17:18--  https://deb.debian.org/debian/
Resolving deb.debian.org (deb.debian.org)... 2001:4f8:1:c::15, 2001:67c:2564:a119::148:14, 2001:41c8:1000:21::21:4, ...
Connecting to deb.debian.org (deb.debian.org)|2001:4f8:1:c::15|:443... connected.
ERROR: The certificate of ‘deb.debian.org’ is not trusted.
ERROR: The certificate of ‘deb.debian.org’ is not yet activated.
The certificate has not yet been activated
```

Millichrome, 2018-09-05:
In [Issue 2020](https://github.com/Fourdee/DietPi/issues/2020) you mention that it's possible to manually add the date (I am trying to install the 6.9 image on an ASUS tinker board). What file do I have to edit to manually set the date on the SD card?

Fourdee, 2018-09-06:
@Millichrome 

> What file do I have to edit to manually set the date on the SD card?

You can set the date in file ```/etc/fake-hwclock.data``` I believe (untested).

Failing that, if NTP sync fails for you, and you've tried various NTP mirrors in ```dietpi-config```, you'll need to manually apply a date eg:
```
date +%Y%m%d -s "20180906"
```

Then re-run the initial setup with:
```
dietpi-software
```

Millichrome, 2018-09-06:
I log into my Tinker Board DietPi install via SSH. On first connection I get a request to update the default password and run the the update program. The update routine fails with the following error: 

https://i.imgur.com/WkcYKSo.jpg

I can't run any commands or do anything once I get this error (at least via SSH, I haven't tried a physical connection). 

On Windows 10, I don't see any way to access the "etc" folder (linux partition), do I need to install a flavour of desktop linux to edit "/etc/fake-hwclock.data?"

Thanks for the help.

MichaIng, 2018-09-06:
@Millichrome 
Thanks for the report.

Can't you just select the <ok> button (use <tab> on keyboard to reach that one) and can run terminal commands then?

Issue is that timesync did not yet update the system clock. You should be able to run
`systemctl start systemd-timesyncd`
as another failsafe, as well run: `apt install -y fake-hwclock`
and then run `dietpi-software` to rerun first run setup (incl. update etc.).

Fourdee, 2018-09-07:
I'll update the asus TB image, will resolve this issue. Due to ```wget``` without ```-no-check-certificates``` prior to ntpd sync check.

Fourdee, 2018-09-07:
RPI ref: https://github.com/Fourdee/DietPi/issues/2050

Fourdee, 2018-09-07:
🈯️ ASUS TB and RPi image redone

Fourdee, 2018-09-07:
@Millichrome 

ASUS TB image updated, should resolve the issue you are experiencing.

Please download:
https://dietpi.com/downloads/images/DietPi_ASUSTB-ARMv7-Stretch.7z

Try this image and let us know if problems are resolved.

Millichrome, 2018-09-07:
@MichaIng

This might be an SSH issue or something with my system setup, but I can't run any commands after I get the error. That's why I was trying edit the date on the old 6.9 TB image. 

@Fourdee 

Awesome! I will test the new image and report back in a few hours.

Fourdee, 2018-09-08:
@Millichrome 

>  Hopefully the network/disk IO is much better than with the B+. :)

Its in a different league :+1:

Fourdee, 2018-09-18:
Note to self:
- VB images still need updating

Fourdee, 2018-10-16:
Only outstanding images to do are Native PC, however, as 99% of those have HW clocks, this issue should not effect those installs.

As such i'll set to low priority for now.

MichaIng, 2018-10-16:
@Fourdee
How do you create native PC images? I have an old unused i386 and could do that. Anyway had the plan do use it for DietPi dev, also since it has WiFi of course.

But not sure how critical the used base system is and/or if some manual adjustments are needed to ensure/increase compatibility?

Fourdee, 2018-10-18:
> How do you create native PC images? 

For BIOS/CSM:
- 1 USB stick with Debian installer
- 1 USB stick for installing to
- Make sure ```fstab``` uses the PARTUUID, eg:
```
PARTUUID=fc9c45cd-01    /    auto    defaults,noatime    0 0
```

Then read the USB stick to an image.

---

UEFI:
- Install Debian onto Z83
- Use clonezilla to save EMMC to image

MichaIng, 2018-10-18:
@Fourdee 
> - Remove all HDD's from system (ensures GRUB doesnt pick them up)
> - After install, remove Debian USB stick
> - Make sure fstab uses the PARTUUID, eg:

Does initramfs (from final image/system) use the fstab entries from the partition/drive it is loaded from? Then of course editing fstab would resolve it. However since I could not find a way to edit the fstab from within initramfs (could not mount, since there is no "file system" to mount to), this needs to be done from another system. Anyway needed to run zerofree e.g.

If the root partition is not read from fstab, but somehow stored within the initrd/grub, then I indeed needed to detach the Debian installer USB stick, which was otherwise recognized as /dev/sda and the destination drive as /dev/sdb. Perhaps this also depends on the IDE/SATA controllers, in which order they recognize the drives.

Fourdee, 2018-10-18:
@MichaIng 

> Does initramfs (from final image/system) use the fstab entries from the partition/drive it is loaded from? 

Actually, ignore everything I just said, UUID will work fine, that is what ```update-grub``` will use.

Can check what gets applied with:
```/boot/grub/grub.cfg```

```
menuentry 'Debian GNU/Linux' --class debian --class gnu-linux --class gnu --class os $menuentry_id_option 'gnulinux-simple-aa010bdf-3471-4ca4-ab$
        load_video
        insmod gzio
        if [ x$grub_platform = xxen ]; then insmod xzio; insmod lzopio; fi
        insmod part_gpt
        insmod ext2
        if [ x$feature_platform_search_hint = xy ]; then
          search --no-floppy --fs-uuid --set=root  aa010bdf-3471-4ca4-ab17-d1488d382949
        else
          search --no-floppy --fs-uuid --set=root aa010bdf-3471-4ca4-ab17-d1488d382949
        fi
        echo    'Loading Linux 4.15.0-0.bpo.2-amd64 ...'
        linux   /boot/vmlinuz-4.15.0-0.bpo.2-amd64 root=UUID=aa010bdf-3471-4ca4-ab17-d1488d382949 ro net.ifnames=0 consoleblank=0 quiet
        echo    'Loading initial ramdisk ...'
        initrd  /boot/initrd.img-4.15.0-0.bpo.2-amd64
}
```

MichaIng, 2019-01-27:
@Fourdee 
I think the RPi image needs to be redone. It's on DietPi v6.17 where I added a bug to exit, if hardware identifier file could not be found, which is meant to not exist on RPi: https://github.com/Fourdee/DietPi/commit/eae7af9d50775d833c2c672a344f364df1b02ae5#diff-c7bcfe6e7ece17b798b4d08a37df308cR37

You fixed that with v6.18: https://github.com/Fourdee/DietPi/commit/0c0995312500865d049c13276032fb94bfb43f89#diff-c7bcfe6e7ece17b798b4d08a37df308c

Most likely related: https://dietpi.com/phpbb/viewtopic.php?f=11&t=5463&p=16292#p16292
I am wondering that we didn't get more reports 😅.

Fourdee, 2019-01-28:
@MichaIng 

If I remember and can, i'll redo the image after v6.20 release.

Heres the finalisation script:
```
#!/bin/bash
{

	#check packages are installed
	if (( ! $(dpkg -l | grep -ci -m1 'gparted') )); then

		apt-get install gparted gdisk gpart zerofree -y

	fi

	#-------------------------------------------------------------------------------
	#MUST LEAVE 50MB+ space or automation autologin may fail due to 0 free space!!!
	#-------------------------------------------------------------------------------

	RK_SINGLE_IMG=0

	CP_PARTITIONS=0

	#fdisk -l -u "$IMAGE_FP/$IMAGE_NAME"

	IMAGE_FP='/mnt/dietpi_userdata/#Backups/_Daniel/Projects/DietPi'

	IMAGE_NAME='DietPi_v6.19_RockPro64-ARMv8-Stretch'
	IMAGE_NAME+='.img'

	INDEX_BOOT_PART=1
	INDEX_ROOTFS_PART=2
	if (( $RK_SINGLE_IMG )); then

		INDEX_ROOTFS_PART=1

	fi

#ROCK 64 after reading the image AND again after shrinking
#to fix:
#	GPT PMBR size mismatch (4458495 != 15523839) will be corrected by w(rite).
#	15523806
#RUN
# gdisk "$IMAGE_FP/$IMAGE_NAME"
#	w | y
#

	if [[ ! -f $IMAGE_FP'/'$IMAGE_NAME ]]; then

		read -p "No file found..."
		exit

	fi

	#resize2fs /dev/loop2p7 500M
	#dumpe2fs -h /dev/loop2p7 | grep 'Block count'
	#Block count * Block size / 1024 / 1024 =MB?

	modprobe loop
	losetup -f
	losetup /dev/loop2 "$IMAGE_FP/$IMAGE_NAME"
	partprobe /dev/loop2

	if (( $RK_SINGLE_IMG )); then

		e2fsck -f /dev/loop2

	else

		if [[ $INDEX_BOOT_PART ]]; then

			e2fsck -f /dev/loop2p$INDEX_BOOT_PART

		fi
		e2fsck -f /dev/loop2p$INDEX_ROOTFS_PART

	fi

	#Optional, copy
	if (( $CP_PARTITIONS )); then

		mkdir -p /mnt/loop2p1
		mkdir -p /mnt/loop2p2
		mount /dev/loop2p$INDEX_BOOT_PART /mnt/loop2p1
		mount /dev/loop2p$INDEX_ROOTFS_PART /mnt/loop2p2

		echo -e "\ncopying data to $HOME\n"
		rm -R $HOME/loop2p1
		rm -R $HOME/loop2p2

		cp -R /mnt/loop2p1 $HOME/
		cp -R /mnt/loop2p2 $HOME/
		rm -R $HOME/loop2p1/lost+found
		rm -R $HOME/loop2p2/lost+found

	fi

	sync

	umount /mnt/loop2p1
	umount /mnt/loop2p2
	umount /mnt/loop2

	#Resize partitions
	echo -e "\Resize partitions\n"
	gparted /dev/loop2

	sync

	#Mount for file changes, if required
	mkdir -p /mnt/loop2p1
	mkdir -p /mnt/loop2p2
	if (( $RK_SINGLE_IMG )); then

		mount /dev/loop2 /mnt/loop2p1

	else

		if [[ $INDEX_BOOT_PART ]]; then

			mount /dev/loop2p$INDEX_BOOT_PART /mnt/loop2p1

		fi
		mount /dev/loop2p$INDEX_ROOTFS_PART /mnt/loop2p2

	fi

	if (( $CP_PARTITIONS )); then

		echo -e "\nCopying data back\n"
		cp -R $HOME/loop2p1/* /mnt/loop2p1/
		cp -R $HOME/loop2p2/* /mnt/loop2p2/

	fi

	read -p "Partitions mounted to /mnt/loop2px, make changes to files if required..."

	sync

	umount /mnt/loop2p1
	umount /mnt/loop2p2

	#Resize partitions
	# echo -e "Resize partitions"
	# gparted /dev/loop2

	# sync

	read -p "If failed press CTRL+C to exit and prevent further action, else, press any key to continue..."

	if (( $RK_SINGLE_IMG )); then

		zerofree -v /dev/loop2

	else

		for (( i=0; i<10; i++ ))
		do

			if [[ -b /dev/loop2p$i ]]; then

				zerofree -v /dev/loop2p$i

			fi

		done

	fi

	losetup -d /dev/loop2

	if (( $RK_SINGLE_IMG )); then

		END_PARTITION='850M'

	else

		# - debug
		# END_PARTITION="$(echo p | parted "$IMAGE_FP/$IMAGE_NAME" | grep ' 1 ' | awk '{print $3}')"

		if (( $INDEX_ROOTFS_PART == 1 )); then

			END_PARTITION=$(( $(fdisk -l "$IMAGE_FP/$IMAGE_NAME" | grep ".img$INDEX_ROOTFS_PART" | awk '{print $4}') + 1 ))

		else

			END_PARTITION=$(( $(fdisk -l "$IMAGE_FP/$IMAGE_NAME" | grep ".img$INDEX_ROOTFS_PART" | awk '{print $3}') + 1 ))

		fi

		END_PARTITION=$(( 512 * ( $END_PARTITION + 256 ) )) #64 bytes for secondary GPT + safety net

	fi

	truncate --size=$END_PARTITION "$IMAGE_FP/$IMAGE_NAME"

	echo -e "\nCleaning up $HOME/loop2px\n"
	rm -R $HOME/loop2p1
	rm -R $HOME/loop2p2

	read -p "Done, press any key to continue..."

	exit

}
```

MichaIng, 2019-02-08:
@Fourdee 
May I add a Odroid C1 Stretch image?
More and more issues appear with Jessie not offering required library/package versions, e.g. now Emby: https://github.com/Fourdee/DietPi/issues/2521

Btw great work to update so much images to v6.20, I recognized meanwhile. That as well prevented many users from running into first run update loop due to the updater issue 👍!!

Fourdee, 2019-02-08:
@MichaIng 

> May I add a Odroid C1 Stretch image?

Please, I still unable to find mine, most likely in a toy box somewhere (kids lol).

> * I know this means a distro upgrade since Meveric does not offer a Stretch image. However if this can succeed, then via DietPi-PREP

Yep, agree Stretch is preferred. However, software from Meverics APT repo (kernel/kodi etc) may not work.

Although, if above is an issue, update to latest kernel, then remove his APT repo from ```/etc/apt/sources.list.d```

MichaIng, 2019-02-08:
@Fourdee 
Ah sorry, I have not Odroid C1 here either 😢. I meant adding it to the ToDo list 😉.

Hmm, kernel packages should work independently from Debian version, at least on RPi this was never an issue (Stretch kernel on Buster, and that time Jessie kernel on Stretch). But of course needs testing.
Are Meverics other repos device specific and for C1 only Jessie available? 🤔 Okay that would be indeed an issue. I mean we can anyway not support it forever, but loosing Kodi and certain media software support is quite a nasty bumper...

MichaIng, 2019-03-08:
Other outdated images:
- [ ] NanoPi M1+
- [ ] RPi AmiBerry
- [ ] x86_64 BIOS
- [ ] x86_64 UEFI: https://github.com/MichaIng/DietPi/issues/1171

I will do the x86_64 BIOS image the next days, UEFI is currently outside of my possibilities.

MichaIng, 2019-03-24:
#### Updated VirtualBox Stretch image: https://dietpi.com/downloads/images/DietPi_VirtualBox-x86_64-Stretch.7z

MichaIng, 2019-04-01:
#### Updated VirtualBox Stretch image: https://dietpi.com/downloads/images/DietPi_WMware-x86_64-Stretch.7z
#### Updated VirtualBox Buster image: https://dietpi.com/downloads/images/DietPi_WMware-x86_64-Buster.7z

Links on homepage have been updated as well (manually).

MichaIng, 2019-04-06:
Update NativePC BIOS image ready for testing: https://dietpi.com/downloads/testing/DietPi_NativePC-BIOS-x86_64-Stretch.7z

MichaIng, 2019-04-13:
#### Update NativePC BIOS image: https://dietpi.com/downloads/images/DietPi_NativePC-BIOS-x86_64-Stretch.7z

MichaIng, 2019-05-14:
#### Updated RPI image ready for testing: https://dietpi.com/downloads/testing/DietPi_RPi-ARMv6-Stretch.7z

MichaIng, 2019-05-18:
Updated RPi image with yesterday released kernel v4.19 + **FIRST Raspbian Buster image available for testing**: https://dietpi.com/downloads/testing/

MichaIng, 2019-07-10:
I close this in favour of: https://github.com/MichaIng/DietPi/issues/2979

closed 2019-07-10 by MichaIng