MichaIng/DietPi #2030: Bug: pihole continues while dhcp get stopped

ichbestimmtnicht, 2018-08-22:
## Creating a bug report/issue:

#### Required Information:
- DietPi version	14
- Distro version	stretch
- Kernel version	Linux server-elis 4.14.62-v7+ #1134 SMP Tue Aug 14 17:10:10 BST 2018 armv7l GNU/Linux

- SBC device		RPi 3 Model B (armv7l)
- Power supply used	5V 3A power)
- SDcard used		 SanDisk ultra class 10

#### Additional Information (if applicable):
- Software title	dietpi-backup
- Was the software title installed freshly or updated/migrated? first use 1 week old install
- Can this issue be replicated on a fresh installation of DietPi? i think so
- `dietpi-bugreport` dont have one because it does not create one.

#### Steps to reproduce:

1. install dietpi with pihole and isc-dhcp-server (and maybe additional software)
2. set it up with dietpi Hotspot
3. run `sudo dietpi-backup` -> create -> starting 

#### Expected behaviour:

- backup shouldn't drop internet connection, pihole and dhcp server are up and running during backup

#### Actual behaviour:

- doesn't stops pihole while backing up but dhcp server goes down

#### Extra details:
- did not checked earlier issues about it

MichaIng, 2018-08-23:
@ichbestimmtnicht 
Thanks for your request.

So far, I think the current behavior is expected:
- During backup and other maintenance tasks, we handle/stop as much services as possible to reduce interferences etc.
- We keep all services active that are required for successful network connectivity, since during other maintenance tasks (software installs, update etc.) network is required.
- `pihole-FTL` needs to stay active, as it is the DNS resolver not only for external clients (using Pi-hole), but also for the host machine. Stopping it would break name resolving, thus download of software, updates etc.
- The DHCP server on the other hand is not required for the host machine, just for external DHCP clients, AFAIK?
- I am anyway wondering, if there is not a DHCP server included into pihole-FTL? At least the since FTLDNS included dnsmasq can serve as DHCP server and I remember that you can enable DHCP functionality via pihole command, can't you? In this case you actually do not need `isc-dhcp-server` in combination with Pi-hole.

ichbestimmtnicht, 2018-08-24:
I understand and like the minimal idea. Thats' why I use Dietpi and you have a seamless argumentation :100: 

Then let's say it would be neat to have stop options for networking services during dietpi maintenance modes. As admin you can setup HAProxy to avoid downtimes. With Dietpie's moore frequent use in personal setup's (i suggest) and the within interconnected limitation of hardware ressources shutting down network switches like dhcpd and haproxy at the same time seems negoitable to me.
While a static 2nd or 3rd dns entry (at the router or device) could solve piholes absence for (kn)own devices during maintenance.

closed 2018-08-24 by ichbestimmtnicht