MichaIng/DietPi #2033: Suspicious entries in 6.14 update

Barishhh, 2018-08-23:
Can someone explain me what is going here? During the upgrade to 6.14, an SSH key gets added into the system and the global password has been changed aswell. I don't trust it, so I changed both entries back. Can someone tell me what those entries are needed for and if the system is safe to use?

[image]

adamotte, 2018-08-23:
Im' not an expert but : 

- `182.101.93.93` point to `diepit.com`
- v6.14 introduce encrypted password, see [CHANGELOG.txt](https://github.com/Fourdee/DietPi/blob/master/CHANGELOG.txt)

Barishhh, 2018-08-23:
That explains it, thanks!

closed 2018-08-23 by Barishhh

MichaIng, 2018-08-23:
@Barishhh 
Jep just to clarify from our side:

SSH host key:
- Entries in `/root/.ssh/known_hosts` contain SSH host keys, so when connecting via SSH, your system can check/prove, that the answering server really is the one that you expect, by comparing the received host key with the one saved in this file-
- `DietPi-Survey`, `DietPi-Bugreport` and since v6.14 `DietPi-Benchmark` connect via SFTP to our server to upload their data/results, if chosen/allowed by you. SFTP basically is an FTP protocol wrapped within an secure/encrypted SSH connection, which is the reason SSH keys are exchanged.
- Already before v6.14 `DietPi-Survey` and `DietPi-Bugreport` used the same method/connection, but via `dietpi.com` as server address, which points to the same IP.
- However since we added the DietPi server to Cloudflare, there seemed to be issues in some cases, where connecting via SFTP to domain instead of IP failed. Jep, can also replicate it here on `v6.13` 🤔: https://support.cloudflare.com/hc/en-us/articles/200169346-Using-FTP-with-Cloudflare-. Thus we changed upload target to the raw IP to bypass the Cloudflare protection mention in the link and as the host key is really just accepted for the domain, we needed to re-add it for the raw IP as well.

Password:
- Jep we changed the behavior of the default global password to be saved encrypted instead of plain within dietpi.txt: https://github.com/Fourdee/DietPi/issues/2021
- This was no critical security issue, as this plain text password was only used as initial default and on first boot users are asked to change their login passwords. Also after accessing first to software web UIs etc. I hope that it's clear for everyone that (especially when opened to the web) the password of the related software should/must always be changed individually, to not match the plain text default anymore 😉. But having even this now encrypted is another shovel of security on top 😃.

Yeah overall we still very actively develop DietPi down to it's core, trying to make it better and better, but this involves certain changes on every update, that we don't want to hide/do silently, so you can see what happens and react to this, ask questions, as done with this thread. I hope we do not bother you guys too much with this 😆.

Barishhh, 2018-08-23:
Thanks alot for the clarification @MichaIng! 

I was just worried. How many times did we not see that software is getting backdoored, so yeah i'd rather make sure it's something legit and check it with you guys. 

Thanks again and keep up the dev work!