MichaIng/DietPi #2034: SSH | Error when SSH client passes unsupported terminal to DietPi SSH server

MichaIng, 2018-08-23:
## Creating a bug report/issue:

#### Additional Information (if applicable):
- Software title	| any SSH server

#### Steps to reproduce:

1. Connect from a remote SSH client with an unsupported terminal type (e.g. `rxvt-256color`) to DietPi SSH server

#### Expected behaviour:

- No error, all software/function working fine

#### Actual behaviour:

- Software/Functions that need the terminal definitions, will fail, e.g. `tput: unknown terminal "rxvt-256color"`

#### Extra details:

- The issue is similar to the $LANG issue when accessing via SSH: https://github.com/Fourdee/DietPi/blob/master/rootfs/etc/bashrc.d/dietpi-posix-set-locale.sh
- Similar to $LANG, also $TERM is passed from the SSH client to the SSH server. This can even on OpenSSH not be suppressed: https://unix.stackexchange.com/a/67540
- If the passed terminal is not supported on the DietPi system, the above errors show up.

#### Solutions:
- Installing [ncurses-term](https://packages.debian.org/en/stretch/ncurses-term) to support more terminal types.
- Resetting $TERM to e.g. `xterm` or another terminal, that is supported on with `ncurses-base`.

#### Other references:
- https://unix.stackexchange.com/questions/67537/prevent-ssh-client-passing-term-environment-variable-to-server
- https://raspberrypi.stackexchange.com/questions/53364/trying-to-ssh-in-using-urxvt-gives-error
- https://unix.stackexchange.com/questions/196488/set-urxvt-to-work-with-256-colors

MichaIng, 2018-08-25:
Check for supported terminals: `toe -a`, e.g. `toe -a | grep -q "$TERM"`

As manually changing the $TERM variable, thus faking a different terminal that what actually is connected, is not the best long term practice:
- Show warning/G_WHIP_MSG that the connecting terminal is not supported, thus SSH client setting/term needs to be changed or `ncurses-term` installed.
- Tell user that xterm (respectively xterm-256color, if `[[ $TERM =~ 256 ]]`) is faked for the current session.

MichaIng, 2018-09-01:
PR is up for testing: https://github.com/Fourdee/DietPi/pull/2049

Fourdee, 2018-09-02:
Note to self:
- General Testing
- Test Allo GUI

Fourdee, 2018-09-02:
@MichaIng 

Yep seems fine, although, been unable to test with any "incompatible terminal" as i know of none 🤣 .

Fourdee, 2018-09-05:
@MichaIng 

> But since xterm seems to be the most common and most supported terminal definition and the issue is anyway a very rare one, it should be good enough and there is not more we can do.

Yep agree 👍 

Great work, I believe this is the best we can do for now and should prevent the initial issue on this ticket. If additional work is required, it could be a question of "is the SSH client" following the standards?

I'll mark as completed. Please reopen if required.

closed 2018-09-05 by Fourdee

Mossy93, 2022-08-02:
Hello,
I am having this error using `ssh` through Alacritty. I have `ncurses-term` installed on my DietPi system.