MichaIng/DietPi #2035: Debian repository does no longer have a Release file.

adamotte, 2018-08-24:
#### Details:
- Date           | Thu  3 Nov 17:17:08 GMT 2016
- Bug report     | N/A
- DietPi version | v6.14
- Img creator    | DietPi Core Team
- Pre-image      | TinkerOS 2.0.5
- SBC device     | Asus Tinker Board (armv7l) (index=52)
- Kernel version | #1 SMP Thu Feb 22 18:23:31 CST 2018
- Distro         | stretch (index=4)
- Command        | G_AGUP
- Exit code      | 100
- Software title | DietPi-Software

#### Steps to reproduce:
First start of a fresh image.

#### Expected behaviour:
Running startup/init/install scripts.

#### Actual behaviour:
Exit with error.

#### Additional logs:
```
Log file contents:
Hit:1 http://repo.linaro.org/ubuntu/linaro-overlay stretch InRelease
Ign:2 http://cdn-fastly.deb.debian.org/debian stretch InRelease
Hit:3 http://cdn-fastly.deb.debian.org/debian stretch-updates InRelease
Hit:4 http://cdn-fastly.deb.debian.org/debian stretch-backports InRelease
Hit:5 http://cdn-fastly.deb.debian.org/debian stretch Release
Ign:6 https://deb.debian.org/debian-security stretch/updates InRelease
Err:7 https://deb.debian.org/debian-security stretch/updates Release
  server certificate verification failed. CAfile: /etc/ssl/certs/ca-certificates.crt CRLfile: none
Reading package lists...
E: The repository 'https://deb.debian.org/debian-security stretch/updates Release' does no longer have a Release file.
```

Fourdee, 2018-08-24:
@adamotte 

Thanks for the report 👍 

Please can you paste results of:
```
cat /etc/apt/sources.list
```

It should match:
```
deb https://deb.debian.org/debian/ stretch main contrib non-free
deb https://deb.debian.org/debian/ stretch-updates main contrib non-free
deb https://deb.debian.org/debian-security/ stretch/updates main contrib non-free
deb https://deb.debian.org/debian/ stretch-backports main contrib non-free
```

Please also paste results of:
```
dpkg -l | grep debian-archive-keyring
```

MichaIng, 2018-08-27:
@adamotte @Fourdee 
> Date | Thu 3 Nov 17:17:08 GMT 2016

It's due to system clock not correctly synchronized. (Network connections appears to be fine.)

Hmm due to `fake-hwclock` removal? Should actually not be an issue as this just influences clock during boot until time sync. And `dietpi-software` separately does a time sync check.

`systemd-timesyncd` settings? Should be `[0-3].debian.pool.ntp.org` on older images and empty (resulting in local gateway, if passed via DHCP and `[0-3].debian.pool.ntp.org` otherwise) on new images.

@adamotte 
Please check the following and paste results to check whether fake-hwclock is correctly (un)installed, based on hwclock availability and systemd-timesyncd servers + status:
```
dpkg --get-selections | grep 'fake-hwclock'
hwclock
cat /etc/systemd/timesyncd.conf
systemctl start systemd-timesyncd
sleep 10
systemctl status systemd-timesyncd
systemctl stop systemd-timesyncd
```

adamotte, 2018-08-27:
@MichaIng You're right it was due to system clock. After manually adjusted it, and restarted dietpi-software, everything ran fine. 
FYI, i had modified dietpi.txt to add softwares to be installed directly on first boot.
Now my clock is perfectly synchronized so i guess timesyncd worked after the all boot/install but if you have a software list to install too, there is a gap of unsynchronized time.

Fourdee, 2018-08-27:
@MichaIng 

Yep, due to removal of ```fake-hwclock```I believe:
```
root@DietPi:~# hwclock
2018-08-27 20:44:37.840811+0100
root@DietPi:~# dpkg -l | grep hwclock
root@DietPi:~#
```

I'll set a patch to reinstall it for all systems, regardless of ```hwclock```. Maybe we can add an option in ```dietpi-config``` to allow users to select if ```hwclock``` is actually available?

Fourdee, 2018-08-27:
Marking as completed:

Planned option to enable RTC:
https://github.com/Fourdee/DietPi/issues/2041

closed 2018-08-27 by Fourdee

adamotte, 2018-08-27:
@Fourdee BTW, Great work, I love DietPi !

Fourdee, 2018-08-27:
@adamotte 

Thanks 👍 

Although its a team effort, don't forget @MichaIng 🥇 😃