MichaIng/DietPi #2041: DietPi-Config | Ability to toggle RTC installed

Fourdee, 2018-08-27:
RTC install = removal of ```fake-hwclock```

Fourdee, 2018-08-27:
Completed.

closed 2018-08-27 by Fourdee

MichaIng, 2018-08-27:
Nice 😃. Maybe we can add a check via `hwclock` if one is available, before allowing to toggle?

For the record on RPi:
```
2018-08-27 23:54:50 root@micha:/var/log# hwclock --verbose
hwclock from util-linux 2.32.1
System Time: 1535406941.759119
Trying to open: /dev/rtc0
Trying to open: /dev/rtc
Trying to open: /dev/misc/rtc
No usable clock interface found.
hwclock: Cannot access the Hardware Clock via any known method.
```
- In such cases there is definitely none available.

On Odroids XU4/HC1/2 have a "RTC backup battery connector", thus a RTC which is also powered, thus the above command succeeds:
- https://www.hardkernel.com/main/products/prdt_info.php?g_code=G150229074080&tab_idx=2
- https://www.hardkernel.com/main/products/prdt_info.php?g_code=G137508214939
> You need to just connect this backup battery to keep the time when your ODROID is offline. This battery can keep the RTC running up to 3 years without external power supply.

Thus even if no battery is attached, the RTC is available, which is in fact ineffective, since it cannot keep the clock without battery. Since I couldn't find any way to check RTC battery state, leaving `fake-hwclock` in place by default is the best solution. If one installs a real hwclock (e.g. [like this on RPi](https://blog.remibergsma.com/2013/05/08/adding-a-hardware-clock-rtc-to-the-raspberry-pi/)) attaches a proper battery etc, the toggle is fine.

Only on VirtualBox, if `hwclock` succeeds, we can be sure that `fake-hwclock` is not needed, since it does not need a battery there 😉.