MichaIng/DietPi #2043: DietPi-Software | Nextcloud dbpassword could not be found in config.php

LexiconCode, 2018-08-28:
I believe this is an issue with the gobal password manifested by installing nextcloud. Additional illegal characters or length limits may be necessary for global passwords

Password being used:

g%8cI#UqCfZwQLX**tcu*XjSf%S77!&GYNM7wRE8e5wL##GDRZbDds@tF@TK%LcvG77r9ZZAE&mZ^ahFDGRjxXit#5Y2aip9UmKWe@S3NtzJD8V4NGET73m3dt8YdjR6T

#### Required Information:
- DietPi version	| `paste -sd '.' /DietPi/dietpi/.version`
- Distro version	| stretch
- Kernel version	| Linux DietPi 4.14.66 armv7
- SBC device		| Odroid XU3/XU4/HC1/HC2 (armv7l)
- Power supply used	| Official oDroid powers
- SDcard used		| SanDisk ultra

#### Additional Information (if applicable):
- Software title	| Nextcloud
- Can this issue be replicated on a fresh installation of DietPi? Yes
- `dietpi-bugreport` dc5140fc-00f7-4313-b0e2-a963c8e3abc3

#### Extra details:
     │ The following info might help:                                                                                       │
     │  1723 root     pts/0     |       \_ /bin/bash /DietPi/dietpi/dietpi-launcher                                         │
     │  1757 root     pts/0     |           \_ /bin/bash /DietPi/dietpi/dietpi-software                                     │
     │  9609 root     pts/0     |               \_ whiptail --title DietPi-Software --msgbox [FAILED] Setting could not be  │
     │ added after desired line.                                                                                            │
     │  The pattern                                                                                                         │
     │ ?'dbpassword'                                                                                                        │
     │ could not be found in file                                                                                           │
     │ ?/var/www/nextcloud/config/config.php                                                                                │
     │  Please retry with valid parameter ($4) or apply the setting manually:                                               │
     │ ?'mysql.utf8mb4' => true, --ok-button Ok --backtitle DietPi-Software | Odroid XU3/XU4/HC1/HC2 (armv7l) | IP:         │
     │ 192.168.1.8 15 120                                                                                                   │
     │ 11067 root     pts/1             \_ /bin/bash /DietPi/dietpi/dietpi-launcher                                         │
     │ 11088 root     pts/1                 \_ /bin/bash /DietPi/dietpi/dietpi-launcher

MichaIng, 2018-08-28:
Thanks for your report.

I will investigate what issue might occur when using such a random string password.

LexiconCode, 2018-08-28:
On a semi-related note. What are the ramifications of changing the global password after installing x software? Does the software prior to the pw change get updated with the new global password?

MichaIng, 2018-08-28:
@LexiconCode 
Nope, we do not change any passwords after the software got installed. The global password is just meant to be an initial default password, so even that software gets installed automated, you know which password you need to enter on first access. But afterwards you should anyway change it individually for every software, especially those who are opened to the web and DietPi does not touch it anymore.

About your issue:
- I believe since `'dbpassword'` could not be found in your Nextclouds config.php, the occ install step must have been failed already. You you please check by pasting output of `cat /var/www/nextcloud/config/config.php`? **Please anonymize personal/domain and password/salt/etc fields!**
- From the password string, there are no characters inside that would make problems inside bash strings at least. Just `"` and `$` should be a problem here, AFAIK. But maybe Nextclouds occ command and/or MariaDB do not allow others and/or strings of that length. You can play around via:
```
mysql -e "grant all privileges on *.* to 'tmp_root'@'localhost' identified by '<yourPassword>' with grant option"
ncc maintenance:install --no-interaction --database "mysql" --database-name "nextcloud" --database-user "tmp_root" --database-pass "<yourPassword>" --admin-user "<yourNextcloudUsername>" --admin-pass "<yourPassword>" --data-dir "<yourDataDirectory>"
```
where `<yourNextcloudUsername>` and `<yourDataDirectory>` should match the ones defined in your dietpi.txt, by default user is `admin` and data dir is `/mnt/dietpi_userdata/nextcloud_data`.
After install worked, you can set the used password as new global password and rerun the install. It will then skip this occ/ncc install step. Ah before rerunning the install, remove the temporary database admin user that is needed for occ/php to access: `mysql -e "drop user 'tmp_root'@'localhost'"`
But I will also try to replicate this now with exactly the same password on VM.

LexiconCode, 2018-08-28:
This is a fresh install and I've tried a fresh install 3 times with the same results.

I'm at work at the moment and won't have access until tomorrow morning to investigate.

closed 2018-08-31 by LexiconCode

MichaIng, 2018-09-01:
@LexiconCode 
Were you able to resolve it?

LexiconCode, 2018-09-04:
It was indeed a failed install of nextcloud. After the fourth and fifth fresh install everything work as expected. Since I can't replicate it again I would consider closed.

MichaIng, 2018-09-04:
@LexiconCode
Okay, strange that it  needed several attempts.

The occ install command should have some error output then, I will try to implement some error handing of this, but it was not too easy AFAIK due to the commands error code and handling. As the error could have been earlier on e.g. MySQL install or configuration, would be nice to have a log file that contains the whole dietpi-software output. But this is again more difficult.

But good that it finally worked.

LexiconCode, 2018-09-04:
What SBC devices do you own personally? @MichaIng.  
Worst-case scenario I can just send you an image file.

MichaIng, 2018-09-04:
@LexiconCode 
I have a RPi2 in use here and some VMs for testing/development purpose.
Have two NanoPi here as well, but currently no time and space to set them up for a real purpose.

But it's okay, if it works for you now, we can leave this topic. I anyway planned to review and tune the ownCloud/Nextcloud install code and update some links to Nextcloud 14. I will add an error handling for the indeed important occ install step them.

Note to self:
- The reason why `occ maintenance:install ...` cannot be easily error handled, is because it is a PHP command (ncc/occ alias for `sudo -u www-data php /var/www/(nextcloud|owncloud)/occ`).
- The `php` command itself does not fail (no error code), the related php scripts are fine, just the output of the command contains an error message.
- To catch an error, we need to catch and scrape the commands output for error markers. Need to create some failure scenarios to find the phrase that is contained in every error output.
- Perhaps asking on Nextcloud forum/GitHub how to do it best.