MichaIng/DietPi #2044: dietpi-backup : progress indicator

adamotte, 2018-08-29:
## Creating a feature request:

#### Is your feature request related to a problem? Please describe:

- Kind of... WIth dietpi-backup, we can sync our data to another drive but during the process, we have no clue about time pending, time passed or percentage of data already done.

#### Describe the solution you'd like:

- Something like an estimal pending time's or percentage progression's .

#### Describe alternatives you've considered:

- Take a :coffee: or :tea:

#### Additional context:

- Actual behavior : animated dots
[image]

adamotte, 2018-08-29:
I just found that we can add *pv* command to monitor progress of data sent via a pipe.

[image]

MichaIng, 2018-08-29:
The problem with G_RUN_CMD is that it generally redirects all output (I guess including all progress bars etc.) into a file: `$G_ERROR_HANDLER_COMMAND &> /tmp/G_ERROR_HANDLER_COMMAND`

I guess we would need to call it without G_RUN_CMD and call G_ERROR_HANDLER just afterwards. The only issue then is, that either we can hide debug (transfered file list) output, which can then also not be reviewed afterwards, or show it as well within terminal, which I really don't like, because it spams/fills the whole terminal history, removing output before the actual transfer.

Not sure if there is a way via rsync option or pv piping to redirect just the transfered file list to a log file?

Fourdee, 2018-08-31:
- 🈯️ DietPi-Backup
- DietPi-Sync

Fourdee, 2018-08-31:
Completed.

closed 2018-08-31 by Fourdee