MichaIng/DietPi #2047: PlexPy failed to start

actraiser30, 2018-09-01:
Hello
On a fresh image updated to 6.14 in a raspberry pi 3 i installed plex server / proftp file server and plexpy all ok rebooted and after upon starting of the services i get Failed in red in PlexPy service

MichaIng, 2018-09-01:
@actraiser30 
Thanks for your report. I will try to replicate, although on VM.

Fourdee, 2018-09-01:
ASUS TB S test (ARMv7):
- 🈴 ```dietpi-software install 42 146 94```

```
[FAILED] DietPi-Services | plexpy       failed (Result: exit-code) since Sat 2018-09-01 20:29:53 BST; 5s ago
● plexpy.service - Tautulli - Stats for Plex Media Server usage
   Loaded: loaded (/etc/systemd/system/plexpy.service; disabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Sat 2018-09-01 20:29:53 BST; 5s ago
  Process: 8154 ExecStart=/opt/plexpy/PlexPy.py --quiet --daemon --nolaunch --config /opt/plexpy/config.ini --datadir /mnt/dietpi_userdata/plexpy (code=exited, status=1/FAILURE)

Sep 01 20:29:53 DietPi PlexPy.py[8154]:     from Tautulli import main
Sep 01 20:29:53 DietPi PlexPy.py[8154]:   File "/opt/plexpy/Tautulli.py", line 35, in <module>
Sep 01 20:29:53 DietPi PlexPy.py[8154]:     import plexpy
Sep 01 20:29:53 DietPi PlexPy.py[8154]:   File "/opt/plexpy/plexpy/__init__.py", line 33, in <module>
Sep 01 20:29:53 DietPi PlexPy.py[8154]:     from apscheduler.schedulers.background import BackgroundScheduler
Sep 01 20:29:53 DietPi PlexPy.py[8154]:   File "/opt/plexpy/lib/apscheduler/__init__.py", line 1, in <module>
Sep 01 20:29:53 DietPi PlexPy.py[8154]:     from pkg_resources import get_distribution, DistributionNotFound
Sep 01 20:29:53 DietPi PlexPy.py[8154]: ImportError: No module named pkg_resources
```

Fourdee, 2018-09-01:
@actraiser30 

Solution is to install the following pre-req package:
```
G_AGI python-setuptools
```

We'll get this fixed for v6.15 also. Thanks for reporting 👍

Fourdee, 2018-09-01:
Fresh test installed passed 🈯️ 

Completed.

closed 2018-09-01 by Fourdee