MichaIng/DietPi #2048: DietPi-Swapfile auto-sizing can try to create it larger than available free space

Jaffa, 2018-09-01:
#### Details:
- Date           | Sat  1 Sep 12:16:01 BST 2018
- Bug report     | N/A
- DietPi version | v6.14
- Img creator    | DietPi Core Team
- Pre-image      | Raspbian Lite
- SBC device     | RPi Zero W (armv6l) (index=1)
- Kernel version | #1134 Tue Aug 14 16:58:07 BST 2018
- Distro         | stretch (index=4)
- Command        | fallocate -l 1567M /var/swap
- Exit code      | 1
- Software title | DietPi-Swapfile

#### Steps to reproduce:
1. Leave default `AUTO_SETUP_SWAPFILE_SIZE=1` in `dietpi.txt`
2. Install v6.13 on to 2GB SD card using Etcher
3. Configure `AUTO_SETUP_AUTOMATED`.
4. Wait

#### Expected behaviour:
`DietPi-Swapfile` configures swap size to take into account free space on device.

#### Actual behaviour:
According to the comment in `dietpi.txt`, `AUTO_SETUP_SWAPFILE_SIZE=1` tries to set the swap file to 2GB - _RAM_. On a Raspberry Pi Zero W, this explains why it's trying to create a 1.5GB swap file.

However, on a 2GB SD card, which Etcher/the default image has partioned into a 1.8GB `/` partition, and ~500MB has been used by the base install, the swap size should probably be maxed out at about half of the free space (~500-600MB).

#### Additional logs:
```
[  OK  ] DietPi-Software | Free space check: path=/ | available=1207 MB | required=500 MB
...
[  OK  ] DietPi-Swapfile | swapoff -a
[ INFO ] DietPi-Swapfile | Deleting existing swapfile (/var/swap)
[ SUB1 ] DietPi-Swapfile > Generating new swapfile
[ INFO ] DietPi-Swapfile | Size     = 1567 MB
[ INFO ] DietPi-Swapfile | Location = /var/swap
[FAILED] DietPi-Swapfile | fallocate -l 1567M /var/swap
fallocate: fallocate failed: No space left on device
```

Jaffa, 2018-09-01:
`/DietPi/dietpi/func/dietpi-set_dphys-swapfile` currently does:

```
SWAP_SIZE_TARGET=$(( 2048 - $(free -m | grep -m1 'Mem:' | awk '{print $2}') ))
```

We can get the disk space available in MB for `$SWAP_LOCATION` with, and half it, with:

```
MAX_SWAP_SIZE=$(( $(df --output=avail /var/opt | tail -n 1) / 1024 / 2))
if (( $SWAP_SIZE_TARGET > $MAX_SWAP_SIZE )); then
  SWAP_SIZE_TARGET=$MAX_SWAP_SIZE
fi
```

Jaffa, 2018-09-01:
Confirmed the following works:

```
		MAX_SWAP_SIZE=$(( $(df --output=avail $(dirname "$SWAP_LOCATION") | tail -n 1) / 1024 / 2))
		if (( $SWAP_SIZE_TARGET > $MAX_SWAP_SIZE )); then

			SWAP_SIZE_TARGET=$MAX_SWAP_SIZE

		fi
```

MichaIng, 2018-09-01:
@Jaffa 
Many thanks for your report and providing already the solution 👍.

We already have a free space check function (`[  OK  ] DietPi-Software | Free space check: path=/ | available=1207 MB | required=500 MB`), but the 500M is just checked as default on `dietpi-software` start, but is not used/checked on swapfile creation. We will fix this for v6.15.

Jaffa, 2018-09-01:
Ah, cool. Great stuff!

Fourdee, 2018-09-01:
@Jaffa @MichaIng 

Resolved for v6.15 (ETA: few days release):
```
root@DietPi:~# /DietPi/dietpi/func/dietpi-set_dphys-swapfile 1000000 /var/swap
[  OK  ] Root access verified.
[FAILED] DietPi-Swapfile | Free space check: path=/var/swap | available=12996 MB | required=1000000 MB
[FAILED] DietPi-Swapfile | Unable to generate swapfile due to insufficient disk space. Swap will be disabled.

 DietPi-Swapfile
─────────────────────────────────────────────────────
 Mode: Disable swapfile

[  OK  ] DietPi-Swapfile | swapoff -a
[ INFO ] DietPi-Swapfile | Deleting existing swapfile (/var/swap)
[ INFO ] DietPi-Swapfile | Setting /tmp tmpfs size: 1006M
[  OK  ] DietPi-Swapfile | mount -o remount,size=1006M tmpfs /tmp
```

Completed.

closed 2018-09-01 by Fourdee

Jaffa, 2018-09-03:
@Fourdee, thanks. That works great when you've explicitly requested a size (i.e. >= 2). When it's set to 1 (AUTO), should it be more intelligent (as I suggest above) in selecting a size? Or should it just give up, as you've done?

Fourdee, 2018-09-04:
@Jaffa 

>   That works great when you've explicitly requested a size (i.e. >= 2). When it's set to 1 (AUTO), should it be more intelligent (as I suggest above) in selecting a size? 

Regardless of the size input (auto or manual), if the required space is lower than what is available, the swapfile will simply be removed (if it exists), then disabled.

Please bear in mind, this fix will be only be available in v6.15 release. Which has been delayed a few days due to RL.

MichaIng, 2018-09-04:
@Jaffa 
Makes sense, although `$AVAILABLE / 2` is perhaps a bid random and still dangerous if the available space is already small. Maybe better `$AVAILABLE - 500M`, as 500M is what we check before starting up dietpi-software, and block creation completely if result is negative, thus less then 500M is available?

Maybe also 200M free is sufficient for daily work + some software install. `dietpi-software` could be adjusted for this, also since downloads are not placed inside RAM and auto removed after extraction. Of course can always fail, if you just select too much software at once, thus would put more responsibility to end user. But actually this is another topic 😄.

Jaffa, 2018-09-04:
Yep, `$AVAILABLE - 500M` (or similar) would be smarter.