MichaIng/DietPi #2055: Dietpi becomes unresponsive

nnovaes, 2018-09-03:
## Creating a bug report/issue:

My dietpi server becomes unresponsive and i can't seem to find the cause of this issue. it happens at random times, after a couple hours the server starts. after starting the server all the installed software works as expected.

this is what i have installed:

[* ] 1    Samba Client: access network shares                                                                      │ 
      │    [* ] 2    Folding@Home: distributed disease research project                                                       │ 
      │    [* ] 42   Plex Media Server: web interface media streaming server                                                  │ 
      │    [* ] 44   Transmission: bittorrent server with web interface (c)                                                   │ 
      │    [* ] 87   SQlite: database                                                                                         │ 
      │    [* ] 96   Samba: feature-rich file server                                                                          │ 
      │    [* ] 101  Log Rotate: rotates log files                                                                            │ 
      │    [* ] 102  Rsyslog: system logging                                                                                  │ 
      │    [* ] 104  Dropbear: ssh server                                                                                     │ 
      │    [* ] 110  NFS Client: network file system client                                                                   │ 
      │    [* ] 115  Webmin: web interface system management                                                                  │ 
      │    [ ] 137  CloudPrint: print server for google cloud print                                                          │ 
      │    [ ] 144  Sonarr: automatically download TV shows                                                                  │ 
      │    [ ] 145  Radarr: automatically download movies                                                                    │ 
      │    [ ] 147  Jackett: api Support for your torrent trackers.                                                          │ 
      │    [ ] 150  Mono: runtime libraries and repo    

i don't suspect of Folding@Home and cloudprint because this is a pre 6.12 issue and i the fact these applications installed didn't change anything.

#### Required Information:
- DietPi version	| 6.14
- Distro version	| stretch
- Kernel version	| echo $G_DISTRO_NAME` or `cat /etc/debian_version
- SBC device		| Virtual Machine (x86_64)
- Power supply used	| N/A
- SDcard used		| N/A

#### Additional Information (if applicable):
- Software title	| (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?
- `dietpi-bugreport` ID

#### Steps to reproduce:

1. start VM
2. wait few random number of hours/days
3. VM processing gets high, memory utilization tops and all services don't reply anymore. server is effectively frozen

#### Extra details:
i was not able to send the bugreport. see below

root@DietPi:~# dietpi-bugreport
[  OK  ] DietPi-Bugreport | Root access verified.
[  OK  ] DietPi-Bugreport | RootFS R/W access verified.

[ INFO ] DietPi-Bugreport | Packing upload archive, please wait...
[ .... ] DietPi-Bugreport | (1/5) Testing connection to 185.101.93.93, please wa[  OK  ] DietPi-Bugreport | Connection test: 185.101.93.93

[FAILED] DietPi-Bugreport | The bug report upload archive appears to be unexpected large. Please inspect and in case clean up the locations to be uploaded, as their size should never be that large:

MichaIng, 2018-09-07:
@nnovaes 
Thanks for your report.

Do you have a chance to check the locations within the bug report failure prompt?
You can use `G_TREESIZE /path/to/location` to check dir and subdir sizes. Perhaps start with `/var/log`, which might as well provide info what exactly failed. Check especially:
```
/var/log/syslog
/var/log/daemon.log
```
for error messages until failure time.
`dmesg` can provide related error messages as well, but not sure if with `rsyslog` this is persistent after reboot or not 🤔.

Fourdee, 2018-09-07:
@MichaIng @nnovaes 

I suspect low memory issue and swapdisk full.

@nnovaes 

How much memory is the VM running, and, what is the swapdisk size?

Please run ```htop``` and take a screenshot, please paste in your reply. Ideally, if you can replicate the slow issue, please take another screenshot of ```htop```

---

Another option to assist with debugging system usage, is to install netdata: http://dietpi.com/phpbb/viewtopic.php?f=8&t=5&start=60#p1611

nnovaes, 2018-09-07:
I hate to say "feel" for such technical stuff, but... i "feel" there's something that, with time, gets stuck and then happens to freeze the VM. 

Here are the outputs:

@Fourdee 
Unfortunately, I can't replicate the issue and by the time it happens it becomes unresponsive. 
Screenshot to htop --> [https://www.evernote.com/l/AV1IIDHD66JK8KkHywoj2l3S0Bcx6ZGCwt4](https://www.evernote.com/l/AV1IIDHD66JK8KkHywoj2l3S0Bcx6ZGCwt4 ).

Allocated memory is 4GB and normal use is around 1.5GB
Memory chart --> https://www.evernote.com/l/AV10NPfJ2khHY5onaUjbNAlyQ4U4ZB_Ukq4
CPU chart --> [https://www.evernote.com/l/AV3vWVdG0RlAfZHrPXMez5CQzjsguqpUujA](https://www.evernote.com/l/AV3vWVdG0RlAfZHrPXMez5CQzjsguqpUujA)

@MichaIng 
_root@DietPi:~# G_TREESIZE /var/log/syslog
[  OK  ] Root access verified.
6.7 MB /var/log/syslog
root@DietPi:~# G_TREESIZE /var/log/daemon.log
46.8 MB /var/log/daemon.log
root@DietPi:~#_

Wondering if may be worth to have a cron job capturing the output of htop every hour or so...

Fourdee, 2018-09-09:
@nnovaes 

> Wondering if may be worth to have a cron job capturing the output of htop every hour or so...

Yep, or ```top -b -n 1 >> /root/debug.txt```

Strange that swap is being used with memory available and ```vm.swapiness=1```.

Could be filesystem IO bottle-necking. Might be worth installing netdata to analyse disk IO usage aswell.

MichaIng, 2018-09-26:
@nnovaes 
Ui time has passed.

So the reason for bug report error is the very large /var/log dir, 115M all together, 47M daemon.log is veeery large. It might be even hard to read this out via `nano`.

Should work to read the last entries:
```
tail /var/log/daemon.log
tail /var/log/syslog
```

I think I would first clean this up:
```
> /var/log/daemon.log
> /var/log/<otherLargeLogFiles>
```
- Perhaps `/var/log/debug` might be large as well?

MichaIng, 2018-10-14:
I will mark this as closed due to missing user reply. However DietPi-Bugreport failed due to /var/log being very large, which might be a software title or system spamming to log due to an issue and being related to the freeze.

Feel free to reopen, especially if a certain misconfiguration can be identified, that we could fix our side.

closed 2018-10-14 by MichaIng