MichaIng/DietPi #2057: DietPi-Software | Nextcloud: Clients can't find CalDav/CardDav services.

LexiconCode, 2018-09-05:
- DietPi version	| `6.14`
- Distro version	| `Stretch`
- SBC device		| `Odroid HC2` or (EG: RPi3)
- Power supply used	|  Official Power Supply
- SDcard used		| ( SanDisk ultra)

#### Additional Information (if applicable):
- Software title	| Nextcloud
- Was the software title installed freshly or updated/migrated? Freshly installed 
- Can this issue be replicated on a fresh installation of DietPi? yes
- `dietpi-bugreport` ID

#### Steps to reproduce:
1. ... Install next cloud
2. ... Try to use a WebDAV client to connect. In this case specifically DavDroid 

#### Expected behaviour:
WebDAV should connect without further configuration from an end-user

#### Actual behaviour:
Nextcloud WebDAV services aren't seen by WebDAV clients

#### Extra details:

[debug.txt](https://github.com/Fourdee/DietPi/files/2353568/debug.txt)

MichaIng, 2018-09-06:
@LexiconCode 
> Checking user-given URL: https://CensoredURL/nextcloud
As we do not place any `.well-known` dirs into the webroot (or into nextcloud directory), or redirect those to the final Cal/Card/WebDAV URLs, you need to add the full URL.

Go to Nextcloud web UI, navigate to calendar and contacts app respectively, open settings at the bottom left. There you can copy the Cal/CardDAV URL. In my case e.g. its:
- Calendar: https://my.domain.org/nextcloud/remote.php/dav/
- Contacts: https://my.domain.org/nextcloud/remote.php/dav/addressbooks/users/Micha/contacts/

DAVdroid btw. accesses Card- and CalDAV, not WebDAV, which is for data transfer, e.g. used by the Nextcloud clients. Can you configure Nextcloud desktop/mobile clients without issues?

If with the above method sync works as expected, we can turn the issue into a feature request to allow adding URLs <domain>/nextcloud only, e.g. via .well-known redirects. Have to check how this can be done best without interfering other web services.

MichaIng, 2018-09-07:
@LexiconCode 
Just another quick question, which webserver do you use?

In case of nginx e.g. the following entries inside `location ^~ /nextcloud {` statement should do it, without creaking any other parallel Cal/CardDAV servers: https://docs.nextcloud.com/server/14/admin_manual/installation/nginx.html#nextcloud-in-a-subdir-of-nginx
```
    location = /nextcloud/.well-known/carddav {
      return 301 $scheme://$host/nextcloud/remote.php/dav;
    }
    location = /nextcloud/.well-known/caldav {
      return 301 $scheme://$host/nextcloud/remote.php/dav;
    }
```
There are similar possibilities for other webservers.

LexiconCode, 2018-09-07:
I'm using the Apache Web server. Normal clients desktop/android next code clients connect as expected. 

I believe the configuration for enabling Apache.
```
Redirect 301 /.well-known/carddav /nextcloud/remote.php/dav
Redirect 301 /.well-known/caldav /nextcloud/remote.php/dav
```

I'll get back to you today but the result of testing.

MichaIng, 2018-09-08:
@LexiconCode 
I believe it's
```
Redirect 301 /nextcloud/.well-known/carddav /nextcloud/remote.php/dav
Redirect 301 /nextcloud/.well-known/caldav /nextcloud/remote.php/dav
```
then, if `your.domain.org/nextcloud` shall work. Yours should further allow DAVdroid to find the endpoints by just entering only `your.domain.org` 🤔.

LexiconCode, 2018-09-08:
Which file do edit and were?
`.htaccess`? `nextcloud.conf`?

MichaIng, 2018-09-09:
@LexiconCode 
Not sure if it works inside `.htaccess`, there are some limitations and more importantly it gets overwritten by Nextcloud on update and occ maintenance commands.

So better add it to `/etc/apache2/sites-available/nextcloud.conf`, inside the <Location> statement as a first try. Should work, but if Apache service complains syntax error, perhaps is needs to be outside the <Location> statement. Effect will be the same anyway.

LexiconCode, 2018-09-11:
I will work on finding and testing some Cal/CardDAV clients tomorrow. DAVdroid works.

MichaIng, 2018-09-13:
I have a good idea how we can handle it and by this implement redirection into other DAV servers as well:
- We place redirection into own config files `dietpi-dav_redirect.conf` or something.
- Baikal (and in case other explizite Cal/CardDAV servers we offer) will overwrite this file, if existent.
- ownCloud/Nextcloud (and other software that optionally can be used as Cal/CardDAV server), will only create it, if the file does not yet exist.
- On software uninstall, the config is only removed, if created by/pointing to the related Cal/CardDAV server.

LexiconCode, 2018-09-13:
Everything checks out!

MichaIng, 2018-09-13:
@LexiconCode 
Great, will finish the PR about this tomorrow. The other new network issues prevented from finish this today 😅.

MichaIng, 2018-09-14:
@LexiconCode 
I added the redirects to the PR: https://github.com/Fourdee/DietPi/pull/2065

For Apache, I followed the method that is used within their `.htaccess`, using `mod_rewrite`. Not sure about the difference/advantage to the `Redirect` directive of `mod_alias`: https://httpd.apache.org/docs/2.2/mod/mod_alias.html#Redirect
- The latter seems to be the easier more direct solution, but seems to have disadvantages, why `mod_rewrite` exists(?).
- I had the same question with HTTPS redirects on my system. Some guides use `Redirect` directive, some mod_rewrite. Certbot default configuration uses the latter as well, so somehow this seems to be beneficial. But I will investigate by times when to use which method.

MichaIng, 2018-09-23:
PR finished, finally 😅: https://github.com/Fourdee/DietPi/pull/2065

Testing on all webservers and distro versions passed here.

MichaIng, 2018-10-12:
We implemented this just the right time. Since Nextcloud v14.0.2 there appear admin panel warnings, if `.well-known/caldav|carddav` is not found 😄:
[image]