MichaIng/DietPi #2058: Tinker S: Initial Setup Error

oekinger, 2018-09-05:
#### Details:
- Date           | Thu  3 Nov 18:22:47 CET 2016
- Bug report     | N/A
- DietPi version | v6.14
- Img creator    | DietPi Core Team
- Pre-image      | TinkerOS 2.0.5
- SBC device     | Asus Tinker Board (armv7l) (index=52)
- Kernel version | #1 SMP Thu Feb 22 18:23:31 CST 2018
- Distro         | stretch (index=4)
- Command        | G_AGUP
- Exit code      | 100
- Software title | DietPi-Software

#### Steps to reproduce:

1. Initial Setup -> change pswd
2. ...

#### Expected behaviour:

- ...

#### Actual behaviour:

- ...

#### Extra details:

- ...

#### Additional logs:
```
Log file contents:
Hit:1 http://repo.linaro.org/ubuntu/linaro-overlay stretch InRelease
Ign:3 https://deb.debian.org/debian stretch InRelease
Get:2 http://dl.ubnt.com/unifi/debian stable InRelease [3,023 B]
Ign:2 http://dl.ubnt.com/unifi/debian stable InRelease
Ign:4 https://deb.debian.org/debian stretch-updates InRelease
Ign:5 https://deb.debian.org/debian-security stretch/updates InRelease
Ign:6 https://deb.debian.org/debian stretch-backports InRelease
Err:7 https://deb.debian.org/debian stretch Release
  server certificate verification failed. CAfile: /etc/ssl/certs/ca-certificates.crt CRLfile: none
Err:8 https://deb.debian.org/debian stretch-updates Release
  server certificate verification failed. CAfile: /etc/ssl/certs/ca-certificates.crt CRLfile: none
Err:9 https://deb.debian.org/debian-security stretch/updates Release
  server certificate verification failed. CAfile: /etc/ssl/certs/ca-certificates.crt CRLfile: none
Err:10 https://deb.debian.org/debian stretch-backports Release
  server certificate verification failed. CAfile: /etc/ssl/certs/ca-certificates.crt CRLfile: none
Reading package lists...
W: GPG error: http://dl.ubnt.com/unifi/debian stable InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 06E85760C0A52C50
W: The repository 'http://www.ubnt.com/downloads/unifi/debian stable InRelease' is not signed.
E: The repository 'https://deb.debian.org/debian stretch Release' does no longer have a Release file.
E: The repository 'https://deb.debian.org/debian stretch-updates Release' does no longer have a Release file.
E: The repository 'https://deb.debian.org/debian-security stretch/updates Release' does no longer have a Release file.
E: The repository 'https://deb.debian.org/debian stretch-backports Release' does no longer have a Release file.
```

Fourdee, 2018-09-05:
@oekinger 

Thanks for the report 👍 

Looks like a date issue.

Please can you paste results of the following commands?
```
date
/DietPi/dietpi/func/run_ntpd 1
```

If successful, please re-run the setup process:
```
dietpi-software
```

If problems persist, please try manually changing the APT mirror:
```
dietpi-config 16 1
```

MichaIng, 2018-09-07:
> Date | Thu 3 Nov 18:22:47 CET 2016

Indeed, we really need to carefully recheck how/when time sync is applied first on (first) boot, in case delay further scripts until this is done, prompt user or such 🤔.

Fourdee, 2018-09-07:
@oekinger 

ASUS TB image updated which resolves issues with dates.

Please re-download and re-write the latest image below:
https://dietpi.com/downloads/images/DietPi_ASUSTB-ARMv7-Stretch.7z

Local testing passed, will mark as completed, however @oekinger, if problems persist, please let us know.

Fourdee, 2018-09-07:
@MichaIng 

> Indeed, we really need to carefully recheck how/when time sync is applied first on (first) boot, in case delay further scripts until this is done, prompt user or such

The issue was 
- ```fake-hwclock``` missing
- ```G_CHECK_URL``` runs before time sync check

Solution is
- ```fake-hwclock``` now installed
- ```G_CHECK_URL``` which runs before timesync check, now ignores certificates and any date lapse ```no-check-certificate```
- Time sync check runs afterwards.

closed 2018-09-07 by Fourdee

Fourdee, 2018-09-08:
@MichaIng 

> First check general network connectivity via non-HTTPS request, e.g. http://google.com or any other website that is for sure always available and not redirected to HTTPS.

yep, the current ```--no-check-certificate``` I added has the same effect as a http check, regardless of redirect to https:
https://github.com/Fourdee/DietPi/blob/testing/dietpi/dietpi-software#L188

> Then sync time. If this then fails, prompt choice to reconfigure time sync settings.

Yep, after this basic internet test, then the NTP time sync check is done.

> Only after this was successful, do other web checks/requests, APT etc.

Already in place:
https://github.com/Fourdee/DietPi/blob/testing/dietpi/dietpi-software#L16011

> Then for regular G_CHECK_URL we should leave SSL checks inside.

Yep, all other G_CHECK_URL have cert check, its only the initial connection test in ```dietpi-software``` which now ignores cert (and date) to allow NTP check to run straight afterwards.

oekinger, 2018-09-08:
@Fourdee  The fix did it :-) everything works as expected. Thank you.