MichaIng/DietPi #2060: Hyper-V images?

ghosted17, 2018-09-07:
Hi any change we can get hyper-v images?

MichaIng, 2018-09-07:
@ghosted17 
Please contribute to the existing topic: https://github.com/Fourdee/DietPi/issues/804

I am still not fully disinclined, but Microsoft does not make it easy to run different VM platforms on the same machine 😠. It's always a question, if it makes sense to go so many steps (+effort) towards a company/solution, whose goal is to be the only provider, obviously. Better focus and support the others, that make their concurrent implementation easy, are open source etc, from a developer point of view?

But yeah, since it is the last of three widely used VM platforms, perhaps we have to ignore Microsofts behaviour here for the benefit of our users?

I will reconsider after v6.15 release and close this topic in favour of the linked one.

closed 2018-09-07 by MichaIng

ghosted17, 2018-09-07:
I also had some issues with hyper-v and couldn't run any other vm soft and had to find workarounds. But hey it's a type 1 VM so can't complain that much as all type 1 have there issues.

I'm sure there are plenty of Windows users that use Windows as there primary os / server os and use linux for exotic lightweight services. Rather then buying VMware or running Vbox. Hyper-V is free and build right into the OS especially for Win server services.

When ever/if you get a chance I'd love to test out an image, it should run great and better then running a full linux disto for some simple service.

Fourdee, 2018-09-09:
@ghosted17 

VM workstation player is free:
https://my.vmware.com/en/web/vmware/free#desktop_end_user_computing/vmware_workstation_player/14_0

Allows you to run multiple VM's on a windows system. Its basically the old Player + multiple VM's and improved control.

ghosted17, 2018-09-09:
thanks for the info but I run Win server 16 with hyper v servers and they are rock solid. Don't know how reliable it would be to run vm players that are on 24/7 server and have them auto start / shutdown with Windows.

MichaIng, 2018-09-09:
@ghosted17 
Not to forget VirtualBox, which allows quite useful VM snapshots and is the only open source solution of this three.

All of those solutions allow autostart with Windows and work reliable, there is no advantage of Hyper-V from that point of view. But I understand that you want to stay with the solutions you are used to and made no bad experience with.

ghosted17, 2018-09-09:
I was also exploring alternatives when i had usb pass through issues but For me it's not worth redoing everything as the vms are second priority after Win. if i had to do it over i might go esxi but again for me there maybe a performance loss to windows which i don't want and then im sure esxi will have other issues that i will have to deal with like having to use vmfs for all drives. I hate that there's hardly any goos recovery software for ntfs I'd imaging vmfs would be a nightmare to recover.

yumiris, 2018-09-19:
In #804, I occasionally post snapshots of Hyper-V images with the bare minimum modifications to get them flawlessly running on Hyper-V.. @ghosted17, I just posted a link to v6.15 over there, if you want to have a look.

I'm up for doing this on a much more regular basis. @MichaIng - perhaps I should make a more "official" issue where I'll regularly put Hyper-V images?

yumiris, 2018-09-20:
@MichaIng - yup, all of my images derive from that set of instructions.

Just to confirm, you want me to get Diet running on Hyper-V & VirtualBox/VMware on the same host? 🤔

MichaIng, 2018-09-20:
> yup, all of my images derive from that set of instructions.

Ah perfect.

> Just to confirm, you want me to get Diet running on Hyper-V & VirtualBox/VMware on the same host? 🤔

Hehe no 😄, as I am already creating VMware and VBox images regularly, I could create the Hyper-V images as well in one rush. But the problem is, that if Hyper-V is even installed regularly, it blocks CPU virtualisieren usage for any other VM software, thus VMware and VBox cannot start any more 😠. But it seems to be a background service that does this block, that can be stopped/disabled. Perhaps, when one wants to create all three images on one machine, this Hyper-V service always need to be toggled accordingly. A bid annoying and shame on Microsoft for this unnecessary thing, but it would work at least.

yumiris, 2018-09-20:
Ah, my bad. Indeed, Hyper-V breaks other hypervisors (and even Vagrant :angry:). Hmm, maybe an alternative is to run a Windows guest in VMware/VirtualBox with Hyper-V installed on it?