MichaIng/DietPi #2061: Nanopc T4 - dietpi-software fails to access repo during initial boot due to incorrect date

bitblt, 2018-09-08:
## Creating a bug report/issue:
dietpi-software fails to access the debian software repos because the system date is incorrect during the initial boot of DietPi on Nanopc-T4.  It reports the following in the log after the 5th attempt:

 Resolving deb.debian.org (deb.debian.org)... 128.31.0.62, 149.20.4.15,
                   │ 130.89.148.14, ...
                   │ Connecting to deb.debian.org (deb.debian.org)|128.31.0.62|:443... connected.
                   │ ERROR: The certificate of ‘deb.debian.org’ is not trusted.
                   │ ERROR: The certificate of ‘deb.debian.org’ is not yet activated.
                   │ The certificate has not yet been activated

#### Required Information:
- DietPi version	| 6.14
- Distro version	| stretch
- Kernel version	| Linux DietPi 4.4.126 #7 SMP Thu Jul 19 22:51:16 EST 2018 aarch64 GNU/Linux
- SBC device		| NanoPC T4 (aarch64)
- Power supply used	| 12VDC 2A [from Friendly kit]
- SDcard used		| eMMC

#### Additional Information (if applicable):
- Software title	| dietpi-software
- Was the software title installed freshly or updated/migrated? fresh install
- Can this issue be replicated on a fresh installation of DietPi? yes
- `dietpi-bugreport` ID

#### Steps to reproduce:

1. Install fresh copy of NanoPC-T4 image
2. Login as root
3. Walk-thru password default prompts [choice doesn't matter]
4. Script begins to update software and logs show certificate has not been activated

#### Expected behaviour:
- timezone set before repo accessed using HTTPS
- time set by using NTP server (systemctl start systemd-timesyncd)
- dietpi-software accesses repos using HTTPS 

#### Actual behaviour:
- initial system date is several months in the past
- HTTPS certificate validation fails

#### Extra details:
- Work-around:
- after dietpi-software fails:
1. `dietpi-config` 
- set timezone under the regional settings
- exit dietpi-config without reboot (exit with cancel)
2. `systemctl start systemd-timesyncd`
3, `reboot`

dietpi-software should run to completion

Fourdee, 2018-09-08:
@bitblt 

Thanks for the report :+1: 

Apologies, this is an known issue with pre v6.14 images. We have the code fix in place, however, we need to update the effected images as this code runs pre-patching:
https://github.com/Fourdee/DietPi/issues/2026#issuecomment-414383868

Please bear with us, we'll try to update this image ASAP. In the mean time, please manually set the date after the initial connection test fails, eg:
```
#YYYYMMDD
date +%Y%m%d -s "20180817"
```

Then you can continue setup with:
```
dietpi-software
```

Fourdee, 2018-09-08:
I'll mark this as closed in favor of: https://github.com/Fourdee/DietPi/issues/2026#issuecomment-414383868

closed 2018-09-08 by Fourdee