MichaIng/DietPi #2064: Odroid C2: Can't run dpkg -i. It says Can't locate POSIX.pm

lateboomer, 2018-09-10:
Hi,

I tried to install LMS package but it failed with message:
```
root@OdrDiet:/home# dpkg -i logitechmediaserver_7.9.2_1527861596_arm.deb
(Reading database ... 18596 files and directories currently installed.)
Preparing to unpack logitechmediaserver_7.9.2_1527861596_arm.deb ...
Can't locate POSIX.pm in @INC (you may need to install the POSIX module) (@INC contains: /etc/perl /usr/local/lib/aarch64-linux-gnu/perl/5.24.1 /usr/local/share/perl/5.24.1 /usr/lib/aarch64-linux-gnu/perl5/5.24 /usr/share/perl5 /usr/lib/aarch64-linux-gnu/perl/5.24 /usr/share/perl/5.24 /usr/local/lib/site_perl /usr/lib/aarch64-linux-gnu/perl-base) at /usr/share/perl5/Debconf/Template.pm line 7.
BEGIN failed--compilation aborted at /usr/share/perl5/Debconf/Template.pm line 7.
Compilation failed in require at /usr/share/perl5/Debconf/Question.pm line 8.
BEGIN failed--compilation aborted at /usr/share/perl5/Debconf/Question.pm line 8.
Compilation failed in require at /usr/share/perl5/Debconf/Config.pm line 7.
BEGIN failed--compilation aborted at /usr/share/perl5/Debconf/Config.pm line 7.
Compilation failed in require at /usr/share/perl5/Debconf/Log.pm line 10.
Compilation failed in require at /usr/share/perl5/Debconf/Db.pm line 7.
BEGIN failed--compilation aborted at /usr/share/perl5/Debconf/Db.pm line 7.
Compilation failed in require at /usr/share/debconf/frontend line 6.
BEGIN failed--compilation aborted at /usr/share/debconf/frontend line 6.
dpkg: error processing archive logitechmediaserver_7.9.2_1527861596_arm.deb (--install):
 subprocess new pre-installation script returned error exit status 2
Errors were encountered while processing:
 logitechmediaserver_7.9.2_1527861596_arm.deb
```

#### Required Information:
- DietPi version	| 6.14
- Distro version	| stretch 9.5
- Kernel version	| Linux OdrDiet 3.16.57+ #1 SMP PREEMPT Sun Aug 19 15:27:56 CEST 2018 aarch64 GNU/Linux
- SBC device		| Odroid C2 (aarch64)
- Power supply used	| 5V 2A
- SDcard used		| SanDisk ultra 16 GB

#### Extra details:
May need to refresh Perl so that can run dpkg -i
- ...

MichaIng, 2018-09-10:
@lateboomer 
Thanks for your report. We try to replicate.

lateboomer, 2018-09-11:
Hi,
This is my Perl on Odroid C2:
```
root@OdrDiet:~# dpkg -l | grep perl
ii  libfile-copy-recursive-perl 0.38-1                           all          Perl extension for recursively copying files and directories
ii  liblocale-gettext-perl      1.07-3+b1                        arm64        module using libc functions for internationalization in Perl
ii  libperl5.24:arm64           5.24.1-3+deb9u4                  arm64        shared Perl library
ii  perl                        5.24.1-3+deb9u4                  arm64        Larry Wall's Practical Extraction and Report Language
ii  perl-base                   5.24.1-3+deb9u4                  arm64        minimal Perl system
ii  perl-modules-5.24           5.24.1-3+deb9u4                  all          Core Perl modules
```

I downloaded Dietpi V6.13 and did fresh install on Odroid C2. After finished it became v6.14. After that I changed hostname to OdrDiet and time zone and language. Then I installed ALSA and also Squeezelite from software menu. Also I had installed OpenSSH  and Samba Server and apt-get install openssh-sftp-server. Then after that I tried to run dpkg -i on LMS package but it failed.

lateboomer, 2018-09-11:
Hi,

I tried "dietpi-software install 35" but I saw it got error message on dpkg -i as shown:

```
[  OK  ] DietPi-Software | G_AGUP
[  OK  ] DietPi-Software | Connection test: https://dietpi.com/downloads/binaries/all/logitechmediaserver_7.9.1_all.deb
[ INFO ] DietPi-Software | G_THREAD_START_0 (PID:3543) | wget https://dietpi.com/downloads/binaries/all/logitechmediaserver_7.9.1_all.deb -O 35.deb
[  OK  ] DietPi-Software | G_THREAD: All threads finished
**[FAILED] DietPi-Software | dpkg -i 35.deb**
[ INFO ] DietPi-Software | Trying automated APT fix
[  OK  ] DietPi-Software | APT fix, please wait...
Reading package lists...
Building dependency tree...
Reading state information...
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

[  OK  ] DietPi-Software | G_AGF
[  OK  ] DietPi-Software | Cleaning download directory
[  OK  ] DietPi-Software | Connection test: https://dietpi.com/downloads/binaries/all/logitechmediaserver_7.9.1_CPAN-armv8-stretch.zip
[ INFO ] DietPi-Software | G_THREAD_START_0 (PID:3608) | wget https://dietpi.com/downloads/binaries/all/logitechmediaserver_7.9.1_CPAN-armv8-stretch.zip -O 35.zip
[  OK  ] DietPi-Software | APT installation for: libxml-parser-perl zlib1g-dev libjpeg-dev libpng-dev libjpeg62-turbo-dev, please wait...
debconf: **Perl may be unconfigured (Can't locate IO/File.pm in @INC** (you may need to install the IO::File module) (@INC contains: /etc/perl /usr/local/lib/aarch64-linux-gnu/perl/5.24.1 /usr/local/share/perl/5.24.1 /usr/lib/aarch64-linux-gnu/perl5/5.24 /usr/share/perl5 /usr/lib/aarch64-linux-gnu/perl/5.24 /usr/share/perl/5.24 /usr/local/lib/site_perl /usr/lib/aarch64-linux-gnu/perl-base) at /usr/share/perl/5.24/FileHandle.pm line 9.
Compilation failed in require at (eval 1) line 3.
BEGIN failed--compilation aborted at (eval 1) line 3.
) -- aborting
Selecting previously unselected package netbase.
(Reading database ... 18596 files and directories currently installed.)
Preparing to unpack .../00-netbase_5.4_all.deb ...
Unpacking netbase (5.4) ...
Selecting previously unselected package libc-dev-bin.
```
After installation of 35, I rebooted Odroid C2 and I ran systemctl status as shown:

```
root@OdrDiet:~# systemctl status squeezeboxserver.service
● squeezeboxserver.service - Squeezebox Server (LMS)
   Loaded: error (Reason: Invalid argument)
   Active: inactive (dead)

Sep 11 11:18:34 OdrDiet systemd[1]: [/etc/systemd/system/squeezeboxserver.service:
6] Executable path is not absolute, ignoring: --prefsdir /var/lib/squeezeboxserver/prefs -
-logfile /var/log/squeezeboxserver/error.log --logdir /var/log/squeezeboxserver/ --cachedi
r /var/lib/squeezeboxserver/cache --user root
Sep 11 11:18:34 OdrDiet systemd[1]: squeezeboxserver.service: Service lacks both E
xecStart= and ExecStop= setting. Refusing.
Sep 11 11:19:03 OdrDiet systemd[1]: [/etc/systemd/system/squeezeboxserver.service:
6] Executable path is not absolute, ignoring: --prefsdir /var/lib/squeezeboxserver/prefs -
-logfile /var/log/squeezeboxserver/error.log --logdir /var/log/squeezeboxserver/ --cachedi
r /var/lib/squeezeboxserver/cache --user root
Sep 11 11:19:03 OdrDiet systemd[1]: squeezeboxserver.service: Service lacks both E
xecStart= and ExecStop= setting. Refusing.
Sep 11 11:19:24 OdrDiet systemd[1]: [/etc/systemd/system/squeezeboxserver.service:
6] Executable path is not absolute, ignoring: --prefsdir /var/lib/squeezeboxserver/prefs -
-logfile /var/log/squeezeboxserver/error.log --logdir /var/log/squeezeboxserver/ --cachedi
r /var/lib/squeezeboxserver/cache --user root
Sep 11 11:19:24 OdrDiet systemd[1]: squeezeboxserver.service: Service lacks both E
xecStart= and ExecStop= setting. Refusing.
Sep 11 11:20:04 OdrDiet systemd[1]: [/etc/systemd/system/squeezeboxserver.service:
6] Executable path is not absolute, ignoring: --prefsdir /var/lib/squeezeboxserver/prefs -
-logfile /var/log/squeezeboxserver/error.log --logdir /var/log/squeezeboxserver/ --cachedi
r /var/lib/squeezeboxserver/cache --user root
Sep 11 11:20:04 OdrDiet systemd[1]: squeezeboxserver.service: Service lacks both E
xecStart= and ExecStop= setting. Refusing.
Sep 11 11:20:26 OdrDiet systemd[1]: [/etc/systemd/system/squeezeboxserver.service:
6] Executable path is not absolute, ignoring: --prefsdir /var/lib/squeezeboxserver/prefs -
-logfile /var/log/squeezeboxserver/error.log --logdir /var/log/squeezeboxserver/ --cachedi
r /var/lib/squeezeboxserver/cache --user root
Sep 11 11:20:26 OdrDiet systemd[1]: squeezeboxserver.service: Service lacks both E
xecStart= and ExecStop= setting. Refusing.

```

Fourdee, 2018-09-11:
@lateboomer 

Thanks for the info 👍 

Very strange, i'am unable to replicate this issue.

Please try the following:
- Redownload the DietPi image, write it again and run the software installations as you did previously.
https://dietpi.com/downloads/images/DietPi_OdroidC2-ARMv8-Stretch.7z

> Then after that I tried to run dpkg -i on LMS package but it failed.

Please ensure you install LMS from the ```dietpi-software``` menu, (also known as Squeezebox).

If problems persist, please let me know the following:
- SD/EMMC card used
- Power supply used

lateboomer, 2018-09-13:
Hi,

I downloaded latest Dietpi as given above and it is v6.14, and after installation it becomes v6.15. This is different from my previous download from the official download site as stated above.

Anyway, good news is I am able to install Squeezeboxserver although the message "[FAILED] DietPi-Software | dpkg -i 35.deb" is still appeared, but bad news is it won't start the Squeezebox server as shown:

```
root@OdrDiet:~# systemctl status squeezeboxserver
● squeezeboxserver.service - Squeezebox Server (LMS)
   Loaded: loaded (/etc/systemd/system/squeezeboxserver.service; disabled; vendor preset: enabled)
   Active: inactive (dead)

Sep 13 21:29:40 OdrDiet squeezeboxserver[1127]: 1. Install the latest version of the module(s) using CPAN: sudo cpan Some::Module
Sep 13 21:29:40 OdrDiet squeezeboxserver[1127]: 2. Update the module's package using apt-get, yum, etc.
Sep 13 21:29:40 OdrDiet squeezeboxserver[1127]: 3. Run the .tar.gz version of Logitech Media Server which includes all required CPAN 
modules.
Sep 13 21:30:51 OdrDiet systemd[1]: Started Squeezebox Server (LMS).
Sep 13 21:30:52 OdrDiet squeezeboxserver[1738]: The following CPAN modules were found but cannot work with Logitech Media Server:
Sep 13 21:30:52 OdrDiet squeezeboxserver[1738]:   Proc::Background (loaded <not found>, need 1.08)
Sep 13 21:30:52 OdrDiet squeezeboxserver[1738]: To fix this problem you have several options:
Sep 13 21:30:52 OdrDiet squeezeboxserver[1738]: 1. Install the latest version of the module(s) using CPAN: sudo cpan Some::Module
Sep 13 21:30:52 OdrDiet squeezeboxserver[1738]: 2. Update the module's package using apt-get, yum, etc.
Sep 13 21:30:52 OdrDiet squeezeboxserver[1738]: 3. Run the .tar.gz version of Logitech Media Server which includes all required CPAN 
modules.
```
All thing is not lost because I don't like this custom package because previously when used with CDPlayer plugin, the Squeezeboxserver had bug and took very long time to play cd and playing was stuttering.

Therefore, I uninstalled 35 and installed the downloaded LMS package "logitechmediaserver_7.9.2_1527861596_arm.deb" using dpkg -i and this time it works as shown:
```
root@OdrDiet:~# service logitechmediaserver status
● logitechmediaserver.service - LSB: Startup script for the Logitech Media Server
   Loaded: loaded (/etc/init.d/logitechmediaserver; generated; vendor preset: enabled)
   Active: active (running) since Thu 2018-09-13 21:43:34 +08; 1min 23s ago
     Docs: man:systemd-sysv-generator(8)
  Process: 294 ExecStart=/etc/init.d/logitechmediaserver start (code=exited, status=0/SUCCESS)
   CGroup: /system.slice/logitechmediaserver.service
           ├─322 /bin/bash /usr/sbin/squeezeboxserver_safe /usr/sbin/squeezeboxserver --prefsdir /var/lib/squeezeboxserver/prefs --lo
gdir /var/log/squeezeboxserver/ --cachedir /var/lib/squeezeboxserver/cache --charset=utf8
           └─325 /usr/bin/perl /usr/sbin/squeezeboxserver --prefsdir /var/lib/squeezeboxserver/prefs --logdir /var/log/squeezeboxserv
er/ --cachedir /var/lib/squeezeboxserver/cache --charset=utf8

Sep 13 21:43:34 OdrDiet systemd[1]: Starting LSB: Startup script for the Logitech Media Server...
Sep 13 21:43:34 OdrDiet logitechmediaserver[294]: Making sure that Logitech Media Server is not running first: No process in pidfile 
'/var/run/logitechmediaserver.pid' found running; none killed.
Sep 13 21:43:34 OdrDiet logitechmediaserver[294]: Starting Logitech Media Server.
Sep 13 21:43:34 OdrDiet systemd[1]: Started LSB: Startup script for the Logitech Media Server.

```

Fourdee, 2018-09-13:
@lateboomer 

Thanks for the info.

I'am still unable to replicate the issue you are experiencing on a fresh install:

Install + log:
https://pastebin.com/tpxaygUD
[image]

I believe the issue was:
- Installing ```dpkg -i logitechmediaserver_7.9.2_1527861596_arm.deb``` first (and not through ```dietpi-software```)
- Causing any future install through ```dietpi-software``` to fail due to unknown/unexpected package installation.

As the issue is now resolved, and, we cannot replicate in testing, i'll mark this as closed. Please reopen if required.

closed 2018-09-13 by Fourdee