MichaIng/DietPi #2066: Dietpi-Cloudshell Issue

potter-91, 2018-09-11:
Hi Forderungen
Bugreport id: f356d26d-a01a-4490-98d9-24eed509a7b0
I just made fresh Dietpi installation on my CS2, because of a corrupted Partition.

The Dietpi cloudshell script has some issues displaying the correct amount of ram.
[image]
IT was also with my last install, but there is thought the reason could be a broken Uninstaller of zram.

Sometimes the Screen just goes blank but the backlight is still staying on.

Greetings from Germany ^^

Fourdee, 2018-09-12:
@potter-91 

Many thanks for the report 👍 

I'll take a look.

Fourdee, 2018-09-12:
Confirmed bug with ```free -m``` result. Appears cached result has moved back 1 ```awk```

Fourdee, 2018-09-12:
@potter-91 

> Sometimes the Screen just goes blank but the backlight is still staying on.

Sounds like screen-blanking not being set, i'll have this applied during init.

Fourdee, 2018-09-12:
RAM and a few other issues resolved.

@potter-91 

In regards to the screen blanking (i'am assuming) this should also be resolved, however, if problems persist, please let us know so we can investigate further.

v6.15 (with this fix) will be released today.

closed 2018-09-12 by Fourdee

potter-91, 2018-09-22:
Hi
Tonight the backlight stays on