MichaIng/DietPi #2067: Changing permissions fails on Samba mount

MichaIng, 2018-09-11:
Ref: https://dietpi.com/phpbb/viewtopic.php?f=11&t=4595#p14233
- Does the issue occur, if the server (Windows?) file system (NTFS?) does not support permissions?
- Is it somehow possible via cifs mount options e.g. to adjust owner and permissions for the client side only, e.g. dietpi:dietpi 775 to allow Sonarr etc users r/w access?
- Could this be a similar issue for other network mounts and is there a similar solution possible?

Fourdee, 2018-09-12:
@MichaIng 

Samba does not support setting remote permissions with NTFS volume:
```
root@DietPi:/mnt/samba# chown -R dietpi:dietpi Video
root@DietPi:/mnt/samba# ls -lha
total 4.0K
drwxr-xr-x 2 root root    0 Sep 11 09:35 .
drwxr-xr-x 8 root root 4.0K Jul 23 21:53 ..
drwxr-xr-x 2 root root    0 Dec 12  2016 #Backups
drwxr-xr-x 2 root root    0 Sep 11 13:43 downloads
drwxr-xr-x 2 root root    0 Jul 21 22:14 Music
drwxr-xr-x 2 root root    0 Jul 21 22:14 Pictures
drwxr-xr-x 2 root root    0 Jul 21 22:21 _PRE_BK
drwxr-xr-x 2 root root    0 Jul 21 22:14 Video
```

Unsure in regards to EXT4, will check.

Fourdee, 2018-09-16:
non-root applications are unable to access ```/mnt/samba``` (eg: radarr, sonarr), is it possible to:
- Mount Samba with DietPi group? Allowing access?
- Provide info for the user to change run-level user back to ```root``` for required applications? (the whole point of doing this was to improve security, with lack of Samba access being the downside, due to inability to set folder specific permissions)

MichaIng, 2018-09-16:
@Fourdee 
Check: https://dietpi.com/phpbb/viewtopic.php?f=11&t=4595&start=10#p14439
- Works well here 😀.

Fourdee, 2018-09-18:
@MichaIng 

> dietpi:dietpi 770 should allow r/w access for Sonarr and such, as these software titles run as their own user, but as dietpi group.

Legend 👍 

Yep, I'll send a commit to make the changes, will run some tests aswell.

Fourdee, 2018-09-18:
Mounts root with:
```
					G_ERROR_HANDLER_INFO_ONLY=1 G_RUN_CMD mount -t cifs -o username="$samba_clientuser",password="$samba_clientpassword",uid=dietpi,gid=dietpi,file_mode=0770,dir_mode=0770,vers=${acifs_versions[$i]} //"$samba_clientname"/"$samba_clientshare" "$samba_fp_mount_target"
```

umount and remount required to apply 😕

Fourdee, 2018-09-18:
Testing required:
- Samba as userdata dir
- Sonarr/radarr etc

Fourdee, 2018-09-18:
Offtopic issue with RockPro64:
[image]

Fourdee, 2018-09-18:
- 🈯️ Samba is completed
- NFS, needs research/testing before we/if implement.

Fourdee, 2018-09-30:
🈯️ NFS supports permissions, at least on a Linux share:
```
root@DietPi:~# ls -lha /mnt/nfs_client/
total 32K
drwxrwxr-x 8 dietpi   dietpi   4.0K Sep 30 18:49 .
drwxr-xr-x 7 root     root     4.0K Sep 30 19:30 ..
drwxrwxr-x 2 dietpi   dietpi   4.0K Sep 19 14:12 downloads
drwxrwxr-x 2 dietpi   dietpi   4.0K Sep 19 14:12 Music
drwxrwx--- 6 statd         112 4.0K Sep 30 19:30 mysql
drwxrwx--- 5 www-data www-data 4.0K Sep 30 18:55 nextcloud_data
drwxrwxr-x 2 dietpi   dietpi   4.0K Sep 19 14:12 Pictures
drwxrwxr-x 2 dietpi   dietpi   4.0K Sep 19 14:12 Video
```

Fourdee, 2018-09-30:
🈴 Samba does not
```
root@DietPi:~# ls -lha /mnt/samba
total 4.0K
drwxrwx--- 2 dietpi dietpi    0 Sep 30 18:49 .
drwxr-xr-x 7 root   root   4.0K Sep 30 19:35 ..
drwxrwx--- 2 dietpi dietpi    0 Sep 19 14:12 downloads
drwxrwx--- 2 dietpi dietpi    0 Sep 19 14:12 Music
drwxrwx--- 2 dietpi dietpi    0 Sep 30 19:35 mysql
drwxrwx--- 2 dietpi dietpi    0 Sep 30 18:55 nextcloud_data
drwxrwx--- 2 dietpi dietpi    0 Sep 19 14:12 Pictures
drwxrwx--- 2 dietpi dietpi    0 Sep 19 14:12 Video
```

However, now mounts with ```dietpi``` user and group, so in theory, userdata applications "should" work, however, this needs to be manually set/tweaked by the user, as it will fail the permissions test during userdata move as of v6.17.

Completed.

closed 2018-09-30 by Fourdee

Dr0bac, 2018-10-02:
@Fourdee i have the same problem with radarr not importing movies but my hdd is external hdd connected directly to rpi3, it says in radarr logs:"
Couldn't import movie /mnt/dietpi_userdata/downloads/Avatar: Acess to the path "/mnt/dietpi_userdata/Plex/Movies/Avatar" is denied..
Then i changed user of /mnt/dietpi_userdata/Plex/movies to radarr:root and now it radarr logs i get
Couldn't import movie /mnt/dietpi_userdata/downloads/Avatar: Acess to the path is denied..
I tried adding radarr user to root group but nothing changed
usermod -a -G root radarr
All my folders in Plex folders have permissions drwxrwxr-x and plex was main user, so i can delete movies from plex menu..

MichaIng, 2018-10-02:
@Dr0bac 
Thanks for your report. But this is a different issue. So not NFS or Samba is the problem in your case, but plex+radarr cross access.

So I guess, the files in flex folder are owned by plex:plex?
To allow access to most other software, installed via dietpi-software, you need to add radarr user to plex group and add plex as well to dietpi group, to allow access of plex to download and other folders as well.

I link the issue to: https://github.com/Fourdee/DietPi/issues/350#issuecomment-423763518
- Fits better there. Like transmission, plexmediaserver is a APT install that brings its own service. We can make the service run as `dietpi` group, which allows access to dietpi_userdata/*, and hope that this also leads to files being created with dietpi group. Or we add a `usermod -a -G` step to dietpi-software to add each software run user to each others group, where cross access is needed. This prevents the need to edit the APT installed service file and allows APT to patch it in case.

Dr0bac, 2018-10-02:
Yes, different problem, but it started recently i guess when 6.14 update was installed.. no it's not plex:plex.. it's plex:root.. and it worked like that perfectly last 6 months..so i guess plex and radarr are in root group? So if radarr don't have permissions to download folder i must add it to dietpi group, not root group?

MichaIng, 2018-10-03:
@Dr0bac 
> plex:root

🤔 Okay this is bad. Hmm I would never want to add software run users to root group actually. Bad repo package default then... I hope for our fix in DietPi install script we can change it the way that files are created as plex:plex or plex:dietpi.
So yeah, in this case you have no other chance then adding radarr user to root group.

> i guess when 6.14 update was installed

It should have worked until v6.12: https://github.com/Fourdee/DietPi/pull/1938
- We changed most software titles run with their own users instead of all as root, for security reasons.
- But indeed that made cross software access quite complicated.

> So if radarr don't have permissions to download folder i must add it to dietpi group, not root group?

Radarr does already run as dietpi group, so should have access to download folder actually.

Verify:
`cat /etc/systemd/system/radarr.service`
- Should show User=radarr and Group=dietpi
`ls -al /mnt/dietpi_userdata/downloads`
- Should be `dietpi:dietpi 775`
With radarr should have access 🤔.

Dr0bac, 2018-10-03:
@Michalng
It's my fault that media folders are plex:root. I saw somewhere i guess on reddit that plex must be owner of media files so i can delete movies with it..also plex plugin subzero for subtitles can't work without changing permissions..
I tried first command and i got like you said
User=radarr
Group=dietpi
After second command i saw that download folder owner is dietpi:dietpi, but downloaded movie folders are under root:root.. and permissions are different..maybe that is the reason..who create that folders, transmission or radarr?
[image]

MichaIng, 2018-10-06:
@Dr0bac 
Whole steps:
```
cat << _EOF_ > /etc/systemd/system/transmission-daemon.service
[Unit]
Description=Transmission BitTorrent Daemon
After=network.target

[Service]
User=debian-transmission
Group=dietpi
Type=notify
ExecStart=/usr/bin/transmission-daemon -f --log-error
ExecStop=/bin/kill -s STOP $MAINPID
ExecReload=/bin/kill -s HUP $MAINPID

[Install]
WantedBy=multi-user.target
_EOF_
G_CONFIG_INJECT '\"umask\":' '\"umask\": 7,' /etc/transmission-daemon/settings.json
```

I will also test now myself 🙂.

MichaIng, 2018-10-06:
PR up for changing Plex + Transmission run as group "dietpi": https://github.com/Fourdee/DietPi/pull/2117

Dr0bac, 2018-11-03:
@MichaIng 
Hey thanks for fixing that permission bug in new dietpi update..now movies are imported nice in plex folder..after update i changed plex movies folders to be plex:dietpi, and also in radarr permission setings i edited so new movies are created with plex:dietpi..previously
It was plex:root

but i have another issue now..
For few days i noticed that subzero plugin that download subtitles for plex did not download subtitles for recent movies i downloaded.si i went to putty to see permissions..and something is not right..radarr is setting strange permissions..look at pictures
[image]

[image]

[image]

New movie folders are created as radarr:dietpi and movies are moved as root:root

Dr0bac, 2018-11-03:
@MichaIng that first picture with permissions is from radarr settings..before, all folders and movie files were created with permissions that was in this settings..before i did put in this settings user plex group root..now it seems that this settings are ignored

MichaIng, 2018-11-03:
@Dr0bac 
Does Radarr actually move or copy the Movies from Transmission download folder? So do they still exist in downloads as well? Would be not ideal due to doubled disk size consumption, but I remember this being the case when I did tests during this permissions fixes...

Actually Radarr might not be always able to adjust the permissions. It runs as radarr:dietpi (check via: `cat /etc/systemd/system/radarr.service`) and as such it can only change permissions of files owned by radarr. If it moves movie files from Transmission downloads, then this is the case.

> subzero plugin that download subtitles for plex did not download subtitles for recent movies i downloaded

Does it save the subtitles within the move container file (thus need write access) or separately? Plex runs as dietpi group so should have access to all files, but not sure if this is the case as well for plugins. Do they run as separate process or inside Plex main process (`htop`)?

Dr0bac, 2018-11-03:
Yes radarr only move video file ( i know, because file is moved quickly, in seconds) and after that radarr delete download move folder and leftovers (like nfo, poster and similar)

Yes, subtitle is stored in same folder where movie is.. in htop all plugins are plex users..i think movie file must also be in plex group so subzero can work..i will try now to change movie file to be radarr:dietpi like folder..and i will check again will it work

MichaIng, 2018-11-03:
@Dr0bac 
> Yes radarr only move video file

Okay that is great in general. But then it makes sense that Radarr cannot change the files permissions, since they are owned by root (in your case) respectively transmission-daemon on current fresh install.

Transmission downloads allow group write access as desired. So Plex needs dietpi group permissions.
Please assure this is the case. On current install script we do:
`echo -e '[Service]\nGroup=dietpi' > /etc/systemd/system/plexmediaserver.service.d/dietpi-group.conf`

In case somehow the plugin sub processes do not take the dietpi group permissions from main systemd unit, you could try `usermod -a -G dietpi plex`

Dr0bac, 2018-11-03:
@MichaIng you were right.. ichanged permissions to 0775/0664 and i see now that permissions are changed from drwxr-xr-x to drwxrwxr-x.. and now subzero works :)

One thing is strange.. before when radarr was root he did create those movie folders and movie files like i wrote in permissions plex:root 0755/0644.. now he lost that ability.. iguess before i created folders with plex owner so i didnt need group write access for subzero to work

MichaIng, 2018-11-03:
@Dr0bac 
Jep as said, since Radarr runs as radarr user now, it can only change permissions of it's own files, not those which are owned by other users. `root` of course can chown every file, so could Radarr when running as root.

Great it works now and thanks for feedback. I will open a new issue to configure sonarr/radarr/lidarr by default using this permissions.

MichaIng, 2018-11-04:
@Dr0bac 
Theoretically you can move it back by removing the `-data=$G_FP_DIETPI_USERDATA/sonarr` option from `/etc/systemd/system/sonarr.service` but this includes all data.

Also having constant I/O to the SDcard increases wear and leads to it breaking faster. I would always want to have as much I/O on external drive instead of SDcard! If there is some software flooding the log that the drive can't sleep, then better take care that software, resolve the issue or reduce logging verbosity, instead of moving this contant I/O to SDcard.

Moving logging to RAM of course would be best. But you don't want other data there. So we would need to find a way to move sonarr logs only to e.g. dietpi-ramlog (/var/log), but keeping other data inside dietpi_userdata 🤔.

Dr0bac, 2018-11-04:
@MichaIng 
Someone asked the same question on reddit and someone from sonarr team answered this..

https://www.google.rs/amp/s/amp.reddit.com/r/sonarr/comments/7vn9c3/how_to_disable_logging/

Is it helpful for you?

So to bring back appdata folder to sd card i need to delete that line
-data=$G_FP_DIETPI_USERDATA/sonarr
From
/etc/systemd/system/sonarr.service
And restart raspberry and everything will be like before?

MichaIng, 2018-11-04:
@Dr0bac 
Move content of /mnt/dietpi_userdata/sonarr to default location first, to not loose config/data. I believe it's /root/.<something>?
Jep the link supports my assumption. So there seems to be no possibility to change the logging location only. Also logging goes to database as well, okay.

**But really, I absolutely not recommend this and future dietpi-updates may break it/require to reapply. It is definitely better to have an external drive running through the night then to have an SDcard being constantly written to. Fix the logging issue, change logging settings, do not break your SDcard and have a full system+data loss instead!**

Within the link an SSD is the assumption. This of course would be a proper solution as well, if for some reason you need that much logging. But SDcard is no option from my point of view.

Wait, a nice solution actually would be to symlink the log file(s) themself to /var/log (with dietpi-ramlog)? Not sure about the remaining database writes. If really all log is doubled to database + logfile then the issue remains of course, but this is not explicitly mentioned, actually.

Dr0bac, 2018-11-04:
@MichaIng 

https://github.com/Sonarr/Sonarr/wiki/Log-Files

On sonarr wiki it is written:

You can clear log files and the logs database directly from the UI, under System -> Logs -> Files and System -> Logs respectively.

So sonarr and radarr are storing logs in txt file and db file..how can we symlink them to var/log?
In appdata folder there are several db files but i can look in date modified times to see which db is for logs, to compare with txt file..