MichaIng/DietPi #2069: Aria2 web interface is showing 403 forbidden error in v6.14 RPi 3B+

sureshachary, 2018-09-12:
## Creating a bug report/issue:

#### Required Information:
- DietPi version	| `paste -sd '.' /DietPi/dietpi/.version`
- Distro version	| `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
- Kernel version	| `uname -a`
- SBC device		| `echo $G_HW_MODEL_DESCRIPTION` or (EG: RPi3)
- Power supply used	| (EG: 5V 1A RAVpower)
- SDcard used		| (EG: SanDisk ultra)

#### Additional Information (if applicable):
- Software title	| (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?
- `dietpi-bugreport` ID

#### Steps to reproduce:

1. ...
2. ...

#### Expected behaviour:

- ...

#### Actual behaviour:

- ...

#### Extra details:

- ...

MichaIng, 2018-09-13:
@sureshachary 
Great to hear. I reopened to not forget fixing this. We try to apply/setup the WebUI the way, that you can access again via `<IP/domain>/aria2`.

Perhaps install size can be reduced by downloading the docs sub directory only and leave out the others.

Also at least we can think about using the additional methods, e.g. allow users to decide whether they want to use it via usual webserver or via NodeJS. The latter could be beneficial is either you have no other use for the webserver or you already have other software that require NodeJS.

sureshachary, 2018-09-13:
Okay

Fourdee, 2018-09-18:
Symlink not compatible:
```
ln -sf /var/www/aria2/docs/index.html /var/www/aria2/index.html
```

[image]

---

This appears to work 🈯️ 
mv /var/www/aria2/docs/* /var/www/aria2/

However, if they have any matching file names added in the future, will break installation.

Therefore, for now, i'll simply update the online doc to reflect the new URL.
🈯️ https://dietpi.com/phpbb/viewtopic.php?f=8&t=5&p=6177#p6177

Completed.

Fourdee, 2018-09-18:
Delete parent dir and move docs over
```
mv /var/www/aria2/docs /tmp/
rm -R /var/www/aria2/*
mv /tmp/docs/* /var/www/aria2/
rm -R /tmp/docs
```

Intresting, this works 🈯️

closed 2018-09-18 by Fourdee