MichaIng/DietPi #2070: Tinkerboard wifi not working (again)

mukamuk1, 2018-09-12:
## Creating a bug report/issue:
wifi is not working when using dietpi.txt.
when using dietpi-config adapter is enabled. It finds network but not getting IP from DHCP.

#### Required Information:
- DietPi version	| 6.15
- Distro version	| 9.5
- Kernel version	| Linux TinkerBox 4.4.71+ #1 SMP Thu Feb 22 18:23:31 CST 2018 armv7l GNU/Linux
- SBC device		| Asus Tinker Board (armv7l)
- Power supply used	| X
- SDcard used		| X

#### Steps to reproduce:

1. Fresh dietpi install
2. dietpi-config-->wifi
3. insert network details
4. wifi works once
5. reboot
6. no wifi

#### Expected behaviour:
Wifi should work

#### Actual behaviour:
wifi is not working

#### Extra details:
Might be same issue: https://dietpi.com/phpbb/viewtopic.php?t=3064

mukamuk1, 2018-09-13:
Issue might be missing net-tools package. Because during attempt to connect wifi I could see error message of missing ifconfig. After installing net-tools wifi connection works just fine.

MichaIng, 2018-09-13:
@mukamuk1 
Thanks for your report.

Related to: https://github.com/Fourdee/DietPi/issues/2071 ?

Could you as well try to move the file mentioned in the other thread out and check if it's working then?

Hmm but you say installing net-tools only solves the issue? This should be not needed anymore. Nowadays `ip` is used to check and apply interface settings. Can you identify which software throws the error? Perhaps some outdated network tool from the TimkerOS?

Fourdee, 2018-09-13:
@MichaIng 

```
root@DietPi:~# dpkg -S /etc/network/if-pre-up.d/wireless-tools
wireless-tools: /etc/network/if-pre-up.d/wireless-tools
```

```
root@DietPi:~# cat /etc/network/if-pre-up.d/wireless-tools
#!/bin/sh

IWCONFIG=/sbin/iwconfig

if [ ! -x $IWCONFIG ]; then
  exit 0
fi

# check if this is a 802.11 device we're supposed to be effecting
case "${IF_WIRELESS:-enable}" in
        wireless-tools|iwconfig)
                # *we* and not some other 802.11 tool should be used
        ;;
        true|yes|enable|1)
                # 802.11 should be used on this device, check for extensions
                $IWCONFIG $IFACE >/dev/null 2>&1 || exit 0
        ;;
        *)
                exit 0
        ;;
esac

# The wireless driver madness:
#
# - Some drivers want everything to be configured before bringing the interface up
# - Some drivers want everything to be configured after bringing the interface up
# - Some drivers want some parameters before, other parameters after bringing the interface up
#
# So, we try to set every parameter when the interface is still down,
# and remember which ones failed to be configured properly.
# If some failed, we bring the interface up, and try the failed ones again.

apply_settings()
{
        if [ -n "$IF_WIRELESS_SENS" ]; then
                $IWCONFIG "$IFACE" sens $IF_WIRELESS_SENS && IF_WIRELESS_SENS= || FAIL=true
        fi

        if [ -n "$IF_WIRELESS_MODE" ]; then
                $IWCONFIG "$IFACE" mode $IF_WIRELESS_MODE && IF_WIRELESS_MODE= || FAIL=true
        fi

        if [ -n "$IF_WIRELESS_AP" ]; then
                $IWCONFIG "$IFACE" ap $IF_WIRELESS_AP && IF_WIRELESS_AP= || FAIL=true
        fi

        if [ -n "$IF_WIRELESS_RATE" ]; then
                $IWCONFIG "$IFACE" rate $IF_WIRELESS_RATE && IF_WIRELESS_RATE= || FAIL=true
        fi

        if [ -n "$IF_WIRELESS_RTS" ]; then
                $IWCONFIG "$IFACE" rts $IF_WIRELESS_RTS && IF_WIRELESS_RTS= || FAIL=true
        fi

        if [ -n "$IF_WIRELESS_FRAG" ]; then
                $IWCONFIG "$IFACE" frag $IF_WIRELESS_FRAG && IF_WIRELESS_FRAG= || FAIL=true
        fi

        if [ -n "$IF_WIRELESS_POWER" ]; then
                $IWCONFIG "$IFACE" power $IF_WIRELESS_POWER && IF_WIRELESS_POWER= || FAIL=true
        fi

        if [ -n "$IF_WIRELESS_POWERPERIOD" ]; then
                $IWCONFIG "$IFACE" power period $IF_WIRELESS_POWERPERIOD && IF_WIRELESS_POWERPERIOD= || FAIL=true
        fi

        if [ -n "$IF_WIRELESS_POWERTIMEOUT" ]; then
                $IWCONFIG "$IFACE" power timeout $IF_WIRELESS_POWERTIMEOUT && IF_WIRELESS_POWERTIMEOUT= || FAIL=true
        fi

        if [ -n "$IF_WIRELESS_TXPOWER" ]; then
                $IWCONFIG "$IFACE" txpower $IF_WIRELESS_TXPOWER && IF_WIRELESS_TXPOWER= || FAIL=true
        fi

        if [ -n "$IF_WIRELESS_RETRY" ]; then
                $IWCONFIG "$IFACE" retry $IF_WIRELESS_RETRY && IF_WIRELESS_RETRY= || FAIL=true
        fi

        if [ -n "$IF_WIRELESS_ENC" ]; then
                eval $IWCONFIG "$IFACE" enc $IF_WIRELESS_ENC && IF_WIRELESS_ENC= || FAIL=true
        fi

        if [ -n "$IF_WIRELESS_DEFAULTKEY" ]; then
                $IWCONFIG "$IFACE" key ["$IF_WIRELESS_DEFAULTKEY"] && IF_WIRELESS_DEFAULTKEY= || FAIL=true
        fi

        if [ -n "$IF_WIRELESS_KEYMODE" ]; then
                $IWCONFIG "$IFACE" key "$IF_WIRELESS_KEYMODE" && IF_WIRELESS_KEYMODE= || FAIL=true
        fi

        if [ -n "$IF_WIRELESS_KEY" ]; then
                eval $IWCONFIG "$IFACE" key $IF_WIRELESS_KEY && IF_WIRELESS_KEY= || FAIL=true
        fi

        if [ -n "$IF_WIRELESS_KEY1" ]; then
                $IWCONFIG "$IFACE" key [1] "$IF_WIRELESS_KEY1" && IF_WIRELESS_KEY1= || FAIL=true
        fi

        if [ -n "$IF_WIRELESS_KEY2" ]; then
                $IWCONFIG "$IFACE" key [2] "$IF_WIRELESS_KEY2" && IF_WIRELESS_KEY2= || FAIL=true
        fi

        if [ -n "$IF_WIRELESS_KEY3" ]; then
                $IWCONFIG "$IFACE" key [3] "$IF_WIRELESS_KEY3" && IF_WIRELESS_KEY3= || FAIL=true
        fi

        if [ -n "$IF_WIRELESS_KEY4" ]; then
                $IWCONFIG "$IFACE" key [4] "$IF_WIRELESS_KEY4" && IF_WIRELESS_KEY4= || FAIL=true
        fi

        if [ -n "$IF_WIRELESS_FREQ" ]; then
                $IWCONFIG "$IFACE" freq $IF_WIRELESS_FREQ && IF_WIRELESS_FREQ= || FAIL=true
        fi

        if [ -n "$IF_WIRELESS_CHANNEL" ]; then
                $IWCONFIG "$IFACE" channel $IF_WIRELESS_CHANNEL && IF_WIRELESS_CHANNEL= || FAIL=true
        fi

        if [ -n "$IF_WIRELESS_NICK" ]; then
                $IWCONFIG "$IFACE" nick "$IF_WIRELESS_NICK" && IF_WIRELESS_NICK= || FAIL=true
        fi

        if [ -n "$IF_WIRELESS_NWID" ]; then
                $IWCONFIG "$IFACE" nwid "$IF_WIRELESS_NWID" && IF_WIRELESS_NWID= || FAIL=true
        fi

        if [ -n "$IF_WIRELESS_ESSID" ]; then
                $IWCONFIG "$IFACE" essid "$IF_WIRELESS_ESSID" && IF_WIRELESS_ESSID= || FAIL=true
        fi

        if [ -n "$IF_WIRELESS_COMMIT" ]; then
                $IWCONFIG "$IFACE" commit && IF_WIRELESS_COMMIT= || FAIL=true
        fi
}

FAIL=
apply_settings 2>/dev/null

if [ -n "$FAIL" ]; then
        FAIL=
        ifconfig "$IFACE" up
        apply_settings
fi
```

Fourdee, 2018-09-13:
Always applies this SSID:
```
[  OK  ] DietPi-Config | Desired setting in /DietPi/dietpi.txt was already set: AUTO_SETUP_NET_WIFI_SSID=MySuperDooperWiFi
```

Fourdee, 2018-09-13:
@mukamuk1 

I'll update the image, please bear with us, will post when its available including the above fix.

Fourdee, 2018-09-13:
Issue effects all recently updated images: 
- https://github.com/Fourdee/DietPi/issues/2026

I won't be redoing them all, due to lack of time. However, i'll redo the popular ones.

Workaround is to either:
- Initially connect with Ethernet, then set up WiFi
- Use DietPi-Automation, to set WiFi creds before 1st boot: https://dietpi.com/phpbb/viewtopic.php?f=8&t=273#p1140

MichaIng, 2018-09-13:
@Fourdee 
I don't get it. Maybe I need to finish the DietPi mug of coffee first 🤣.
- You just changed to method of offering and applying the SSIDs from scan? Instead of writing scan output to file and filling content into an array to loop through, now it is directly looped through the scan output, which is indeed better coding.
- But the $WIFI_SSID value afterwards should be the same 🤔?
- Also the method worked in previous version?

My goodness, took a time until I understand the purpose the the pre-up script:
- Some adapters need to have the interface up, before applying WiFi settings (via iwconfig) works, others allow or need the interface to be still down.
- So the script first tries to apply the settings pre-up, then if this fails, first brings the interface up via ifconfig, before retrying to apply settings.
- It is indeed still an issue (from my point of view), that it uses ifconfig at all, without having the package as dependency:
```
root@VM-Stretch:~# apt depends wireless-tools
wireless-tools
  Depends: libc6 (>= 2.14)
  Depends: libiw30 (>= 30~pre1)
```
- And of course net-tools are widely known/marked as obsolete/outdated: https://linux.die.net/man/8/ifconfig
> This program is obsolete! For replacement check ip addr and ip link. For statistics use ip -s link.

Fourdee, 2018-09-13:
@MichaIng 

> But the $WIFI_SSID value afterwards should be the same

Yep, still unsure of the cause, however, the rewrite has resolved the issue in testing. I need to do more testing regardless.

> Maybe I need to finish the DietPi mug of coffee first 🤣.

Same here 👍

Fourdee, 2018-09-13:
Notes:
- 🈯️ WiFi working fine with tinkerOS 2.0.7 and updated ```dietpi-config```. 
- 🈯️ Also tested after reboot, still fine. 
- Unable to replicate the previous issue with re-written dietpi-config code update, unsure of cause at this time.

Fourdee, 2018-09-13:
@MichaIng 

> Can we edit the script to use ifup "$IFACE" instead? Or is it just for Ethernet interfaces? In case no iproute2 needed, but ifupdown.

We could edit script, i'd rather not, but seems until Debian can resolve this, we need to.

Fourdee, 2018-09-13:
@MichaIng 

Reboot with WiFi only, however, WiFi is connected
```
root@DietPi:~#systemctl status networking -l
● networking.service - Raise network interfaces
   Loaded: loaded (/lib/systemd/system/networking.service; enabled; vendor prese
t: enabled)
   Active: active (exited) since Thu 2018-09-13 20:57:23 BST; 54s ago
     Docs: man:interfaces(5)
  Process: 643 ExecStart=/sbin/ifup -a --read-environment (code=exited, status=0
/SUCCESS)
  Process: 639 ExecStartPre=/bin/sh -c [ "$CONFIGURE_INTERFACES" != "no" ] && [
-n "$(ifquery --read-environment --list --exclude=lo)" ] && udevadm settle
(code=exited, status=1/FAILURE)
 Main PID: 643 (code=exited, status=0/SUCCESS)
   CGroup: /system.slice/networking.service
```

```
root@DietPi:~# dpkg -l | grep net
ii  dirmngr                   2.1.18-8~deb9u2                armhf        GNU privacy guard - network certificate management service
ii  ethtool                   1:4.8-1+b1                     armhf        display or change Ethernet device settings
ii  ifupdown                  0.8.19                         armhf        high level tools to configure network interfaces
ii  iproute2                  4.9.0-1+deb9u1                 armhf        networking and traffic control tools
ii  iputils-ping              3:20161105-1                   armhf        Tools to test the reachability of network hosts
ii  libip4tc0:armhf           1.6.0+snapshot20161117-6       armhf        netfilter libip4tc library
ii  libnettle6:armhf          3.3-1+b2                       armhf        low level cryptographic library (symmetric and one-way cryptos)
ii  libnl-3-200:armhf         3.2.27-2                       armhf        library for dealing with netlink sockets
ii  libnl-genl-3-200:armhf    3.2.27-2                       armhf        library for dealing with netlink sockets - generic netlink
```

Fourdee, 2018-09-13:
@mukamuk1 

I've updated the ASUS TB image, please try it and let us know if this resolves the issue:
https://dietpi.com/downloads/images/DietPi_ASUSTB-ARMv7-Stretch.7z

mukamuk1, 2018-09-14:
@Fourdee Wifi works fine now.

Thx both of u!

Fourdee, 2018-09-15:
@mukamuk1 

Excellent 👍 

I'll mark this as completed. New ticket for resolving external issue:
https://github.com/Fourdee/DietPi/issues/2074

closed 2018-09-15 by Fourdee