MichaIng/DietPi #2072: Home Assistant script failing at python / pyenv stage

narfel, 2018-09-14:
#### Required Information:
- DietPi version	| 6.15
- SBC device		| RPi Zero W
- Power supply used	| Benchpowersupply 5V
- SDcard used		| SanDisk ultra 16G

#### Additional Information (if applicable):
- Software title	| (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated?
fresh
- Can this issue be replicated on a fresh installation of DietPi?
yes
- `dietpi-bugreport` ID
looks pretty similar to build essentials missing issue, but they are installed

#### Steps to reproduce:
dietpi-software -> select to install Home assistant

#### Expected behaviour:
hass should be installed

#### Actual behaviour:
 DietPi-Software
─────────────────────────────────────────────────────
 Mode: Installing Home Assistant: open-source home automation platform

[ INFO ] DietPi-Software | ha_user: homeassistant
[ INFO ] DietPi-Software | ha_userroot: /home/homeassistant
[ INFO ] DietPi-Software | ha_srvroot: /srv/homeassistant
[ INFO ] DietPi-Software | ha_pyenv_activation: export PATH="/home/homeassistant/.pyenv/bin:$PATH"; eval "$(pyenv init -)"; eval "$(pyenv virtualenv-init -)"
[ INFO ] DietPi-Software | ha_python_version: 3.6.3
[  OK  ] DietPi-Software | APT installation for: libssl-dev git cmake libc-ares-dev uuid-dev daemon curl libgnutls28-dev libgnutlsxx28 nmap net-tools sudo libglib2.0-dev libudev-dev swig libssl-dev libusb-1.0-0 gcc libssl-dev libffi-dev libbz2-dev zlib1g-dev libreadline-dev libsqlite3-dev libncurses5-dev libncursesw5-dev, please wait...

[  OK  ] DietPi-Software | G_AGI: libssl-dev git cmake libc-ares-dev uuid-dev daemon curl libgnutls28-dev libgnutlsxx28 nmap net-tools sudo libglib2.0-dev libudev-dev swig libssl-dev libusb-1.0-0 gcc libssl-dev libffi-dev libbz2-dev zlib1g-dev libreadline-dev libsqlite3-dev libncurses5-dev libncursesw5-dev
[  OK  ] DietPi-Software | APT installation for: libmariadbclient-dev, please wait...

[  OK  ] DietPi-Software | G_AGI: libmariadbclient-dev
Adding system user `homeassistant' (UID 108) ...
Adding new user `homeassistant' (UID 108) with group `nogroup' ...
Creating home directory `/home/homeassistant' ...
Adding group `homeassistant' (GID 1001) ...
Done.
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  2194  100  2194    0     0   4418      0 --:--:-- --:--:-- --:--:--  4432
Cloning into '/home/homeassistant/.pyenv'...
remote: Counting objects: 610, done.
remote: Compressing objects: 100% (423/423), done.
remote: Total 610 (delta 293), reused 274 (delta 97), pack-reused 0
Receiving objects: 100% (610/610), 356.94 KiB | 0 bytes/s, done.
Resolving deltas: 100% (293/293), done.
Cloning into '/home/homeassistant/.pyenv/plugins/pyenv-doctor'...
remote: Counting objects: 11, done.
remote: Compressing objects: 100% (8/8), done.
remote: Total 11 (delta 1), reused 6 (delta 1), pack-reused 0
Unpacking objects: 100% (11/11), done.
Cloning into '/home/homeassistant/.pyenv/plugins/pyenv-installer'...
remote: Counting objects: 17, done.
remote: Compressing objects: 100% (15/15), done.
remote: Total 17 (delta 2), reused 9 (delta 0), pack-reused 0
Unpacking objects: 100% (17/17), done.
Cloning into '/home/homeassistant/.pyenv/plugins/pyenv-update'...
remote: Counting objects: 10, done.
remote: Compressing objects: 100% (6/6), done.
remote: Total 10 (delta 1), reused 5 (delta 0), pack-reused 0
Unpacking objects: 100% (10/10), done.
Cloning into '/home/homeassistant/.pyenv/plugins/pyenv-virtualenv'...
remote: Counting objects: 54, done.
remote: Compressing objects: 100% (48/48), done.
remote: Total 54 (delta 10), reused 19 (delta 0), pack-reused 0
Unpacking objects: 100% (54/54), done.
Cloning into '/home/homeassistant/.pyenv/plugins/pyenv-which-ext'...
remote: Counting objects: 10, done.
remote: Compressing objects: 100% (6/6), done.
remote: Total 10 (delta 1), reused 4 (delta 0), pack-reused 0
Unpacking objects: 100% (10/10), done.

WARNING: seems you still have not added 'pyenv' to the load path.

# Load pyenv automatically by adding
# the following to ~/.bash_profile:

export PATH="/home/homeassistant/.pyenv/bin:$PATH"
eval "$(pyenv init -)"
eval "$(pyenv virtualenv-init -)"
Downloading Python-3.6.3.tar.xz...
-> https://www.python.org/ftp/python/3.6.3/Python-3.6.3.tar.xz
Installing Python-3.6.3...

#### Extra details:
I checked build-essentials and python to be installed. Both are present and working.

narfel, 2018-09-14:
Looks like the warning above led me to believe the script was hung while apparently it wasn't. It was just good old patience that was missing :)

MZorzy, 2018-11-27:
Upgraded DietPI to v6.18.4, rebbot, now homeassistand don't start
(worked on 6.17.x) . command line to launch ?

Fourdee, 2018-11-30:
@MZorzy 

Possible python version issue, from what I can translate + google.

Which device are you running (EG: RPi)? I'll try to replicate.

MZorzy, 2018-11-30:
Hi @Fourdee ,
i'm on DietPi_VirtualBox-x86_64-Stretch from your site. 
start as dietpi 6.17, added pihole (it work), reboot, than added home-assistant (it work).
no other software.

after upgrade dietpi from 6.17 to 6.18 pihole working as usual, but home-assistant don't autostart 

de-installed homeassistant. rm /home/homeassistant/.cache, dietpi-software install 157, reboot, now work again.

Fourdee, 2018-12-01:
@MZorzy 

> after upgrade dietpi from 6.17 to 6.18 pihole working as usual, but home-assistant don't autostart

Strange, must of been an APT upgrade that caused the issue.

> de-installed homeassistant. rm /home/homeassistant/.cache, dietpi-software install 157, reboot, now work again.

Ok, we'll try to replicate this on our VM.

Fourdee, 2018-12-01:
Test:
- VB fresh image
```
Full Path         : C:\Users\Administrator\Desktop\DietPi_VirtualBox-x86_64-Stretch\DietPi_VirtualBox-x86_64-Stretch.ova
Modified Time     : 15/11/2018 12:52:55
```
- Do not update to v6.18
- Install Pi-hole + HA
- Update to v6.18 🈯️ | no issues
- Reinstall Python via APT (to test APT update cause) 🈯️  | no issues
```
root@DietPi:~# dpkg --get-selections | grep python | awk '{print $1}'
dh-python
libpython3-stdlib:amd64
libpython3.5-minimal:amd64
libpython3.5-stdlib:amd64
python3
python3-minimal
python3.5
python3.5-minimal
root@DietPi:~# apt-get install --reinstall $(dpkg --get-selections | grep python | awk '{print $1}')
```

@MZorzy 

Was unable to replicate in our testing, unsure as to cause your end. However, as the issue is now resolved on your system, i'll mark this as resolved.

closed 2018-12-01 by Fourdee