MichaIng/DietPi #2074: /etc/network/if-pre-up.d/wireless-tools | requires net-tools

Fourdee, 2018-09-15:
However, not listed as a APT dep:
https://github.com/Fourdee/DietPi/issues/2070#issuecomment-421088364

Appears only required, when all other methods fail, and, should not effect WiFi usage:
https://github.com/Fourdee/DietPi/issues/2070#issuecomment-421058572

MichaIng, 2018-09-16:
Bug solved on current version: https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=908886#10
- Download available on unstable/sid repo so far only: https://packages.debian.org/sid/amd64/wireless-tools/download
- Only change that I could bind is indeed the ifconfig line being replaced by: `/sbin/ip link set dev "$IFACE" up`, which we can do ourself on v6.16 patch.

Fourdee, 2018-09-16:
@MichaIng 

> Bug solved on current version: https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=908886#10

Legend, great work 👍