MichaIng/DietPi #2075: Jessie | DietPi-Preboot causes SystemD to skip networking service

Fourdee, 2018-09-15:
> https://github.com/Fourdee/DietPi/issues/2071#issuecomment-421127903
https://github.com/Fourdee/DietPi/issues/2071#issuecomment-421145760

Non-issue for Stretch+

> re: @MichaIng 
networking wants to run before sysinit.target
But sysinit.target runs before basic.target, which is required by all our services.
So it's: networking -> sysinit -> basic -> preboot -> networking

MichaIng, 2018-09-16:
Manual workaround works well here on Jessie VM: https://dietpi.com/phpbb/viewtopic.php?f=11&t=4726&p=14440#p14440

Fourdee, 2018-09-26:
Note to self:
- Setup Jessie system, investigate issue.

MichaIng, 2018-09-26:
@Fourdee 
I found a perhaps cleaner solution: https://unix.stackexchange.com/a/239978

- Adding `Wants=network-pre.target` to pull in this target
- Adding `Before=network-pre.target` to let it run before all networking services.
- Removing all other network targets+services from `Before=` to remove the dependency loop.

Works well here on Jessie VM, but should be tested on some other Jessie devices.

Fourdee, 2018-09-26:
@MichaIng 

Tweaked the services, appears fine on Stretch:
- before https://dietpi.com/downloads/testing/before.html
- after https://dietpi.com/downloads/testing/after.html

Just need to test Jessie.

MichaIng, 2018-09-26:
Nice tool btw.

Fourdee, 2018-09-26:
@MichaIng 

Went for the workaround for Jessie. Not ideal, however, all our official current images are Stretch.

Re-test using PREP
- 🈯️ Stretch
- 🈯️ Jessie

Completed.

closed 2018-09-26 by Fourdee

MichaIng, 2018-09-26:
@Fourdee 
Okay, so found that basic.target and sysinit.target are by default added as required, which can be overridden via `DefaultDependencies=no`.
But I guess, this would be needed for ramdisk+ramlog as well, to completely resolve the loop. Not sure, if there is any needed service starting between local-fs.target and sysinit.target/basic.target on Jessie.

Why do ramdisk/ramlog actually start before `boot.mount`, when they are actually set to start `After=local-fs.target boot.mount`?? `DietPi.mount` is already done, so /boot definitely mounted. I see `boot.automount` starting at very early stage, maybe that solves/replaces somehow `boot.mount` 🤔.

However, I will test a bid with `DefaultDependencies=no`, but at least we have a workaround now.

sdubrick, 2018-10-07:
I just wanted to mention that I was having issues with my Odroid C1 not connecting eth0 on startup in v6.16, and the workaround didn't fix it.  The changes @Fourdee made DID fix it though, thanks.

MichaIng, 2018-10-07:
@sdubrick 
The `systemctl start networking` inside preboot script was the change that finally fixed it for Jessie systems (G_DISTRO=3), if want does not want to apply all those changes 😉: https://github.com/Fourdee/DietPi/blob/be64a2c84aae9215de1bce6d2684f2d84a1bbd1d/dietpi/preboot#L269-L274