MichaIng/DietPi #2079: DietPi Menu standards

Fourdee, 2018-09-19:
- If the item contains a changeable value, it must contain ```[VALUE]```?
- All menu array 2nd items, must contain ```:``` at start to separate the 1st item?

🈯️ Currently, this looks fine:
[image]
[image]

🈴 But this looks a mess, as the text blends into each other:
[image]
[image]

MichaIng, 2018-10-01:
@Fourdee 
We could make `G_WHIP_INIT` add the beautiful delimiters automatically. So the array only needs to contain the raw text.
Allows us to change it easily for all together, if something beautiful was found.

Fourdee, 2018-10-01:
@MichaIng 

> We could make G_WHIP_INIT add delimiters automatically. So on array creation we only need to add the raw text.

Yep, thought the same already 👍 , but processing on large menu's may be an issue to add and re-shuffle the chars?

MichaIng, 2018-10-01:
@Fourdee 

For `G_WHIP_MENU`:
```
for (i=1;i<${#G_WHIP_MENU_ARRAY[@]};i+2)
do

	G_WHIP_MENU_ARRAY[$i]=": ${G_WHIP_MENU_ARRAY[$i]}"

done
```

For `G_WHIP_CKECKBOX`:
```
for (i=1;i<${#G_WHIP_MENU_ARRAY[@]};i+3)
do

	G_WHIP_MENU_ARRAY[$i]=": ${G_WHIP_MENU_ARRAY[$i]}"

done
```

As very first step, before estimating windows size and such?

Fourdee, 2018-10-07:
@MichaIng 

Nice 👍 

Although my issue is with any performance impact. The RPI 1/Zero is borderline slow with ```dietpi-software``` list generation. Any slower and it will become painful lol.

I think for simplicity and keeping current performance, we simply need to add ```:``` manually to ```G_WHIP``` items as we see them?

MichaIng, 2018-10-07:
@Fourdee 
Okay for now, as mentioned, there seem to be some other parts where we can improve performance. Then we do some tradeoff with the above 😉.

MichaIng, 2018-10-13:
PR to speed up G_WHIP: https://github.com/Fourdee/DietPi/pull/2136

Fourdee, 2018-10-13:
@MichaIng 

Thanks 👍 

Issue with missing text:
https://github.com/Fourdee/DietPi/pull/2136#issuecomment-429524800

Fourdee, 2018-10-13:
New format:
[image]

MichaIng, 2018-10-13:
@Fourdee 
Fixed with: https://github.com/Fourdee/DietPi/commit/6570a9949f84f18d68f8476061f1796814cfb25d

However somehow found another issue:
- $WHIP_MESSAGE is not shown, debug:
```
                        # - Calculate lines required, including addition lines, if the text is longer than X
                        [[ $WHIP_MESSAGE ]] && echo -e "$WHIP_MESSAGE" | while read -r line
                        do

                                lines_required_whip_y=$(( lines_required_whip_y + 1 + ${#line} / ( $WHIP_SIZE_X - $whip_x_internal_margin) ))
                                echo $lines_required_whip_y

                        done
                        echo $lines_required_whip_y

                        #       Process final whiptail size
                        #       Add internal whiptail margins to end line total
                        lines_required_whip_y=$(($lines_required_whip_y+$whip_y_internal_margin))
```
```
root@VM-Stretch:~# G_WHIP_MSG "TEST\nTEST2\nTEST3"
1
2
3
0
```
- `lines_required_whip_y` increase is lost outside the while read loop 🤔. Strange since we used this as well before...

MichaIng, 2018-10-15:
Added to v6.17 milestone, so we not forget it an apply to scripts that we touch anyway:

> New format:
> - ```:``` on all 2nd array items at start
> - ```[ ]``` indicates a changeable option, with current value inside.

MichaIng, 2018-12-14:
There are now several menus, where the values are shown within the menu entries and above in whiptail text again.
- When adding the value to the menu entry, we should remove the related text above.

From code side, with last PRs I added real linebreaks to long whiptail texts (replacing `\n`) to make it more readable and reduce the need to scroll left and right within code.

Fourdee, 2018-12-15:
@MichaIng 

> There are now several menus, where the values are shown within the menu entries and above in whiptail text again: When adding the value to the menu entry, we should remove the related text above.

Yep, agree 👍

Fourdee, 2019-01-28:
Will mark as completed:
- I believe all menus now have the new standard ```'X' ': X'```
- Will further reduce double current value print out in ```G_WHIP_MENU``` when I come across them 😄

closed 2019-01-28 by Fourdee

MichaIng, 2019-01-28:
@Fourdee 
I think I saw one missing case, cloudshell or another misc program. Will check quickly.

MichaIng, 2019-01-28:
Done: https://github.com/Fourdee/DietPi/commit/b6635e0ca426a2c20e3169ddd192bf8c58a2a3ed
- Applied to all menus, besides where second entry is "Seconds" or "Hours"