MichaIng/DietPi #2081: Show changelog url along with update message on login.

msdos, 2018-09-19:
## Creating a feature request:

#### Is your feature request related to a problem? Please describe:

- No.

#### Describe the solution you'd like:

- I just logged in and received this message: `dietpi-update    = Run now to update DietPi (from v6.15 to v6.16).` Since I have a really customized installation, I would like to read the changelog to know if I manual action will be needed before updating.

#### Describe alternatives you've considered:

- The alternative is going to google and searching for this changelog, because I always forget the exact place to search for this. I believe more people have the same feeling.

#### Additional context:

- Just change the message from `dietpi-update    = Run now to update DietPi (from v6.15 to v6.16).` to `dietpi-update    = Run now to update DietPi (from v6.15 to v6.16). Changelog: https://raw.githubusercontent.com/Fourdee/DietPi/master/CHANGELOG.txt`. The changelog url https://raw.githubusercontent.com/Fourdee/DietPi/master/CHANGELOG.txt can even receive a shortened url from you so you can track who is really clicking on it: since the url is pointed to master, you wouldn't even need to change it (but it's up to you if you want to point this url to a specific commit).

yumiris, 2018-09-20:
To expand on the idea, what if `dietpi-update` could show the changelog when asking the user to confirm their update? Or maybe a new `dietpi-changelog` command, which in its most basic form would do a `curl` request to the `CHANGELOG.txt` file? 🤔

msdos, 2018-09-20:
@yumiris I think it gets too complicated for a first implementation. The idea here (specially if using a shortened url with statistics) is to see if people will really use this feature. I think it can be evolved to your idea if more people start using the url.

yumiris, 2018-09-20:
Haha, that's true 😛 - I was thinking it from the perspective that URLs might be difficult for users to manually type (and therefore result in lower statistics). Although using a customised URL (e.g. `short.url/dietpi-changelog`) might make it easier.

msdos, 2018-09-20:
People can copy the url using a mouse. Since it's a message shown in a terminal session, I think it's not for the complete newbie and this won't be a problem.

Can be `short.url/dpchangelog` or even something along the lines of `short.url/HjYeV` to be even shorter, but I leave that to DietPi maintainers.

I knew that more people would be interested :laughing:

yumiris, 2018-09-20:
That's actually a very valid point. Throughout this entire conversation, I completely forgot that desktop environments are actually a thing (even though I use both daily!).

Yup, keeping it short and succinct should work wonders. 😛

MichaIng, 2018-09-20:
@msdos @yumiris 
Actually, since the changelog is already downloaded with the update archive (https://github.com/Fourdee/DietPi/blob/master/CHANGELOG.txt), it would be easiest to offer opening this file into a pager/whiptail dialog?
... ah but you want to see them before downloading the update of course.

Still, since the file is a tiny text file, we could offer to download+view the changelog within die `dietpi-update` dialog, instead of just [Yes] [No]?
Or would you prefer a URL, e.g. in case of small server screens/for comfortable desktop view?
Of course also both solution together are possible.

msdos, 2018-09-20:
I don't know how a pager/whiptail dialog with `Do you want to download the changelog for this version at https://github.com/Fourdee/DietPi/blob/master/CHANGELOG.txt?` would work. I'm just afraid of adding another click to the update action and make it more complicated, that's why I just gave the url suggestion. Maybe when updating, you directly shows what comes from that version. A `csplit` in the changelog take everything before `--------------------------------------------------` and show in the dialog before the user saying [Yes] or [No].

yumiris, 2018-09-20:
The changelog could be shown in this section rather than a separate one:

[image]

More information, but the number of clicks is kept the same. In case of long changelogs, maybe use a scrolling view:

```
+--------------------------------| DietPi-Update |------------------------------+
|                                                                               |
| Update available   : https://goo.gl/G9ikqn                                    |
| Installed version  : 6.15                                                     |
| Latest version     : 6.16                                                     |
|                                                                               |
| ------------------------------------Notice----------------------------------- |
| The benefit of DietPi is we use standard linux configurations and commands... |
| ----------------------------------------------------------------------------- |
|                                                                               |
| ----------------------------------Change Log--------------------------------- |
| Changes / Improvements / Optimizations:                                       |
| DietPi-Config | WiFi: Added support for applying up-to 5 SSIDs [...]          |
|                                                                               |
|                                 <Scroll Down>                                 |
| ----------------------------------------------------------------------------- |
|                                                                               |
| Do you wish to continue and update DietPi to v6.16?                           |
|                                                                               |
|                 <Ok>                                       <Exit>             |
|                                                                               |
+-------------------------------------------------------------------------------+
```

Similar to the software selection list in `dietpi-software`. 😛

Fourdee, 2018-09-21:
Great idea, thanks for sharing @msdos @yumiris 👍 

If we are to use existing functions, we'd need to do ```G_WHIP_MENU``` with:
- Update now
- Read changelog
- OK/Cancel button

Then read changlog will use ```G_WHIP_SCROLLBOX```

---

To support scrolling with ```G_WHIP_YESNO```, needs an additional flag.

Fourdee, 2018-09-21:
```--scolltext``` does not indicate any ability to scroll, tab required to use ability to scroll, best avoided. Menu options should be preference.

[image]

Fourdee, 2018-09-21:
Curl lacks the ability to ignore downloading headers, if the file does not exist on server, always returns 0 due to header DL. wget does not have this issue:
```
root@DietPi:~# curl --header -k -L https://dietpi.com/test
<!DOCTYPE HTML PUBLIC "-//IETF//DTD HTML 2.0//EN">
<html><head>
<title>404 Not Found</title>
</head><body>
<h1>Not Found</h1>
<p>The requested URL /test was not found on this server.</p>
<hr>
<address>Apache/2.4.25 (Debian) Server at dietpi.com Port 443</address>
</body></html>
```

Fourdee, 2018-09-21:
[image]
[image]

Fourdee, 2018-09-21:
@msdos 

>  A csplit in the changelog take everything before -------------------------------------------------- and show in the dialog before the user saying [Yes] or [No].

Good idea 👍 

However, we have alot of users who update from say v6.12 to v6.16, so the whole logfile needs to be shown.

Fourdee, 2018-09-21:
- 🈯️ All options tested with master and testing branches

Completed.

Fourdee, 2018-09-21:
@msdos @yumiris 

Thanks guys, great suggestion 👍, will be available once v6.17 is released, updated, then next updates (due to ```dietpi-update``` not updating till after the update is completed, eg: the next run)

closed 2018-09-21 by Fourdee