MichaIng/DietPi #2082: Simple Minecraft Server

symbios24, 2018-09-20:
Hello, i would like when there is time to create for dietpi a simple minecraft server unlike the gui versions that need a lot of configurations 
its something like the pe/nukkit version but it works on general/official minecraft version
a simple straight loading server without anything to setup.for users that want to start right away.

The best instructions i can find are on this two pages:

https://www.makeuseof.com/tag/setup-minecraft-server-raspberry-pi/

https://pimylifeup.com/raspberry-pi-minecraft-server/

----------------------------------------------------------
Its a server that will use the latest build at this time 1.13.1
With the NoSpawnChunk plugin to save memory
Works on older pi with the right configuration Model B and up and on latest ones 2/3/3B+
and it can start at boot.
------------------------
i can help if you want.

Thanks

MichaIng, 2018-10-14:
@symbios24 
Thanks for your request. However not that it might take a while, as there are other interesting software requests in the pipe.

Also I think we will not implement any of those with v6.17 where we implement a new Beta system that needs to be announced and tested first.

MichaIng, 2018-12-26:
Official link: https://www.spigotmc.org/wiki/about-spigot/

Added to FeatHub, feel free to vote for it: https://feathub.com/Fourdee/DietPi/+15

MichaIng, 2019-02-27:
No votes (more down == up) so far. Will mark as closed for general cleanup. However the FeatHub page stays in place to allow further voting. If it reaches closer to the top we will reopen.

closed 2019-02-27 by MichaIng

ravenclaw900, 2021-01-17:
Just want to point out, PaperMC is an optimized version of SpigotMC.  It can basically run all Spigot plugins and is much more performant.