MichaIng/DietPi #2083: General | Making DietPi releases more stable

MichaIng, 2018-09-20:
Currently we relatively often push new releases with one or more serious bugs, affecting network connectivity, time sync => SSL errors, permissions and such. This is due to a quite fast development with many new features, core improvements and such.
 
Results are hotfix releases, which are not always separated from other new features (currently multiple WiFi SSIDs e.g.) possibly leading to other new bugs and unhappy users.

Reason is clearly that we have on the one hand many great ideas from users and devs about new features and improvements but lack the devs/time and in my case possibilities to test those changes on every device and possible setup.

My suggestion is, even that it is an additional effort and slows down development a bid, to implement an open beta/RC branch.
E.g. we do a moreless monthly release with new features and improvements. But we do a beta/rc release one week earlier, that hopefully many interested users can test, and concentrate our efforts then on fixing bugs that may have been introduced, before merging it into final release.
This would mean e.g. that we develop always in a new "development"/"dev" branch, but backport only bug fixes to the RC branch (could be the testing branch) in regular RCs during the testing week, using subsubversion strings to allow updates for the beta users.

Additional effort is the backports that need to be done, but at least bug fixes are usually small code snippets. And it should clearly increase our release quality, at least if we find enough users that are willing to join the beta/testing branch.

Another topic is bugs, that break APT, time sync and such and by this break the ability to update without manual pre-fixes. Perhaps we can move current emergency update steps into an own small pre-update file/script, that is applied as very first update step, at best available via non-SSL source? But maybe there is an easier solution, as I think APT and SSL are the only reason why this might be needed. In the past we did ATP upgrade before applying the incremental patch file, but that brings other possible issues when we expect current packages for a certain steps.

These are just suggestions and the current ideas and solutions I have in mind. My most important intention is to start discussion/idea collection about how to make DietPi more robust for production systems.

Fourdee, 2018-09-21:
@MichaIng 

> Reason is clearly that we have on the one hand many great ideas from users and devs about new features and improvements but lack the devs/time and in my case possibilities to test those changes on every device and possible setup.
My suggestion is, even that it is an additional effort and slows down development a bid, to implement an open beta/RC branch.

Great idea, agree 👍 

- Master = Stable
- Beta = Pre-release testing, open to all users
- Testing = In dev, potentially unstable

The only issue I see, is after we roll out Testing > Beta and need beta users to update, we'd essentially need to increase the subversion to notify update. Which results in master branch skipping a few versions per release?

> Perhaps we can move current emergency update steps into an own small pre-update file/script, that is applied as very first update step, at best available via non-SSL source? 

Yep, i've been thinking about this for a while, moving EMR patches to a ```patch_prefile```, which runs:
- After 1st initial internet connection test
- Before systemD timesync check

It can then be used for pre-timesync emergency patches.

> at best available via non-SSL source? 

We can use SSL source with simply the ```--no-check-certificate``` option to prevent non-timesync'd system failing. Essentially, we could use Github for ```patch_prefile``` I believe.

>  My most important intention is to start discussion/idea collection about how to make DietPi more robust for production systems.

Yep I agree. However, this will only improve with these new systems, if we have users who are willing to help test.

MichaIng, 2018-09-21:
@Fourdee 
> Beta = Pre-release testing, open to all users
> Testing = In dev, potentially unstable

Beta = testing and testing = dev is perhaps a bid confusing? This is why I thought to use the current "testing" branch for testing/beta and implement a new "development"/"dev" branch for what we currently do on "testing"? But this means that we need to switch e.g. local git branches and such, so when this is too much effort, your suggestion would work as well of course 😉.

> The only issue I see, is after we roll out Testing > Beta and need beta users to update, we'd essentially need to increase the subversion to notify update. Which results in master branch skipping a few versions per release?

I would add sub-subversion for this, e.g. v6.17.0 for initial RC/beta, then v6.17.1 for next etc.

> We can use SSL source with simply the --no-check-certificate option to prevent non-timesync'd system failing. Essentially, we could use Github for patch_prefile I believe.

Ah that's true, the wget download does also work with ignoring untrusted SSL cert, right?

> Yep I agree. However, this will only improve with these new systems, if we have users who are willing to help test.

Jep, I hope we find enough that it is worth it, making some advertising for it on twitter and forum e.g. 😄.

Fourdee, 2018-09-21:
@MichaIng 

>  the wget download does also work with ignoring untrusted SSL cert, right?

Yep, however, using http:// has the same effect. Both are the only option we have pre time sync.

> Jep, I hope we find enough that it is worth it, making some advertising for it on twitter and forum e.g.

Yep 👍 

> This is why I thought to use the current "testing" branch for testing/beta and implement a new "development"/"dev" branch for what we currently do on "testing"?

Yep, probably a better idea, we should rename the testing branch to 'dev'.

Then with the beta branch, we can roll out dev to beta, when everything is completed. This will then allow users to update and test.

> I would add sub-subversion for this, e.g. v6.17.0 for initial RC/beta, then v6.17.1 for next etc.

Yep, this would allow for beta updates from 'dev' branch once we fix issues.

---

Ok so?
- Rename 'testing' to 'dev'
- Create 'beta' branch, which is a copy of 'dev' once we are ready for release
- Allow 'beta' branch to test for X days, before we merge to 'master'
- Use an additional RC version to allow for 'beta' branch to update when 'dev' updates are applied, during the RC stage prior to release.

MichaIng, 2018-09-25:
@Fourdee 
> Yep, however, using http:// has the same effect. Both are the only option we have pre time sync.

Ah this is great. So we can pre-patch everything besides broken network itself 👍.

> Ok so?
> 
> * Rename 'testing' to 'dev'
> * Create 'beta' branch, which is a copy of 'dev' once we are ready for release
> * Allow 'beta' branch to test for X days, before we merge to 'master'
> * Use an additional 'beta' version to allow for 'beta' branch to update when 'dev' updates are applied, during the RC stage
> * 'beta' version ignored on 'master' branch.
> * Release 'beta' to 'master' when stability testing is completed after X days
> * Patch notes:
>   -- Switch 'testing' to 'dev'
> * Web notes:
>   -- Create BETA_BRANCH.md
>   -- rename 'testing' to 'dev'

Sounds great. I am thinking, if we should use sub-sub-versions for beta updates only and simply ignore them on master/release update check, or as well implement them for master releases (for smaller, properly still needed hotfix releases). As long as every system just checks it's own GitHub branch for current server version string, both should work fine, but has impact on patch_file complexity.

Also about sub-sub-version updates:
- If sub-version is not up-to-date, update via patch_file sub-version patches only.
- If sub-version is up-to-date, check for sub-sub-version update and use separate patch file. We then merge this sub-sub-version patches into the regular sub-version patch, so there always only need to be a single sub-sub-version patch_file for the current beta.

As we already touch the updating system:
- Is your plan to never allow updating to new main version (DietPi v7)?
- Personally I would always try to allow updates, since flashing a new image is always a large effort for users.
- Would love to announce the above via DietPi v7.0 to highlight the step to more stability/reliability and hopefully get attention from potential beta testers. This would mean already a first pre-patch task to allow patching main versions.

MichaIng, 2018-10-01:
Further thinking about pre-patch, downloaded via HTTP. Actually I am not too happy with this for security reasons. I know there are many other cases (3rd party software, non-HTTPS APT sources, ...) where simple HTTP is used, but I would like to use HTTPS where we have control.
- Theoretically someone evil could otherwise implement harmful code and end user has no chance to verify it is from us. Maybe cons < pros but just to think about.

Sadly this case HTTPS would break one of the main reasons for pre-patching: time sync or other HTTPS related issues. Maybe we can instead further improve our G_CHECK_URL to check output for HTTPS related errors. If check fails, but `--no-check-certificate` does not, e.g. this would be a clear indication after which we can print some usual checks+solutions.

Also I thought about some kind of (optional) news/announcement inside `dietpi-banner`, or a "New announcement available, call "dietpi-announcement" print, which only appears until the script is called once (and not unchecked by user, dietpi.txt entry, allowed to toggle within the script itself?).
There we could publish some current issues, important bugs and such, possibly based on distro, device and/or installed software of end user. Or simply inform about other interesting things. Stay closer in touch with users, who are open to it.

Fourdee, 2018-10-03:
@MichaIng 

> Sadly this case HTTPS would break one of the main reasons for pre-patching: time sync or other HTTPS related issues. 

Yep, its a catch 22. We need one to do the other.

> Maybe we can instead further improve our G_CHECK_URL to check output for HTTPS related errors. If check fails, but --no-check-certificate does not, e.g. this would be a clear indication after which we can print some usual checks+solutions.

Nice idea 👍 

> Also I thought about some kind of (optional) news/announcement inside dietpi-banner, or a "New announcement available, call "dietpi-announcement" print, which only appears until the script is called once (and not unchecked by user, dietpi.txt entry, allowed to toggle within the script itself?).
There we could publish some current issues, important bugs and such, possibly based on distro, device and/or installed software of end user. Or simply inform about other interesting things. Stay closer in touch with users, who are open to it.

Yep, nice idea 👍 

Although we could simplify this and have a link to https://github.com/Fourdee/DietPi/issues/2093 in the banner, then simply keep the ticket open and nudge v6.16 down after v6.17 release?

Fourdee, 2018-10-04:
Ok in regards to changes:
#### ```testing``` > ```dev```
- v6.17 patch , change ```GITBRANCH=testing``` to ```GITBRANCH=dev```
- Clone ```testing``` to ```dev```
- Now work on ```dev```

Fourdee, 2018-10-04:
@MichaIng 

I've made a start on this: https://github.com/Fourdee/DietPi/issues/2083#issuecomment-426842537

Please do any work on the new 'dev' branch:
https://github.com/Fourdee/DietPi/tree/dev

MichaIng, 2018-10-04:
@Fourdee 
> RC version only used for 'beta' 'dev' branches. Master branch is always X.X.0

Why not keep RC version for release? This would also allow beta users to update to most current stable, when switching branch setting. E.g. they just test 1st RC, then for some reason miss 2. and 3. and release. Then they want to switch to release. But update would now be shown if server version is X.X.0.

Ah but let me test and revisit the code first before I can give qualified feedback 🤣.

Fourdee, 2018-10-04:
@MichaIng 

> Why not keep RC version for release?

To allow us to reset RC versions for each subversion release. As the RC version is used to allow updates for 'beta' branch.

Ah I see what you mean, just let the RC version roll upwards and never reset. Should work fine aswell.

MichaIng, 2018-10-12:
@Fourdee 
Do we want to add branch info to `dietpi-banner`, e.g. into the version string line?
`DietPi v6.17.3 (beta)`/`(release)`/`(MichaIng/custom)`

Fourdee/master could be replaced by "release" and Fourdee/beta by "Beta" and "Fourdee/dev" by "development", but all custom owner/branches leave as "<owner>/<branch>". So users clearly know which DietPi version they use.

Only issue is, that we need to store the git owner+branch separately on update instead of reading from `dietpi.txt`, so the info is always correct, even that users change `dietpi.txt` but did not yet update? On the other hand, why would someone change git branch/user, if not want to do an update afterwards...

Fourdee, 2018-10-13:
@MichaIng 

> Do we want to add branch info to dietpi-banner, e.g. into the version string line?
DietPi v6.17.3 (beta)/(release)/(MichaIng/custom)

I believe for dev/beta we can add additional info. But with master branch, leave as is to keep it clean:
```
 ───────────────────────────────────────
 DietPi     | Update available
 ───────────────────────────────────────
 v6.17.3    | Odroid C2 (aarch64)
 ───────────────────────────────────────
 eth0       | 192.168.0.113
 ───────────────────────────────────────
```

> On the other hand, why would someone change git branch/user, if not want to do an update afterwards...

Exactly 👍. I added a global for branch ```G_GITBRANCH```, we could probably do the same for ```G_GITFORK```.

Performance wise, should be faster to store/read the above info in ```.version``` as dedicated line. Might be worth doing.

MichaIng, 2018-10-13:
@Fourdee 
Great commit! Looks way more consistent like this. Tiny remark:
- Having this branch selection renders adding at least manual `G_GITBRANCH` export to PREP useless, besides one can cancel the whip menu.
- Furthermore we cannot assure that on every `G_GITOWNER` repo these branches are available.
- Easiest would be properly to only ask for this, if `G_GITBRANCH` and `G_GITOWNER` were not exported to script, thus as well only when `G_GITOWNER` default=Fourdee. Otherwise assume, users wants it the way provided by variables.
- More elegant would be a selection menu based on available owners + branches:
  - Scan https://github.com/Fourdee/DietPi/network/members for available owners. Puhh no way too much and most, if not all of them are just left-overs from a PR. Then better: `Fourdee (official)` as owner selection, `Custom (unsupported)` as second option to allow textbox entry.
  - Scan available branches, with, on Fourdee, master, beta and dev at the top: https://github.com/Fourdee/DietPi/branches/all
  - Puhh, but quite complex, perhaps stay with: "Only prompt branch selection, if both, `G_GITBRANCH` and `G_GITOWNER` are not yet available (not exported/added to script).

Found another inconsistency:
- `dietpi-globals` read `G_GITOWNER` and `G_GITBRANCH` from new `.version` now, which is the currently installed version.
- `dietpi-update` however use this values now as well, instead of the ones from `dietpi.txt` which is the desired/target version.
- We should separate this: Use the above globals as "current" info, but let `dietpi-update` read dietpi.txt for "target" branch/owner to use for server URLs.

MichaIng, 2018-10-13:
Shit, accidentally commited to dev instead of PR: https://github.com/Fourdee/DietPi/commit/f62675e0ceaf93d4720db22d2c4b29dddecb0888

However will do some testing with this.

Fourdee, 2018-10-16:
@MichaIng 

> Isn't writing and then reassigning the exactly same variables somehow doubled effort?

Yep, not sure why I did that lol 👍 

> Personally I would reduce export to the vars, where it's really needed:

Agree, will send commit.

Fourdee, 2018-10-22:
Beta system now live, we can mark this as closed.

closed 2018-10-22 by Fourdee