MichaIng/DietPi #2085: Add guide on compiling qBittorrent 4.x on forum entry for qBittorrent.

Millichrome, 2018-09-21:
I wrote a guide on compiling qBittorrent 4.x on DietPi for ARM devices. DietPi ships with a really old version of qBittorrent from late 2016. I think users of qBittorrent could find this guide useful. I propose adding a link to the guide on the qBittorrent software entry page.  

**Compilation guide**
https://github.com/qbittorrent/qBittorrent/wiki/Compiling-qBittorrent-nox-4.x-on-DietPi-6.16-(Debian-9.0)-for-ARM-devices

**DietPi Entry for qBittorrent**
https://dietpi.com/phpbb/viewtopic.php?t=5&start=70#p2272

MichaIng, 2018-09-22:
@Millichrome
Thanks for this. We'll have a look if we can implement this into our install script.

Millichrome, 2018-09-24:
Let me know if there is any way I can help with testing. I have a Raspberry Pi 3 B+ that I can test on and an ASUS tinker board as my "production" system. 

I don't have a lot of experience with linux, but I would be interested in trying to write a compilation script for qBittorrent 4.x.

MichaIng, 2018-10-08:
@Millichrome 
Btw. many thanks for your guide.

However some remarks, to have this running better on DietPi:
- For security reasons, run qbottorrent service with it's own user and for cross-access reason with group "dietpi". This allows download managers, sonarr and such, as well as media libraries/players to access the downloads of qbittorrent.
Add user: `useradd -rm qbittorrent -G dietpi -s /usr/sbin/nologin`.
Adjust service:
```
[Service]
User=qbittorrent
Group=dietpi
```
- Since `dietpi-services` handles (known) service start/stop, they do not need to be enabled (systemd controlled), at best stay disabled.

Besides that, great work. Will try it out tomorrow.

MichaIng, 2019-08-11:
I mark this as closed. We shift our images to Buster already.
However interested Stretch users can still follow this guide: https://github.com/qbittorrent/qBittorrent/wiki/Compiling-qBittorrent-nox-4.x-on-DietPi-or-Raspbian-(Debian-9.0)-for-ARM-devices

closed 2019-08-11 by MichaIng

crankynet, 2021-03-30:
What happened to the guide at
https://github.com/qbittorrent/qBittorrent/wiki/Compilation:-DietPi-and-Raspbian-(Debian-9.0) ?