MichaIng/DietPi #2086: Issue with Xserver and booting LXDE desktop on Rock64

pfordmedia, 2018-09-22:
Total coding newbie here, but I'm having issues. Also, please excuse my lack of knowledge of how GitHub works as I'm new to that as well.
#### Required Information:
- DietPi version	| 6.16 
- Distro version	| Debian 9 stretch
- Kernel version	| Linux DietPi 4.4.132 #1 SMP Thursday Jul 26 08:22:22 UTC 2018 aarch64
- SBC device		| Pine64 Rock64 ARM64 4gb.
- Power supply used	| Pine64 5v 3A
- SDcard used		| SanDisk Ultra 32GB

#### Additional Information (if applicable):
- Software title	| 
- Was the software title installed freshly or updated/migrated? DietPi flashed with PineInstaller64 for Windows.
- Can this issue be replicated on a fresh installation of DietPi? Yes
- `dietpi-bugreport` ID

#### Steps to reproduce:

1.  Format SD card using Windows disk utility
2.  Flash DietPi 6.16 Rock64 ARMv8 Stretch to SD using Etcher
3.  Boot from SD 
4. Install DietPi
5. Install LXDE from `dietpi-software` without setting  autostart options
6. Reboot and login
7. In console, run `startx`

#### Expected behaviour:

- ... LXDE desktop should begin and open.

#### Actual behaviour:

- `root@DietPi:~# startx`
`xauth: file /root/ .Xauthority does not exist`
 
`X.org X Server 1.19.2`

and then below XServer info:

`(==) Log file: "/var/log/Xorg.0.log", Time: Sat Sep 22 03:46:37 2018`
`(==) Using config directory: "/etc/X11/xorg.conf.d"`
`(==) Using system config directory "usr/share/X11/xorg.conf.d"`
`(EE)`
`Fatal server error:`
`(EE) no screens found (EE)`
`(EE)`
`Please consult The X.Org Foundation support`
`at http://wiki.x.org`
`for help`
`(EE) Please also check the log file at "/var/log/Xorg.0.log" for additional information.`
`(EE)`
`(EE) Server terminated with error (1). Closing log file.`
`xinit: giving up`
`xinit: unable to connect to X server: Connection refused`
`xinit: server error`

#### Extra details:

- When I check the log file using `tail -f`it basically gives me the same message.  Is this a result of a missing video driver on my chipset or a driver that is incompatible with the Rock64 chipset? The X.org wiki shows this issue in their FAQ, however their fixes did not have any effect. Please see this page: https://www.x.org/wiki/FAQErrorMessages/#index8h2

Fourdee, 2018-09-22:
@pfordmedia 

Hi, 

Many thanks for the report 👍  I'll try to replicate.

Fourdee, 2018-09-22:
Confirmed, taking a look now.

Fourdee, 2018-09-22:
> https://github.com/rockchip-linux/rk-rootfs-build/tree/master/packages/armhf/libmali

🈴 
```
dpkg --add-architecture armhf
G_AGUP
wget https://github.com/rockchip-linux/rk-rootfs-build/raw/master/packages/armhf/libmali/libmali-rk-utgard-450-r7p0_1.6-1_armhf.deb
dpkg -i *.deb
G_AGF
```

Fourdee, 2018-09-22:
@pfordmedia 

Workaround is to remove the Mali GPU xorg definition:
```
mv /etc/X11/xorg.conf.d/20-armsoc.conf /root/
startx
```

This allows X to init under software mode.

I'am not having much success with HW GPU currently:(https://github.com/Fourdee/DietPi/issues/2086#issuecomment-423741172)

pfordmedia, 2018-09-22:
@Fourdee I was able to install a stable version from Pine Installer which has version 156 preloaded before. Then I reflashed with 6.16 and did an apt-get update and from then on out I’ve been getting this error. 

Haven’t tried to flash an older version though after the apt-get update. I may try that now.

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

pfordmedia, 2018-09-22:
Sorry I just saw the workaround. This works great! Thanks so much!

Fourdee, 2018-09-27:
Completed.

However, as above, we only offer software rendering for the GPU at this time.

closed 2018-09-27 by Fourdee

denni-isl, 2018-11-28:
> @pfordmedia
> 
> Workaround is to remove the Mali GPU xorg definition:
> 
> ```
> mv /etc/X11/xorg.conf.d/20-armsoc.conf /root/
> startx
> ```
> 
> This allows X to init under software mode.
> 
> I'am not having much success with HW GPU currently:([#2086 (comment)](https://github.com/Fourdee/DietPi/issues/2086#issuecomment-423741172))

Did work like a charm.    Quick fix.  Thanks!