MichaIng/DietPi #2087: wifi adapter doesnt work on 6.16 (fresh install)

sesshomaru86, 2018-09-22:
When I'm upgrading from 6.14 to 6.16 the wifi adapter stopped working .I can't scan for a network through the GUI and even after i manually insert the wireless SSID and PASSWORD stil doesnt resolve it and it gives me no ip ( 0.0.0.0.). If i put a static one doesn't seem to work too.
I tried 2 more times (clean install) from the image with wifi preconfigured in the dietpi.txt as also with ethernet. Worked everything in the clean install of the 6.14 but after the initial update to the 6.16 the wifi adapter breaks. Ethernet still working without a problem. Is there a method to remain to the 6.14 as i need to use the raspi wirelessly. I tried putting CONFIG_CHECK_DIETPI_UPDATES=0 but it still update my initial 6.14 version. Should i try put a fake adress on the APT mirrors ?Also can i try to put on DEV_GITBRANCH=testing on the dietpi.txt to see if the problem is resoved on the 6.17 ? Thanx for your constant work and help !

- DietPi version	       6.16
- Distro version	        Stretch 9.4
- Kernel version	        Linux DietPi 4.14.70-v7+ #1144 SMP Tue Sep 18 17:34:46 BST 2018 armv7l GNU/Linux
- SBC device		        RPi 3 Model B (armv7l)

- Power supply used	5V 2.5A
- SDcard used		        Sansidk ultra 16gb class 10 a1

#### Additional Information (if applicable):

- Was the software title installed freshly or updated/migrated? no additional software installed 
- Can this issue be replicated on a fresh installation of DietPi? yes 
- `dietpi-bugreport` ID   19f71c42-ac59-462a-85ff-27e8fab4d517

#### Steps to reproduce:
downlaod image from dietpi site flash it on the sd and then install normally 

#### Expected behaviour:
Wireless should work 

#### Actual behaviour:
it doesn't

[image]

Fourdee, 2018-09-22:
@sesshomaru86 

Hi, thanks for the report 👍 

I'll try to replicate.

Fourdee, 2018-09-22:
Confirmed bug with current image, redoing the image now.

Fourdee, 2018-09-22:
@sesshomaru86 

Image updated which resolves WiFi issues, please re-download and write the following image:
https://dietpi.com/downloads/images/DietPi_RPi-ARMv6-Stretch.7z

sesshomaru86, 2018-09-22:
Hmm..when i go to dietpi-config to change the network adapter settings im getting a message :

 [FAILED] DietPi-Config | DEBUG: /var/lib/dietpi/dietpi-wifi.db does not exist, unable to continue. If you did not manually remove this file, please report this issue to DietPi.

It's a fresh install with the new image posted above without tinkering anything at all.

Fourdee, 2018-09-23:
@sesshomaru86 

Image updated which applies the fix automatically for this issue:
https://github.com/Fourdee/DietPi/issues/2087#issuecomment-423836528

Image:
https://dietpi.com/downloads/images/DietPi_RPi-ARMv6-Stretch.7z

Completed.

Fourdee, 2018-09-24:
@sesshomaru86 

```
ifup: failed to bring up wlan0
```

Ok appears to be an issue with bringing up adapter, is the onboard WiFi the only connected WiFi adapter?

Ideally, we need to check the system configuration at this state when the WiFi fails. Please can you ensure ethernet is connected, get the WiFi to fail as above, then send me a new bug report:
```
dietpi-bugreport
```

I can then check the logs etc, see where the issue is.

Fourdee, 2018-09-24:
- 🈯️ Unable to replicate in local testing, fresh install then straight to ```dietpi-config``` to setup wifi
[image]

sesshomaru86, 2018-09-24:
Thanx for the reply !

I just download a new image from the site to be sure im not messing with the images reflash it test it again as new without any modifications. same problem.

Bug report sent, reference code: 155689e2-5ada-4461-8d04-051cbfc8a627

Fourdee, 2018-09-24:
@sesshomaru86 

Thanks 👍 

Appears the WiFi chip is working fine:
```
[   97.698885] brcmfmac: brcmf_c_preinit_dcmds: Firmware version = wl0: Oct 23 2017 03:55:53 version 7.45.98.38 (r674442 CY) FWID 01-e58d219f
[   97.699539] brcmfmac: brcmf_c_preinit_dcmds: CLM version = API: 12.2 Data: 7.11.15 Compiler: 1.24.2 ClmImport: 1.24.1 Creation: 2014-05-26 10:53:55 Inc Data: 9.10.39 Inc Compiler: 1.29.4 Inc ClmImport: 1.36.3 Creation: 2017-10-23 03:47:14 
```

```
brcmfmac              307200  0
brcmutil               16384  1 brcmfmac
cfg80211              573440  1 brcmfmac
rfkill                 28672  2 cfg80211
```

---

Ok I believe the issue may be with the WiFi creds. I'll need to see these. I've updated the bugreport script to include them, please run the following commands to resend bug report:
```
 wget https://raw.githubusercontent.com/Fourdee/DietPi/testing/dietpi/dietpi-bugreport -O /DietPi/dietpi/dietpi-bugreport
dietpi-bugreport
```

Also, please can you confirm which WiFi encryption mode is setup on your router (eg: WPA-PSK)?

sesshomaru86, 2018-09-24:
Tried again and this time when i pressed scan doesnt do anything so i put my credentials again manually.

I use a Unifi AP  AC LR as acess point and the encryption is WPA2 -AES . Can it be that the wifi password includes these type of characters [  `   ?

Tried with the above command 

dietpi-bugreport
[  OK  ] DietPi-Bugreport | Root access verified.
[  OK  ] DietPi-Bugreport | RootFS R/W access verified.

[ INFO ] DietPi-Bugreport | Packing upload archive, please wait...
[ .... ] DietPi-Bugreport | (1/5) Testing connection to ssh.dietpi.com, please w[  OK  ] DietPi-Bugreport | Connection test: ssh.dietpi.com
stat: cannot stat '155689e2-5ada-4461-8d04-051cbfc8a627.7z': No such file or directory
/DietPi/dietpi/dietpi-bugreport: line 155: ((: <= 10000000 : syntax error: operand expected (error token is "<= 10000000 ")
[FAILED] DietPi-Bugreport | The bug report upload archive appears to be unexpected large. Please inspect and in case clean up the locations to be uploaded, as their size should never be that large:
CMD_OUT.txt
G_ERROR_HANDLER_GITREPORT
/var/log/*
/DietPi/*
/boot/dietpi.txt
/boot/config.txt
/boot/dietpi/*
/tmp/.G*
/var/tmp/dietpi/logs/*
/var/lib/dietpi/*
/etc/bash.bashrc
/etc/bashrc.d/*
/root/.bashrc
/home/*/.bashrc
/etc/profile
/etc/profile.d/*
/root/.profile
/home/*/.profile
/etc/rc.local
/etc/X11/xorg.conf
/etc/asound.conf
/etc/network/interfaces
/etc/wpa_supplicant/wpa_supplicant.conf
/var/lib/dietpi/dietpi-wifi.db
/etc/fstab
/etc/sysctl.conf
/etc/sysctl.d/*
/etc/init.d/*
/etc/systemd/system/*
/lib/systemd/system/*
/etc/apt/sources.list
/etc/apt/sources.list.d/*

sesshomaru86, 2018-09-24:
Tried just now with a hotspot from my phone and it works so it must be a problem with my router or my password.

Fourdee, 2018-09-24:
@sesshomaru86 

> includes these type of characters [ ` ?

Yep, I believe ` is the issue, try escaping that character when adding in the key:
\`

eg:
thisismy\`key

sesshomaru86, 2018-09-24:
It worked ! Thanx a lot !

Fourdee, 2018-09-24:
@sesshomaru86 

Excellent 👍 i'll add some more info when entering WiFi keys, to mention backslash required for those chars.

MichaIng, 2018-09-24:
Theoretically we could also add those backslashes via ${value/[/\[} but maybe this confuses more than it helps?

Fourdee, 2018-09-24:
@MichaIng 

> Theoretically we could also add those backslashes via ${value/[/[} but maybe this confuses more than it helps?

👍 Should be fine as long as we mention that backslashes will automatically be added for the required chars?

Fourdee, 2018-09-24:
[image]

Fourdee, 2018-09-24:
Completed.

sesshomaru86, 2018-09-24:
Just for the reference . I tried another new install but i couldnt connect it to the wifi again . Moreover every time i was pasting the password automatically was closing the field with the password. At the end it couldnt connect even to a hotspot with no passwrod at all. I proceed with the installation through ethernet , i rebooted and that time got everything flawlessly.

Fourdee, 2018-09-25:
@sesshomaru86 

> reproducing steps : Put a password in the pass tab of the ssid and then exit back to the menu with all the networking options , re enter again in the wifi adapter select network and the password is only half

Many thanks, i'll reopen for testing 👍

Fourdee, 2018-09-26:
```
aWIFI_KEY[0]=`$"'test
```

```
					# - Add required escapes for bash related items
					#	\"`$
					sed -i 's/\\/\\\\/g' /var/lib/dietpi/dietpi-wifi.db
					sed -i 's/\"/\\"/g' /var/lib/dietpi/dietpi-wifi.db
					sed -i 's/`/\\`/g' /var/lib/dietpi/dietpi-wifi.db
					sed -i 's/\$/\\$/g' /var/lib/dietpi/dietpi-wifi.db

					cat /var/lib/dietpi/dietpi-wifi.db | grep 'aWIFI_KEY\[0\]'
```

```
 DietPi-Set_Hardware
─────────────────────────────────────────────────────
 Mode: wificreds (set)

aWIFI_KEY[0]=\`\$\"'test
/var/lib/dietpi/dietpi-wifi.db: line 6: unexpected EOF while looking for matching `''
/var/lib/dietpi/dietpi-wifi.db: line 96: syntax error: unexpected end of file
[  OK  ] wificreds set | Completed
```

MichaIng, 2018-09-26:
@Fourdee 
Try to avoid `sed` and `grep` here. as they expand/interpret several magic chars on each call.
`${value\\[/\\[}` and such should be safer. Jep mixed backslash with slash above, hope with doubled backslash a single backslash is added. The initial double backslash makes the change apply for every occurrence instead of just the first one.

Fourdee, 2018-09-27:
Notes:

- Can only be applied to one pattern per use:
```
root@DietPi:~# echo $test
`$dkfoasfdkao

root@DietPi:~# echo ${test//\`/\\\`}
\`$dkfoasfdkao
```

Fourdee, 2018-09-27:
Ok testing:
- WiFi key in router = ```testtest`"$1```
- Entered key = ```testtest\`\"\$1```
- 🈴 Does not connect
- wpa_supplicant.conf
```
network={
        ssid="XXXXXXXXXXXX"
        scan_ssid=1
        key_mgmt=WPA-PSK
        psk="testtest\`\"\$1"
}
```

MichaIng, 2018-09-28:
@Fourdee 
Sadly I have no chance to test/debug WiFi here 😞.

However my suggestions:
- `aWIFI_KEY[$i]="$( echo ${aWIFI_KEY[$i]} | sed 's/\\/\\\\\\/g' )"`
  - Since `sed` interprets it's own magic chars again, better use `${aWIFI_KEY[$i]//\\/\\\\}` to double backslashes etc.
- Currently `dietpi-wifi.db` creation/applying happens in both, `dietpi-config` and `dietpi-set_hardware`. I see the need currently that dietpi-config needs to safe the data and dietpi-set_userdata wants to recreate it for sanity, especially on first run setup, when user uses `dietpi-wifi.txt` 🤔. Anyway maybe we find a more elegant solution.
  - Actually since now the `dietpi-wifi.db` is effectively recreated two times, one time within dietpi-config and a second time within dietpi-set_hardware, this also doubles the need to backslash magic chars. We should really try to touch strings and file as less as possible. Maybe in this case it is best to create a separate script, e.g. dietpi-wifi_creds/wifi_setup or something like that? Then open menu, when this is called without argument (from dietpi-config, Network > WiFi > dietpi-wifi), allow to simply apply existing `dietpi-wifi.db` and perhaps as separate 1st run option, allow to copy from `dietpi-wifi.txt`. This would reduce the need to read/write those settings just a single time.
- `for (( i=0; i<5; i++ ))`: Use `for i in ${!aWIFI_SSID[@]}` instead to not pre-generate all 5 arrays and safe dummy arrays to file? Also this allows theoretically unlimited SSIDs. Pre-generate a new array index when user chooses (new menu entry) to add a new one: `new_index=$((${#aWIFI_SSID[@]}))`
  - In case of removed SSIDs, better find first empty index: `for (( i=0; i<=${#aWIFI_SSID[@]}; i++ )); do [[ ! ${aWIFI_SSID[$i]} ]] && new_index=${aWIFI_SSID[$i]}; pre-generate array etc...`

MichaIng, 2018-10-28:
Did further enhancements about this: https://github.com/Fourdee/DietPi/pull/2193
- No manual escaping required any more. Testing on Jessie would be great, just to verify that Jessie wpa_supplicant parses the config file identically.

MichaIng, 2018-11-18:
@panjisn 
If your password is "`wakanda`", so does not include any special characters, there shouldn't be any issues.

If your password contains one of $ \` " \ , then on current v6.17 your need to escape those via backslashes, e.g. "`w$k\nd"`" should be "`w\$k\\nd\"`" then.

With v6.18 we do changed the way the pw string is stored (from double quotes " to single quotes '), which allows more special characters, and do the remaining escaping script internally. So no manual escaping is required any more.

Fourdee, 2018-11-18:
We can actually mark this as completed. Testing passed, no need to escape chars in v6.18.

closed 2018-11-18 by Fourdee