MichaIng/DietPi #2088: DietPi-Software | myMPD not funcitional

Fourdee, 2018-09-22:


Fourdee, 2018-09-22:
```
[FAILED] DietPi-Services | mympd        failed (Result: exit-code) since Sat 2018-09-22 15:44:47 BST; 4min 15s ago
● mympd.service - myMPD
   Loaded: loaded (/lib/systemd/system/mympd.service; disabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Sat 2018-09-22 15:44:47 BST; 4min 15s ago
  Process: 9452 ExecStart=/usr/bin/mympd /etc/mypd/mympd.conf (code=exited, status=1/FAILURE)
 Main PID: 9452 (code=exited, status=1/FAILURE)

Sep 22 15:44:47 DietPi systemd[1]: Started myMPD.
Sep 22 15:44:47 DietPi mympd[9452]: Parsing config file: /etc/mypd/mympd.conf
Sep 22 15:44:47 DietPi mympd[9452]: Can't load config file "/etc/mypd/mympd.conf"
```

Fourdee, 2018-09-22:
Actually a bug our end 😮 

Resolved for v6.17.

## Pre v6.17 solution, is to copy and paste all after installation:
```
cat << _EOF_ > /lib/systemd/system/mympd.service
[Unit]
Description=myMPD
After=mpd.service

[Service]
Type=simple
#User=mympd # Forks its own process using mympd user
ExecStart=$(which mympd) /etc/mympd/mympd.conf

[Install]
WantedBy=multi-user.target
_EOF_
systemctl daemon-reload
dietpi-services restart
```

Fourdee, 2018-09-22:
- 🈯️ Reinstall test passed

Completed.

sfb2, 2018-09-22:
hi there,
just to confirm->ran into the same issues->"Can't load config file "/etc/mypd/mympd.conf"

with your workaround it works !! (and absolutely funny, have just used dietpi after weeks, only to "see" the praised mympd, had the failure, and saw that u´ve just working on it ;-) )

BUT: update database didnt find my attached usb-drive (O!mpd does) and it didnt show a poster at all from the song (is it not connected to an online database ?)

but many thx anyway ;-)

Fourdee, 2018-09-23:
@sfb2 

> BUT: update database didnt find my attached usb-drive

You'll need to attach the drive using ```dietpi-drive_manager```, then move the userdata to that drive using the same program.

After thats done, put your music in
```
/mnt/dietpi_userdata/Music
```

Then refresh database.

Fourdee, 2018-09-23:
- 🈴 Issue with non-samba playback and ```mpd``` under these permissions.
```
Sep 23 23:14 : exception: Failed to open "DietPi Alsa" [alsa]
Sep 23 23:14 : exception: nested: Failed to open ALSA device "default": Permission denied
```

Fourdee, 2018-09-24:
- 🈴 Issue with non-samba playback and ```mpd``` under these permissions.
https://github.com/Fourdee/DietPi/issues/2067
```
Sep 23 23:14 : exception: Failed to open "DietPi Alsa" [alsa]
Sep 23 23:14 : exception: nested: Failed to open ALSA device "default": Permission denied
```

Fourdee, 2018-09-24:
Ok, so the issue is running the process with group ```dietpi```, seems to ignore the system groups which are available to it:
```
root@DietPi:~# cat /var/log/mpd/mpd.log
Sep 24 15:43 : avahi: Failed to create client: Daemon not running
Sep 24 15:44 : exception: Failed to open "DietPi Alsa" [alsa]
Sep 24 15:44 : exception: nested: Failed to open ALSA device "default": Permission denied
Sep 24 15:44 : exception: Failed to open "DietPi Alsa" [alsa]
Sep 24 15:44 : exception: nested: Failed to open ALSA device "default": Permission denied

root@DietPi:~# groups mpd
mpd : mpd audio dietpi
```

Fourdee, 2018-09-24:
Confirmed working for local and remote samba userdata directories.

Completed.

closed 2018-09-24 by Fourdee