MichaIng/DietPi #2089: multiple errors

symbios24, 2018-09-23:
Hello upon installation of some software in a raspberry pi 3

i got multiple errors

image gallery: 

single_file_php_gallery_4.7.0.zip
exit code 5
spider mode enabled
check if remote file exists
error the certificate of dietpi.com is not trusted
the certificate of dietpi.com is not yet activated

i also got an error trying to install sickrage/rtorrent/ about an *unrar.deb* file that cannot be found with the same dietpi.com certificates errors up i assume this unrar file is used in many software

thanks

symbios24, 2018-09-23:
i forgot i also have this errors in plex server

[FAILED] DietPi-Process_tool | Plex Media Server () Affinity  0-3
[FAILED] DietPi-Process_tool | An issue has occured

The gui loads ok but i did not proceed further than the create account page

Fourdee, 2018-09-23:
@symbios24 

> error the certificate of dietpi.com is not trusted

Appears to be an issue with date on the system. Please try running:
```
systemctl start systemd-timesyncd
```

Then check status with:
```
systemctl status systemd-timesyncd -l
```

It may take some time, however, once the time is sync you should see the below:
```
 systemd-timesyncd[1591]: Synchronized to time server 178.
62.16.103:123 (0.debian.pool.ntp.org).
```

Once completed, software installations should be functional.

Which DietPi version are you running?

And please paste results
```
systemctl status plex* -l
```

symbios24, 2018-09-24:
Latest DietPi 6.16,its strange because i did some installations successfully,and after i got this errors on others.

symbios24, 2018-09-24:
i cannot paste all this from the raspberry i have to write and its not possible.

symbios24, 2018-09-24:
i cannot paste all this from the raspberry i have to write and its not possible its too long.can you replicate the error?

symbios24, 2018-09-24:
i cannot paste all this from the raspberry i have to write it and its not possible its too long.can you replicate the error?

symbios24, 2018-09-24:
i tried before with the phone,but its not possible to take a picture on a widescreen with all the info in it in such an old phone.

thanks anyway

symbios24, 2018-09-24:
i tried before with the phone,but its not possible to take a picture on a widescreen with all the info in it in such an old phone.i will wait for future fix since i`m unable to understand it my self

thanks anyway

symbios24, 2018-09-24:
the time synchronize trick worked the image gallery installed successful.

symbios24, 2018-09-24:
i tried before with the phone,but its not possible to take a picture on a widescreen with all the info in it in such an old phone.i will wait for future fix for plex server but still i dont know if this failed messages affect the operation of the server.

The time synchronize trick worked the image gallery installed successful.

thanks anyway

MichaIng, 2018-09-26:
@symbios24 
> [FAILED] DietPi-Process_tool | Plex Media Server () Affinity 0-3

We had some other case, where the process tool was not able to apply settings because of temporary spawned processes that disappear while process tool is working. But this was resolved with v6.15: https://github.com/Fourdee/DietPi/commit/14cba4e4cf840b7934a33a23d3ea01701c73e54a#diff-7e4ac76a426f5cc1dd4b18222d378606
- However there is still a smaller time window, where the process could disappear, nothing we can 100% solve currently. **But the failure has no negative impact anyway, it just looks ugly 😉.**

But to be sure, are you on current DietPi v6.16?

> The gui loads ok but i did not proceed further than the create account page

Is this still the case?
What does `systemctl status plex* -l` show?

symbios24, 2018-10-01:
Im on Dietpi v6.16 the systemctl shows a lot of things.that i cannot write here.it will be more easy to duplicate the process to see for your self.

MichaIng, 2018-10-06:
@symbios24 
Fixed the plex issue: https://github.com/Fourdee/DietPi/commit/7abf38d3eae5596f229d612702f108fd87982e04

Would be great if you could apply this tiny change as well to your `/DietPi/dietpi/dietpi-process_tool` and check if the error is gone when reapplying: `/DietPi/dietpi/dietpi-process_tool 1`

Fourdee, 2018-10-07:
@symbios24 @MichaIng 

🈯️ Issue resolved for v6.17, with following commit:
https://github.com/Fourdee/DietPi/commit/3cddaed25cee948a9f1346df6f4e167149bec5d6#diff-7e4ac76a426f5cc1dd4b18222d378606

🈯️ Verified with local testing.

Completed.

closed 2018-10-07 by Fourdee

MichaIng, 2018-10-07:
Two fixes in parallel, however no problem, since both are exactly the same 😄.

symbios24, 2018-10-12:
Hi, i dont get it what i type in the command line to fix the error?